arduino/arduino-ide #740: espota.py: error: option -p: invalid integer value: '{upload.port.properties.port}'

jandrassy, 2022-01-05:
network upload with esp8266 core doesn't have `upload.port.properties.port` property evaluated from network port discovery

IDE 2
```
.arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3" -I "/home/duro/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/espota.py" -i "192.168.1.113" -p "{upload.port.properties.port}" "--auth=a" -f "/tmp/arduino-sketch-41571935C07681B86282A62DC15FAFF8/Blink.ino.bin"
Usage: espota.py [options]

espota.py: error: option -p: invalid integer value: '{upload.port.properties.port}'
Upload error: Failed uploading: uploading error: exit status 2
```
IDE 1
>.arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3 -I /home/duro/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/espota.py -i 192.168.1.113 -p 8266 --auth= -f /tmp/arduino_build_169908/Blink.ino.bin

jandrassy, 2022-01-06:
the mdns record is
```
= wlp58s0 IPv4 esp8266-f0e0c7                                _arduino._tcp        local
   hostname = [esp8266-f0e0c7.local]
   address = [192.168.1.113]
   port = [8266]
   txt = ["tcp_check=no" "ssh_upload=no" "board="ESP8266_WEMOS_D1MINILITE"" "auth_upload=no"]
```
since `auth_upload` is not read too, IDE 2 opens the password dialog. IDE 1 doesn't ask for password.

silvanocerza, 2022-01-10:
@jandrassy by any chance have you tested with the Arduino CLI too? This could probably be an issue with it.

jandrassy, 2022-01-10:
> @jandrassy by any chance have you tested with the Arduino CLI too? This could probably be an issue with it.

with CLI (bundled with IDE) it works. the port is provided. (but asks for password too)
```
juraj@nuc:~/arduino-ide_2.0.0-rc3_Linux_64bit/resources/app/node_modules/arduino-ide-extension/build$ ./arduino-cli upload /home/juraj/Documents/Arduino/BasicOTA -p 192.168.1.113 -l network -b esp8266:esp8266:d1_mini_lite -v
Uploading to specified board using network protocol requires the following info:
Password: 
"/home/juraj/.arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3" -I "/home/juraj/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/espota.py" -i "192.168.1.113" -p "8266" "--auth=a" -f "/tmp/arduino-sketch-5ABD2AC4C1A55FCE95C4A36C8699AE33/BasicOTA.ino.bin"
Uploading...........................................................................................................................................................................................................................
```

octoplayer2, 2022-01-25:
I have found the same issue under Windows 64; again the CLI workaround works. 
I have not yet set a password on the board, and CLI demands one.  Entering <Return> gives Authentication Fail, but entering a dummy password Authenticates.
 BTW, would fixing the Port declaration issue to allow IP format  also address #770, to allow empty Port numbers?

octoplayer2, 2022-02-22:
Any sign of a fix for this? - I was using the CLI workaround, but when I upgraded to CLI 0.21 the OTA authenticates, but then says it cannot get a response from the device.. So I have reverted to 1.18.16, as I need the OTA.

me-no-dev, 2022-04-02:
Issue is that Port properties are being lost in the IDE, because in between the code it has been mapped to a different Port structure that does not even have `properties` object (the one that holds network port and so on).
Here is an example of what config is given to upload:
```json
{
    "wrappers_": {
        "1": {
            "wrappers_": null,
            "arrayIndexOffset_": -1,
            "array": [
                1
            ],
            "pivot_": 1.7976931348623157e+308,
            "convertedPrimitiveFields_": {
            }
        },
        "4": {
            "wrappers_": {
                "5": {
                    "arr_": [

                    ],
                    "valueCtor_": null,
                    "map_": {
                    },
                    "arrClean": true
                }
            },
            "arrayIndexOffset_": -1,
            "array": [
                "192.168.254.153",
                "esp32-7cdfa1e00e58 at 192.168.254.153",
                "network",
                "Network Port",
                [

                ]
            ],
            "pivot_": 1.7976931348623157e+308,
            "convertedPrimitiveFields_": {
            }
        }
    },
    "arrayIndexOffset_": -1,
    "array": [
        [
            1
        ],
        "espressif:esp32:esp32s3:PSRAM=disabled,FlashMode=qio,FlashSize=4M,LoopCore=1,EventsCore=1,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PartitionScheme=default,CPUFreq=240,UploadSpeed=921600,DebugLevel=debug",
        "/Users/ficeto/Documents/Arduino2/BasicOTA",
        [
            "192.168.254.153",
            "esp32-7cdfa1e00e58 at 192.168.254.153",
            "network",
            "Network Port",
            [

            ]
        ],
        true,
        true
    ],
    "pivot_": 1.7976931348623157e+308,
    "convertedPrimitiveFields_": {
    }
}
```

CLI in a mean time gives the following boards to the IDE:
```json
[
  {
    "port": {
      "address": "192.168.254.153",
      "label": "esp32-7cdfa1e00e58 at 192.168.254.153",
      "protocol": "network",
      "protocol_label": "Network Port",
      "properties": {
        ".": "esp32s3",
        "auth_upload": "no",
        "board": "esp32s3",
        "hostname": "esp32-7cdfa1e00e58.local.",
        "port": "3232",
        "ssh_upload": "no",
        "tcp_check": "no"
      }
    }
  },
  {
    "port": {
      "address": "/dev/cu.Bluetooth-Incoming-Port",
      "label": "/dev/cu.Bluetooth-Incoming-Port",
      "protocol": "serial",
      "protocol_label": "Serial Port"
    }
  },
  {
    "port": {
      "address": "/dev/cu.usbserial-1124110",
      "label": "/dev/cu.usbserial-1124110",
      "protocol": "serial",
      "protocol_label": "Serial Port (USB)",
      "properties": {
        "pid": "0xEA60",
        "serialNumber": "44b562375972ea118b553a1dcf25bb41",
        "vid": "0x10C4"
      }
    }
  },
  {
    "port": {
      "address": "/dev/cu.SLAB_USBtoUART",
      "label": "/dev/cu.SLAB_USBtoUART",
      "protocol": "serial",
      "protocol_label": "Serial Port (USB)",
      "properties": {
        "pid": "0xEA60",
        "serialNumber": "44b562375972ea118b553a1dcf25bb41",
        "vid": "0x10C4"
      }
    }
  }
]
```

The "bad" `Port` class interface is here: https://github.com/arduino/arduino-ide/blob/main/arduino-ide-extension/src/common/protocol/boards-service.ts#L145 It is part of `CoreService.Upload.Options` which is given to `doUpload` here: https://github.com/arduino/arduino-ide/blob/main/arduino-ide-extension/src/node/core-service-impl.ts#L130

I just could not trace how/where it was turned into that offending `Port` interface and lost all properties.

per1234, 2022-08-17:
I can still reproduce the issue with 2.0.0-rc9.2-snapshot-9e2b73a and esp8266 platform version 3.0.2

closed 2022-08-25 by kittaakos

bartgrefte, 2024-03-07:
@per1234 Is https://github.com/earlephilhower/arduino-pico/issues/1728#issuecomment-1732504337 the same as this? I'm running into this as well, see the comment after that.

zeroalphagit, 2024-04-05:
Any fix for this its happening for me on Arduino IDE 2.3.2 and esp8266 when attempting to do OTA upload

**espota.py: error: option -p: invalid integer value: '{upload.port.properties.port}'**

JAndrassy, 2024-04-05:
@zeroalphagit with what version of the esp32 platform?

zeroalphagit, 2024-04-08:
> @zeroalphagit with what version of the esp8266 platform?

I am using: Arduino IDE 2.3.2 with ESP8266 3.1.2 Library. 

I'm getting the same error **(espota.py: error: option -p: invalid integer value: '{upload.port.properties.port}')** on an older nodemcu (esp8266) and a newer Wemos D1 mini with ESP8266EX

dylandn, 2025-12-17:
I just got this error as well, IDE v2.3.6, Heltec WiFi Kit 32 with v3.0.3 libs

commit 3db92541 (2022-08-24): Use port properties from the discovery.

Closes #740
PR #1335 (2022-08-25): #740, #1331 Use port properties
### Motivation

To show the `hostname` of the `network` port in the boards dropdown and `Tools` > `Ports` menu (#1331)
To fix the upload caused by missing port `properties` (#740) 

### Change description

 - Removed unnecessary `id` from the `Port` type. It can be derived from the `address` and `protocol`.
 - Introduced a shared `Unknown` i18n string.
 - Removed the deprecated `getAttacheBoards`/`getAvailablePorts` service APIs.
 - Use the `addressLabel` instead of the `address` in the boards dropdown and `Tools` > `Ports` menu. (#1331)
 - Introduced an optiona [... cut]
commit f7f644cf (2022-08-25): Use port properties from the discovery.

Closes #740