arduino/arduino-ide #743: Allow editing of libraries, the core, and the toolchain from the Editor

JM-FRANCE, 2022-01-07:
The current version of the IDE prevents you from doing "risky" activities and puts libraries, the core, and the toolchain in read only mode, for example when access through the "Go to definition..." and "Peek definition" features.

This is meant to protect beginners from modifying key files which could lead to unplanned and hard to debug consequences outside their current project. So this is good.

But this also prevents advanced users from debugging libraries on the spot (fix warnings etc) and they need to use a separate editors for those files making the workflow less seamless. 

Feature request: Add an option to overrule the read-only mode for those files with proper warning / double confirmation.

DaveTur, 2022-01-20:
I agree that the ability to edit library files (especially your own files being developed) will be a very welcome addition.

darianbjohnson, 2022-10-23:
This used to be possible by adding a .development file in the root of the library folder; that doesn't seem to work in IDE 2.0.0

truher, 2022-11-06:
++votes

corneo, 2023-03-26:
I experienced this [nonsense, in my opinion] today and was shocked to learn that the 'feature' cannot be disabled.  In lieu of that, it would be nice to provide a way to get an explanation of why I can't edit a file rather than making me have to go hunt it down. (Thank goodness for Google and for people that speak up in the right forum to get 'usability' concerns like this attended to.)  I am surprised to see that it has not been addressed.  I read in the dialog that prompted the creation of this issue that the Arduino IDE is maintained by volunteers, so if there is much to be done I get that there is a backlog.

I'm writing what is becoming a relatively complex sketch, so I am motivated to move the classes I have written into the header(.h) files and implementation (.cpp) files, as is typically done.  I CANNOT figure out how to make it happen in the Arduino IDE.  Perhaps supporting that is considered 'too complex' for beginners.  I AM a beginner, and I have noticed the pattern (refactoring classes, as such) and want to emulate it.  I'm usually pretty good at figuring out how to use 'tools' such as the Arduino IDE but, I must say, I find very little of the basics to be intuitive.

Making it a bit easier to create and edit files other than .ino files would be very useful, in my opinion.

I get that you want to help people to avoid shooting themselves in the foot.  But the file I tried to create and edit IS under the folder that holds the 'main' -- well, only, so far -- .ino file.  I can't fathom any rational reason for that.  In cases such as these, I start out with the assumption that I'm probably doing something 'wrong.'  If so, I can't figure out what that is.

Anyway... I think it should be addressed.  I appreciate all that has already been done to make otherwise very complex issues easier to deal with, so thank you for all the good work so far.  But... moving on...