arduino/arduino-ide #746: Auto Format deletes first line of non-`.ino` sketch files

per1234, 2022-01-10:
## Describe the bug

In addition to the `.ino` file extension of the Arduino programming language, Arduino sketches may contain files of other programming languages, indicated by the appropriate file extension (e.g., `.c`, `.cpp`, `.h`).

:bug: When the Arduino IDE's Auto Format feature is used to format a sketch file with one of these file extensions other than `.ino`, the first line of the sketch is erased.

## To Reproduce

1. Select **File > New** from the Arduino IDE menus.
1. Click the downward pointing triangle button on the right side of the tab bar.
1. Select "**New Tab**" from the menu.
1. In the "**Name for new file**" field, type a filename that ends in `.c`, `.cpp`, or `.h` (e.g., `SomeTab.h`).
1. Click the <kbd>OK</kbd> button.
1. Add some code that is not correctly formatted. For example:
   ```
             // unformatted line
   ```
   (inappropriate leading space)
1. Select **Tools > Auto Format** from the Arduino IDE menus.

:bug: The first line of the file is erased.

## Expected behavior

Format the code of any supported sketch file without deleting any code.

## Desktop

- OS: Windows 10
- Version: 2.0.0-rc3-snapshot.4c831c6
Date: 2022-01-05T11:13:17.867Z
- Arduino CLI Version: 0.20.2 [13783819]
- Arduino Language Server version: 0.6.0
- clangd version: 13.0.0

## Additional context

The issue does not occur in `.ino` files.

---

Language server logs: [SomeSketch.zip](https://github.com/arduino/arduino-ide/files/7837032/SomeSketch.zip)

---

Originally reported at https://forum.arduino.cc/t/ctrl-t-deletes-first-line-s-of-file/941177

Pushpendra766, 2022-01-19:
Got the same issue.

per1234, 2022-06-08:
Closing as fixed by https://github.com/arduino/arduino-ide/pull/1019

closed 2022-06-08 by per1234