arduino/arduino-ide #754: Sketch not saved on Run

TS200010, 2022-01-16:
**Describe the bug**
Sketch is not saved on compile. When the IDE exits one is prompted this is easy to overlook and work is trashed. This behaviour is different from earlier versions.

**To Reproduce**
Onen Sketch.
Make edit
Run.
Exit - dialog pops up edits have not been saved.

**Expected behavior**
Edits saved on Run as current version.

**Screenshots**
None

**Desktop (please complete the following information):**
- OSX Catalina

**Additional context**
None

per1234, 2022-01-16:
Hi @TS200010. Thanks for your report.

Although it is true that Arduino IDE 2.x does not have the **File > Preferences > Save when verifying or uploading** feature of the classic Arduino IDE, it does have an equivalent **File > Preferences > Auto save** feature, which automatically saves the sketch every 500 ms.

This more modern approach to automatically saving the sketch is IMO superior to the old system because if I want my work to be automatically saved, I want everything to be saved, rather than whatever I had in the editor the last time I happened to compile or upload.

Just as it was for **Save when verifying or uploading**, **Auto save** is enabled by default.

closed 2022-01-16 by per1234

TS200010, 2022-01-17:
Am I missing something? Two days ago I lost all my work because I clicked through the dialog box that popped up when I closed the sketch. I lost all may work of several hours, not just the changes since the last compile.

per1234, 2022-01-17:
Do you have "**Auto Save**" enabled in the **File > Preferences** dialog?

If no, then just enable "**Auto Save**" in the preferences.

If yes, please provide detailed and precise instructions I can follow to easily reproduce the issue. Tell me exactly how you opened the sketch you are editing.

TS200010, 2022-01-17:
Hi,

I did not have auto save set.

I would suggest that AutoSave is enabled by default as this most closely mirrors the previous version.

When I clicked on don’t save I expected that the edits that I had made since the last time I compiled would not be saved. To have the edits revert back to three hours previously with umpteen (that’s a large number) compile and run cycles is not imo opinion behaviour that one would expect. I certainly do not call it modern behaviour. 

You need to consider the totality of the experience. Those that set the feature and those that do not. Here are my thoughts...

Those that set auto save: Save at predetermined intervals and always on compile and run.

Those that do not set auto save. Save changes in a temporary file on each compile and run. If the user exits without saving then warn the user that ALL changes since loading will be lost. 

In both cases, keep the last unsaved change in a backup file so the user can recover in the event they click though on the wrong button. In my case, the Fiile was nowhere… not in Time Machine nor was it in OneDrive backup. Poof it had gone.\

Just some thoughts. In my case I have Auto set now so I should be good.

Regards
Tony

per1234, 2022-01-17:
>I would suggest that AutoSave is enabled by default

It is.

>Save at predetermined intervals

The interval is 500 ms by default (configurable via the advanced settings).

>always on compile and run

There is no need. The worst case scenario is already to lose 500 ms worth of work.

>Save changes in a temporary file on each compile and run.

It is.

>If the user exits without saving then warn the user that ALL changes since loading will be lost.

It does. However, as I mentioned, there are several known bugs that affect exit with unsaved changes.

>keep the last unsaved change in a backup file

I think it is a reasonable request. There is a previous formal feature request for it here: https://github.com/arduino/Arduino/issues/3539