arduino/arduino-ide #756: Structure View / Method explorer

Jawomo, 2022-01-17:
Handling bigger projects can be a real pain with this IDE.

A Structure View where you have a list of variables and methods in a side window really helps with navigation and editing. Clicking on a method directly brings you to that method.

Would be great if we could select contents of structure view with small buttons. One for variables and one for methods e.g.

This is a basic feature which would make Arduino IDE som much better!

Thanks!

kittaakos, 2022-01-17:
> View where you have a list of variables and methods in a side window really helps with navigation and editing. Clicking on a method directly brings you to that method.

There is an _Outline View_. See here: https://github.com/arduino/arduino-ide/issues/237#issuecomment-801882266

Jawomo, 2022-01-17:
Why is it hidden that way? Why not add it to sidemenu by default? Why do i have to activate it again for every project?

Anyways its not working. It says "No outline informationavailable"

per1234, 2022-01-17:
Hi @Jawomo. Thanks for taking the time to submit an issue.

>Why is it hidden that way? Why not add it to sidemenu by default?

This is being tracked at: https://github.com/arduino/arduino-ide/issues/448

>Why do i have to activate it again for every project?

This is being tracked at: https://github.com/arduino/arduino-ide/issues/538

> Anyways its not working. It says "No outline information available"

This is being tracked at: https://github.com/arduino/arduino-ide/issues/124

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the others.

If you end up with additional information to share, feel free to comment in the appropriate thread.

closed 2022-01-17 by per1234