arduino/arduino-ide #757: Missing support for "open with" and file association

fstasi, 2022-01-19:
## Motivation
We need to provide file associations between .ino files and IDE2.

This involves implementing the following behaviours:

- associate a file type (namely, .ino files)
- allow the drag and drop of a file over the icon (macOS only) and open the relevant files

### Related

- https://forum.arduino.cc/t/file-open-with-right-mouse-key-does-not-work/856111
- https://forum.arduino.cc/t/incorrect-ino-file-loaded/879801
- https://forum.arduino.cc/t/beta-10-and-prior-wont-load-a-script-via-file-explorer-open-with/894968
- https://forum.arduino.cc/t/how-do-i-set-ide-2-0/897607
- https://forum.arduino.cc/t/arduino-2-0-problem/902476
- https://forum.arduino.cc/t/opening-sketch-in-file-explorer-doesnt-work/904514

fstasi, 2022-01-25:
assigned to Mark

per1234, 2022-02-01:
Closing as duplicate of https://github.com/arduino/arduino-ide/issues/72 and https://github.com/arduino/arduino-ide/issues/370

closed 2022-02-01 by per1234