arduino/arduino-ide #763: Splash screen

rozrabiak, 2022-01-22:
Hi, add do preferences "disable splash screen" OR change splash screen to not to show ALWAYS ON TOP. When Arduino IDE 2.0 starting splash screen is always ON TOP over all running applications.

per1234, 2022-01-22:
Hi @rozrabiak. Thanks for taking the time to submit a request.

>OR change splash screen to not to show ALWAYS ON TOP

We are already tracking this issue at https://github.com/arduino/arduino-ide/issues/324 and even have a proposed fix here: https://github.com/arduino/arduino-ide/pull/717

So this issue will be dedicated solely to your request:

>add to preferences "disable splash screen"

per1234, 2022-08-16:
Obviated by the removal of the splash screen entirely in https://github.com/arduino/arduino-ide/pull/1287

closed 2022-08-16 by per1234