arduino/arduino-ide #766: Example code for JoymouseControl can't find Mouse.h

iplayfast, 2022-01-22:
**Describe the bug**
I've installed this on linux, and when trying out the sample code USB\JoyMouseControl it can't seem to find the Mouse.h, even though it is there. 
I've done this on both the old ide and the new one.

**To Reproduce**
Steps to reproduce the behavior:
1. Open File/Examples/USB/Mouse/JoyMouseControl
2. Compile
3. Error  Mouse is not declared.

**Desktop (please complete the following information):**
 - OS: Linux Mint
 - Version 1.8.16, and 
 - Version: [e.g. 2.0.0]

per1234, 2022-01-23:
Hi @iplayfast. Thanks for taking the time to submit an issue.

I'm going to ask you to provide the full verbose compilation output.

Please do this:

1. Select **File > Preferences** from the Arduino IDE's menus.
1. Check the box next to "**Show verbose output during: > ☐ compilation**"
1. Click the **<kbd>OK</kbd>** button.
1. Select **Sketch > Verify/Compile** from the Arduino IDE's menus.
1. Wait for the compilation to fail.
1. Right click on the black **"Output"** pane at the bottom of the Arduino IDE 2.x window.
1. From the context menu, click **"Copy All"**.
1. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error messages in a reply here.
   In order to ensure the text is not corrupted, please use [code fencing](https://guides.github.com/features/mastering-markdown/).

per1234, 2022-01-23:
Problem was identified over on the forum. This is the normal and expected behavior when attempting to compile a sketch using the Mouse library for a board that is not compatible.

closed 2022-01-23 by per1234