arduino/arduino-ide #769: IDE 2 has to use the latest "pluggable monitor" gRPC APIs (was Pluggable monitor IDE-2)

fstasi, 2022-01-24:
*Jira issue:* https://arduino.atlassian.net/browse/ATL-85

None

PaulStoffregen, 2022-05-01:
@AlbyIanna - Any chance for pluggable monitor to be implemented by RC7?

AlbyIanna, 2022-05-02:
Hi @PaulStoffregen! Sadly, this was delayed for multiple reasons, the main ones being the complexity of the change (a pretty big refactor was necessary). BUT we're working on it! I can't make promises, but I believe it will be working soon.

If you're curious, this is the branch we're working on: https://github.com/arduino/arduino-ide/tree/pluggable-monitor

PaulStoffregen, 2022-05-02:
@AlbyIanna - Glad it's happening.  Do you have a Teensy board?  If not and you want one for testing, just say the word and I can send you one.  As far as I know, Teensy is the only 3rd party board (so far) using the new pluggable monitor API.

AlbyIanna, 2022-05-03:
Thank you so much @PaulStoffregen 😄  but I believe @ubidefeo can lend me one of his Teensy boards for testing!

closed 2022-06-08 by fstasi