arduino/arduino-ide #770: Allow uploading to Digispark

plampix, 2022-01-24:
**Is your feature request related to a problem? Please describe.**
Compiling a project for a Digispark works great, however, uploading doesn't work (it doesn't do anything if I hit upload). Presumably because the Digispark doesn't present a 'port' to upload to.

**Describe the solution you'd like**
I'd like to be able to upload a project when there's no port selected, for hardware that doesn't upload using a static port.

**Describe alternatives you've considered**
Uploading the .hex from CLI. It works, but would be nice if we could do it from the GUI, like we used to in Arduino 1.x

per1234, 2022-01-25:
Thanks so much for reporting this @plampix.

I see there was a regression of https://github.com/arduino/arduino-pro-ide/issues/364, which had been previously fixed by https://github.com/arduino/arduino-ide/commit/2831acc5b5b9f0c8744c946f1721925aaf5f8c9c

I bisected the regression to [the 2.0.0-rc1 release](https://github.com/arduino/arduino-ide/releases/tag/2.0.0-rc1). Uploading to boards without a port selected was still possible when using Arduino IDE [2.0.0-beta.12](https://github.com/arduino/arduino-ide/releases/tag/2.0.0-beta.12)

## To Reproduce

In case it will be helpful to the developers, I'll provide instructions to reproduce the issue.

1. Quit the IDE if it is running.
1. Delete the configuration folder:
   - Windows:
     ```
     C:\Users\<user name>\AppData\Roaming\arduino-ide\
     ```
   - Linux:
     ```
     ~/.config/arduino-ide/
     ```
   - macOS:
     ```
     ~/Library/Application Support/arduino-ide/
     ```
   This step is done because the IDE currently makes it otherwise difficult for those with internal serial ports to achieve the required state of having no port selected, due to the odd fallback port selection behavior (e.g., https://github.com/arduino/arduino-ide/issues/180).
1. Start the Arduino IDE.
1. Select **Tools > Board > Arduino AVR Boards > Arduino Uno** from the Arduino IDE menus.
   NOTE: I chose the Uno because you likely already have it installed, and the issue is not actually board-specific. You can install the ["Digistump AVR Boards" platform](https://raw.githubusercontent.com/ArminJo/DigistumpArduino/master/package_digistump_index.json) and select the Digispark if you prefer a more realistic reproduction rather than a bare minimum demo.
   You should now have Arduino Uno selected, but no port selection.
1. Select **Sketch > Upload** from the Arduino IDE menus.
   :bug: Note that nothing at all happens.
1. Click the "Upload" button.
   :bug: Note that nothing at all happens.

Of course, in the case of the Uno, a port selection is needed for an upload, but that is not the case with all boards. We decided previously that, rather than making assumptions, the IDE should always trigger an upload and let the upload tool decide what to do. Those unwarranted assumptions were removed by https://github.com/arduino/arduino-ide/commit/2831acc5b5b9f0c8744c946f1721925aaf5f8c9c

## Workaround

The problem is caused by the requirement imposed by the IDE that a port must be selected before an upload may be triggered, so the workaround is to select any serial port from the **Tools > Port** menu in the Arduino IDE.

The Digispark upload tool doesn't care about the port, so it makes no difference what the port is. If you don't happen to have a port available already, connect some device such as a regular Arduino board to your computer and use its port.

## Related

- https://github.com/arduino/arduino-ide/issues/702

plampix, 2022-01-25:
@per1234 Thanks a lot for confirming the issue. I didn't look at the beta releases, so I assumed it never worked in v2.

mklemarczyk, 2022-03-28:
Hello I have the same issue with IDE version rc5 and Arduino Mbed OS RP2040 Boards v3.0.0.
I use Raspberry Pi Pico from official distribution. Tested under Windows 10 64-bits.

The workaround works when the device is in boot mode that mount the flash as a drive.
I connected any Arduino board, just to be able to select COM port for the project.
The project was properly loaded to the Pico device.

On the other hand, similar device Arduino RP2040 Connect does not need the workaround. Even the device is mounted as storage device, the IDE detects some COM port for Arduino Connect.

It is good to mention, once the program is loaded the device COM port is detected. And program can be loaded without need of fake COM port.

AndKe, 2022-04-22:
@per1234   Thank you - the workaround worked as advertised :)

Farhankhosravi, 2022-07-27:
The problem still exist - we always need to make a fake serial port to make ST LINK working... 
it's not hard to fix, please work on it in next updates.

closed 2022-10-20 by per1234