arduino/arduino-ide #771: Can't select language packet in Preferences

ChrSchultz, 2022-01-24:
**Describe the bug**
I am not able to select other language packs for ui.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to prefernces
2. Click on '.language 'en'
3. try to select cn or de or each other...
4. remark, you are not able to select any other language , but en.
5. than I have a look into the installation Directory in laguage and all other languages are installed.

**Expected behavior**
I expect, selecting many other languages

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Window 10 21H2 -64bit
 - Version: 2.0.0 rc3

**Additional context**
none

per1234, 2022-01-25:
Hi @ChrSchultz it is possible to install the stock "language pack" extensions by following the instructions [here](https://github.com/arduino/arduino-ide/pull/625).

However, that only provides the base UI translations. None of the Arduino IDE-specific translations are in those extensions.
My understanding is that the integration of the Arduino translations depends on https://github.com/arduino/arduino-ide/issues/712

closed 2022-04-08 by AlbyIanna

qizaizi, 2022-04-10:
The plugin directory is in the installation location arduino2.0\Arduino IDE\resources\app\plugins