arduino/arduino-ide #776: Arduino IDE 2.0 RC on Ubuntu is a total disaster!

maciejmiklas, 2022-01-26:
I've installed latest RC on latest Ubuntu. I'm not getting any error on console, but the whole IDE is totaly unstable, so I'm not sure how this could be a release candidate?

1) I have small sketch, only few files, but for most of the time I can see in bottom corner Indexing .... and Building....  Manual build takes 2 seconds.
2) Formater stops working after a while, IDE has to be restarted in order to fix it
3) Menu displays key shortcuts but those are already used in Ubuntu, and instead other key shortcuts are working. Meaning: displayed shortcuts are incorrect. 
4) I have few cpp and h files, after few minutes IDE shows compilation errors but there are none. Restart helps here.
5) Jumping to declaration stops working after a few minutes as well.

Basically I have to restart IDE every few minutes, otherwise is shows non existing errors, formater is not working.  

I've been using Eclipse for development and I'm going back to Eclipse, because it's not possible to work with RC 2.0

per1234, 2022-01-26:
Hi @maciejmiklas. Thanks so much for giving the release candidate of Arduino IDE 2.0.0 a try and for your feedback!

>most of the time I can see in bottom corner Indexing .... and Building....

This is expected after every change to the sketch and after changing the board selection. This is because, unlike the classic Arduino IDE, Arduino IDE 2.x has [a language server](https://github.com/arduino/arduino-language-server) that provides parameter hints, problem detection, hover, goto, and other capabilities. The language server must understand your program, including the core, toolchain, and library dependencies, and so this is the reason for the frequent build and indexing processes.

There was a recent cosmetic bug (https://github.com/arduino/arduino-ide/issues/431) that caused the status bar to show the IDE was in the "indexing" state even though it was finished, but that has been fixed already. The fix was made after the 2.0.0-rc3 release, and so is currently only available for users of [the nightly build](https://github.com/arduino/arduino-ide#nightly-builds).

>Formater stops working after a while, IDE has to be restarted in order to fix it

We have an existing bug report about this here: https://github.com/arduino/arduino-ide/issues/396
I believe the auto format has been made much more reliable by the work on the Arduino Language Server that has occurred since the time of the 2.0.0-rc3 release, so I would recommend anyone experiencing such issues to try [the latest nightly build](https://github.com/arduino/arduino-ide#nightly-builds).

>Menu displays key shortcuts but those are already used in Ubuntu, and instead other key shortcuts are working. Meaning: displayed shortcuts are incorrect.

Please provide a list of the incorrect shortcuts.

>I have few cpp and h files, after few minutes IDE shows compilation errors but there are none.

Please provide detailed step by step instructions I can follow to reproduce this bug.

>Jumping to declaration stops working after a few minutes as well.

Please provide detailed step by step instructions I can follow to reproduce this bug.

Which board do you have selected?

What is the "jump to declaration" procedure you are following?

Does it occur with any code and any declaration, or only under specific conditions?

maciejmiklas, 2022-01-27:
I do not know exact steps to reproduce this problem, but I think that it happens when you do lots of refactoring. My project contains a few .ccp/.h files and it's in early stage of development, so I'm renaming methods/variables, renaming files, adding new files and new methods.  Sometimes I use build in refactor function, but sometimes I just rename method by hand.

Here is the project: https://github.com/maciejmiklas/CapacitivePlantMonitor/tree/main/Arduino/CapacitivePlantMonitor

Once this problem occurs I observe flowing:
1. Editor shows errors on code that has been refactored, but compilation is fine,
2. "go to definition" does not work (CTRL+click, F12),
3. code completion (CTRL+SPACE) shows "loading...." or similar message

maciejmiklas, 2022-01-27:
I've tried nightly build and this one seams to be working stable so far. At least I did not get such problem withing an hour. Please close this ticket.

closed 2022-01-27 by maciejmiklas