arduino/arduino-ide #778: OSError: [Errno 16] Device or resource busy: '/dev/ttyUSB0'

larryn46, 2022-01-26:
I'm using Linux Mint 20 and an ESP32 DEVKIT V1 on port /dev/tty/USB0. The Aduino IDE is arduino-ide_2.0.0-rc3_Linux_64bit. I compiled and ran a simple program Blink_simple. It uploaded and flashed the built-in LED as it should. I tried to upload another version of the same program, and got the message: "OSError: [Errno 16] Device or resource busy: '/dev/ttyUSB0' "
Here are the programs:
First program 

const char * programName = "dv7-6187 ~/Arduino/blink_simple";
/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/Blink
*/
#define LED LED_BUILTIN // LED_BUILTIN
// the setup function runs once when you press reset or power the board
void setup() {
  Serial.begin(115200);
  delay(100);
  Serial.println(programName);
  Serial.println(programName);
  Serial.println(programName);
  Serial.printf("LED is on pin: %d ", LED);
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED, HIGH);   // turn the LED on (HIGH is the voltage level)
  Serial.println("LED is HIGH");
  delay(2000);                       // wait for a second
  digitalWrite(LED, LOW);    // turn the LED off by making the voltage LOW
  Serial.println("LED is LOW");
  delay(2000);                       // wait for a second
}

Second program:
/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  https://www.arduino.cc/en/Tutorial/BuiltInExamples/Blink
*/

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}

Apparently, the Serial output from the first program doesn't allow the upload of the second program.

per1234, 2022-01-27:
Hi @larryn46. Thanks for your report.

Did you have Serial Monitor open when the upload failed?

larryn46, 2022-01-28:
yes

per1234, 2022-01-28:
Thanks for the information. So I think it must be this bug: https://github.com/arduino/arduino-ide/issues/726

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-01-28 by per1234