arduino/arduino-ide #787: Not operating as expected

cike-567, 2022-01-29:
**Describe the bug**
A clear and concise description of what the bug is.

After running the program, stop in the loading interface. (First picture)

The icon of the loading interface is located on the top layer to block other interfaces (Second picture)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Run program

**Expected behavior**
A clear and concise description of what you expected to happen.

If additional resources are needed, they can be packaged into the Arduino ide. Or provide the download address. In some areas, the network environment is not friendly, and the online download will be slow or unable to download.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

[image]

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Version: [e.g. 2.0.0]

Windows 11 22000.469

**Additional context**
Add any other context about the problem here.

The portable Arduino IDE is used。
processor：AMD Ryzen 5 3600X
Graphics card：xfx rx6700xt

per1234, 2022-01-29:
Hi @cike-567 

>The icon of the loading interface is located on the top layer to block other interfaces (Second picture)

This is already being tracked here: https://github.com/arduino/arduino-ide/issues/324 and we have a proposed improvement here: https://github.com/arduino/arduino-ide/pull/717

So we should restrict this issue exclusively to the discussion of the other thing you reported:

>After running the program, stop in the loading interface. (First picture)

I'm going to ask you to provide some additional information that might help us to identify the problem.

Please do this:

1. Start **File Explorer** (AKA "**Windows Explorer**").
1. Open the Arduino IDE installation folder
   (usually `C:\Program Files\Arduino IDE` or `C:\Users\<user name>\AppData\Local\Programs\Arduino IDE`).
1. In the folder listing of the Arduino IDE installation folder, hold the **<kbd>Shift</kbd>** key while clicking the right hand button on the mouse.
1. From the context menu, click "**Open PowerShell window here**".
   PowerShell will now open with the current directory set to the Arduino IDE installation folder.
1. Type `& ".\Arduino IDE"`.
1. Press the **<kbd>Enter</kbd>** key.
   Debug output should now be printed to the PowerShell window as the Arduino IDE attempts to start up.
1. Wait until the Arduino IDE start up hangs.
1. Switch back to the PowerShell window if the Arduino IDE window took the focus.
1. Right click on the title bar of the PowerShell window.
   This will open a context menu.
1. Select **Edit > Select All** from the context menu.
   This will select all the text in the PowerShell window.
1. Press **<kbd>Ctrl</kbd>+<kbd>C</kbd>**.
   This will copy the contents of the PowerShell window to the clipboard.
1. Paste (**<kbd>Ctrl</kbd>+<kbd>V</kbd>**) the output in a reply to this GitHub issue thread.

cike-567, 2022-01-29:
PowerShell 7.2.1
Copyright (c) Microsoft Corporation.

https://aka.ms/powershell
Type 'help' to get help.

PS C:\Users\46143\Downloads\Compressed\arduino-ide_2.0.0-rc3_Windows_64bit> & ".\Arduino IDE"
PS C:\Users\46143\Downloads\Compressed\arduino-ide_2.0.0-rc3_Windows_64bit>
Starting backend process. PID: 816
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///c%3A/Users/46143/.theia'
root ERROR (node:816) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
root INFO Theia app listening on http://localhost:50053.
root WARN The local plugin referenced by local-dir:/c%3A/Users/46143/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/c%3A/Users/46143/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:C:\Users\46143\.arduinoProIDE\plugins does not exist.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO Deploying backend plugin "cortex-debug@0.3.10" from "C:\Users\46143\Downloads\Compressed\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\plugins\cortex-debug\extension\dist\extension"
root INFO Deploying backend plugin "vscode-arduino-tools@0.0.2-beta.1" from "C:\Users\46143\Downloads\Compressed\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\plugins\vscode-arduino-tools\extension\dist\bundle"
root INFO Deploying backend plugin "cpp@1.52.1" from "C:\Users\46143\Downloads\Compressed\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\plugins\vscode-builtin-cpp\extension"
root INFO Deploying backend plugin "json@1.46.1" from "C:\Users\46143\Downloads\Compressed\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\plugins\vscode-builtin-json\extension"
root INFO Deploying backend plugin "json-language-features@1.46.1" from "C:\Users\46143\Downloads\Compressed\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\plugins\vscode-builtin-json-language-features\extension\client\dist\jsonMain"
root INFO Deploy plugins list took: 48.9 ms
root WARN Backend ConfigServiceImpl.onStart is slow, took: 110.6 ms
daemon INFO Starting daemon from C:\Users\46143\Downloads\Compressed\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
daemon INFO Using config file: c:\Users\46143\.arduinoIDE\arduino-cli.yaml
daemon INFO arduino-cli.exe version 0.20.2
daemon INFO Executing `arduino-cli daemon`
daemon INFO Setting up Prometheus metrics on :9090/metrics
daemon INFO {"IP":"127.0.0.1","Port":"50054"}
daemon INFO Daemon is running.

per1234, 2022-01-29:
I'm not seeing any red flags in the output unfortunately. There is usually a lot more output after that.

Does it hang at that point forever?

Were you ever able to use Arduino IDE 2.0.0, or have you experienced this hanging on startup problem from the very beginning?

About this statement:

>If additional resources are needed, they can be packaged into the Arduino ide. Or provide the download address. In some areas, the network environment is not friendly, and the online download will be slow or unable to download.

Do you suspect this could be the cause of the IDE hanging for you? Is there anything unusual or unfriendly about your network environment?

I have recently done some testing of the Arduino IDE when it does not have Internet access and identified some cases where it is a poor user experience (https://github.com/arduino/arduino-ide/issues/784, https://github.com/arduino/arduino-ide/issues/786), but nothing that would cause the IDE to hang.

cike-567, 2022-01-30:
I spent hours testing。From RC 3 to beta 12 were tested. All stay in the loading interface. But RC 3 ran normally once yesterday, and then it didn't run normally anymore.
Stay in the loading interface for more than one hour. This is the longest time I have tested, during which the Arduino ide has not changed. A month ago, I ran the Arduino ide on an earlier version of windows 11, and the Arduino ide worked normally.The Arduino IDE is not unresponsive. The options in the following figure still work

[image]

The following is the operation of Arduino ide rc 3

[image]

[image]

I tested it on the virtual machine and the program runs normally. （windows 11 dev 22543.1000）
The following is the log
Windows PowerShell
版权所有（C） Microsoft Corporation。保留所有权利。

安装最新的 PowerShell，了解新功能和改进！https://aka.ms/PSWindows

PS C:\Users\46143\Desktop\arduino-ide_2.0.0-rc3_Windows_64bit> & ".\Arduino IDE"
PS C:\Users\46143\Desktop\arduino-ide_2.0.0-rc3_Windows_64bit>
Starting backend process. PID: 11580
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///c%3A/Users/46143/.theia'
root ERROR (node:11580) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
root INFO Theia app listening on http://localhost:65341.
root WARN The local plugin referenced by local-dir:/c%3A/Users/46143/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/c%3A/Users/46143/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:C:\Users\46143\.arduinoProIDE\plugins does not exist.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO Deploying backend plugin "cortex-debug@0.3.10" from "C:\Users\46143\Desktop\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\plugins\cortex-debug\extension\dist\extension"
root INFO Deploying backend plugin "vscode-arduino-tools@0.0.2-beta.1" from "C:\Users\46143\Desktop\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\plugins\vscode-arduino-tools\extension\dist\bundle"
root INFO Deploying backend plugin "cpp@1.52.1" from "C:\Users\46143\Desktop\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\plugins\vscode-builtin-cpp\extension"
root INFO Deploying backend plugin "json@1.46.1" from "C:\Users\46143\Desktop\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\plugins\vscode-builtin-json\extension"
root INFO Deploying backend plugin "json-language-features@1.46.1" from "C:\Users\46143\Desktop\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\plugins\vscode-builtin-json-language-features\extension\client\dist\jsonMain"
root INFO Deploy plugins list took: 79.2 ms
root WARN Backend ConfigServiceImpl.onStart is slow, took: 189.7 ms
daemon INFO Starting daemon from C:\Users\46143\Desktop\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
daemon INFO Using config file: c:\Users\46143\.arduinoIDE\arduino-cli.yaml
daemon INFO arduino-cli.exe version 0.20.2
daemon INFO Executing `arduino-cli daemon`
daemon INFO Setting up Prometheus metrics on :9090/metrics
daemon INFO {"IP":"127.0.0.1","Port":"65344"}
daemon INFO Daemon is running.
root INFO Git askpass helper is listening on http://127.0.0.1:65349.
daemon INFO Checking if CLI is Bundled into the IDE
daemon INFO Adding libraries dir [dir: C:\Users\46143\Documents\Arduino\libraries, location: user]
daemon INFO URL: https://downloads.arduino.cc/packages/package_index.json
daemon INFO Updating index [url: https://downloads.arduino.cc/packages/package_index.json]
root ERROR Error: Git installation not found.
    at C:\Users\46143\Desktop\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\node_modules\find-git-exec\lib\find-git-exec.js:69:69
    at async DefaultGitInit.init (C:\Users\46143\Desktop\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\node_modules\arduino-ide-extension\lib\node\theia\git\git-init.js:22:49)
root WARN Frontend CommonFrontendContribution.configure is slow, took: 491.5 ms
root WARN Failed to resolve module: @types/body-parser
root WARN Failed to resolve module: @types/cookie
root WARN Failed to resolve module: @types/dompurify
root WARN Failed to resolve module: @types/express
root WARN Failed to resolve module: @types/fs-extra
root WARN Failed to resolve module: @types/lodash.debounce
root WARN Failed to resolve module: @types/lodash.throttle
root WARN Failed to resolve module: @types/react
[... 117 lines cut ...]
[... cut ...]
    at Stream.stream.write (C:\Users\46143\Desktop\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\node_modules\map-stream\index.js:104:7)
    at Stream.ondata (internal/streams/legacy.js:17:31)
    at Stream.emit (events.js:223:5)
    at drain (C:\Users\46143\Desktop\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\node_modules\through\index.js:36:16)
    at Stream.stream.queue.stream.push (C:\Users\46143\Desktop\arduino-ide_2.0.0-rc3_Windows_64bit\resources\app\node_modules\through\index.js:45:5)
serial-service INFO >>> Disposing serial service...
serial-service INFO <<< Disposed serial service.

per1234, 2022-02-01:
Thanks for the information!
> RC 3 ran normally once yesterday, and then it didn't run normally anymore.

I want to be sure I understand the timeline correctly. Please tell me which of the two possible scenarios described below is correct:

## Scenario A

1. IDE ran normally the first time you used it.
2. IDE started hanging at splash screen on subsequent attempts to use it.

## Scenario B

1. IDE started hanging at splash screen.
1. IDE ran normally during some random attempt.
2. IDE went back to hanging at splash screen.

> The following is the log

Is that the log from the virtual machine where the IDE is running normally?

cike-567, 2022-02-01:
Scenario B is correct。

This log is the log when the IDE is running normally in the virtual machine

per1234, 2022-02-15:
Hi @MArimont3. Please post the full output you got from attempting to start the IDE from the command line.

ednieuw, 2022-03-07:
Same issue here with RC3 and now also with RC4.
I use W11.

tahunus, 2022-03-22:
Same issue with RC5.  Running on Windows 11 OS Build 19043.1586 (64 bit).

JachinY, 2022-04-29:
I meet the same problem, how to deal it

tahunus, 2022-05-26:
RC6 running on Windows 10 Enterprise Version 21H1 64 bit (OS Build 19043.1706) w/Intel Core i5-10310 @ 1.70Ghz & 16GB RAM takes on average (tried several times) 16.5 minutes to start
[image]

ednieuw, 2022-05-27:
RC3 was installed as "Only for me" instead "For all users".

When RC5 was installed "For all users" RC3 was not properly uninstalled.
I removed RC5 and restarted the PC
I re-installed RC3 "Only for me" and uninstalled it again and restarted the PC
Then I removed the folder c:\Users\ednie.arduinoIDE\
And Installed RC5 again.
Now it takes 10 seconds to see the editor

I also saw this solution:
I set my arduino.exe to run as Administrator and it is now fast when i open it
at [https://forum.arduino.cc/t/very-slow-startup-ide-2/953115/9](url)

Addharuqutni, 2023-05-30:
PS C:\Program Files\Arduino IDE> & ".\Arduino IDE"
PS C:\Program Files\Arduino IDE>
Arduino IDE 2.1.0
Starting backend process. PID: 8300
Using browser-only version of superagent in non-browser environment
Configuration directory URI: 'file:///c%3A/Users/msi%20gaming/.theia'
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2023-05-30T16:21:48.593Z root INFO Backend Object.initialize: 3.4 ms [Finished 1.089 s after backend start]
2023-05-30T16:21:48.593Z root INFO Backend Object.configure: 2.8 ms [Finished 1.089 s after backend start]
2023-05-30T16:21:48.593Z root INFO Backend MessagingContribution.onStart: 1.3 ms [Finished 1.089 s after backend start]
2023-05-30T16:21:48.594Z root INFO Backend OSBackendApplicationContribution.configure: 0.3 ms [Finished 1.119 s after backend start]
2023-05-30T16:21:48.595Z root INFO Backend DefaultWorkspaceServer.onStart: 0.3 ms [Finished 1.119 s after backend start]
2023-05-30T16:21:48.595Z root INFO Backend PluginLocalizationBackendContribution.configure: 0.1 ms [Finished 1.119 s after backend start]
2023-05-30T16:21:48.596Z root INFO Backend PluginLocalizationBackendContribution.initialize: 30.0 ms [Finished 1.119 s after backend start]
2023-05-30T16:21:48.596Z root INFO Backend TaskBackendApplicationContribution.onStart: 0.2 ms [Finished 1.119 s after backend start]
2023-05-30T16:21:48.597Z root INFO Backend ElectronTokenBackendContribution.configure: 0.8 ms [Finished 1.120 s after backend start]
2023-05-30T16:21:48.597Z config INFO >>> Initializing CLI configuration...
2023-05-30T16:21:48.597Z root INFO Backend PluginDeployerContribution.initialize: 0.4 ms [Finished 1.121 s after backend start]
2023-05-30T16:21:48.598Z root INFO Backend ConfigServiceImpl.onStart: 0.7 ms [Finished 1.121 s after backend start]
2023-05-30T16:21:48.598Z config INFO Loading CLI configuration from c:\Users\msi gaming\.arduinoIDE\arduino-cli.yaml...
2023-05-30T16:21:48.599Z root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.3 ms [Finished 1.122 s after backend start]
2023-05-30T16:21:48.599Z root INFO Backend ArduinoDaemonImpl.onStart: 0.9 ms [Finished 1.123 s after backend start]
2023-05-30T16:21:48.599Z root INFO Backend NodeFileUploadService.configure: 1.6 ms [Finished 1.123 s after backend start]
2023-05-30T16:21:48.599Z daemon INFO Starting daemon from C:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
2023-05-30T16:21:48.599Z discovery-log INFO start
2023-05-30T16:21:48.599Z discovery-log INFO start new deferred
2023-05-30T16:21:48.599Z root INFO Backend HostedPluginLocalizationService.initialize: 0.3 ms [Finished 1.123 s after backend start]
2023-05-30T16:21:48.599Z root INFO Backend BoardDiscovery.onStart: 0.7 ms [Finished 1.124 s after backend start]
2023-05-30T16:21:48.599Z root INFO Backend FileDownloadEndpoint.configure: 0.6 ms [Finished 1.124 s after backend start]
2023-05-30T16:21:48.599Z root INFO Backend AuthenticationServiceImpl.onStart: 0.3 ms [Finished 1.125 s after backend start]
2023-05-30T16:21:48.599Z root INFO Backend PluginApiContribution.configure: 0.4 ms [Finished 1.125 s after backend start]
2023-05-30T16:21:48.600Z root INFO Backend HostedPluginReader.configure: 0.3 ms [Finished 1.126 s after backend start]
2023-05-30T16:21:48.600Z root INFO Backend HostedPluginReader.configure: 0.0 ms [Finished 1.126 s after backend start]
2023-05-30T16:21:48.600Z root INFO Backend PlotterBackendContribution.configure: 0.7 ms [Finished 1.127 s after backend start]
2023-05-30T16:21:48.604Z root INFO Theia app listening on http://localhost:49688.
2023-05-30T16:21:48.604Z root INFO Finished starting backend application: 4.9 ms [Finished 1.130 s after backend start]
2023-05-30T16:21:48.612Z root WARN The local plugin referenced by local-dir:/c%3A/Users/msi%20gaming/.arduinoIDE/plugins does not exist.
2023-05-30T16:21:48.613Z root WARN The local plugin referenced by local-dir:/c%3A/Users/msi%20gaming/.arduinoIDE/extensions does not exist.
2023-05-30T16:21:48.613Z root WARN The local plugin referenced by local-dir:C:\Users\msi gaming\.arduinoIDE\plugins does not exist.
2023-05-30T16:21:48.628Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-05-30T16:21:48.628Z root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@1.5.1" with engines:
2023-05-30T16:21:48.628Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-05-30T16:21:48.629Z root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.8" with engines:
2023-05-30T16:21:48.629Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-05-30T16:21:48.629Z root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
2023-05-30T16:21:48.629Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-05-30T16:21:48.630Z root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
2023-05-30T16:21:48.630Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-05-30T16:21:48.630Z root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
2023-05-30T16:21:48.630Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-05-30T16:21:48.631Z root INFO Resolved "vscode-language-pack-bg" to a VS Code extension "vscode-language-pack-bg@1.48.3" with engines:
[... cut ...] n": 41.5 ms [Finished 2.000 s after backend start]
2023-05-30T16:21:49.474Z root INFO Deploy plugins list: 69.3 ms [Finished 2.000 s after backend start]
Opening channel for service path '/services/electron-window-ext'.
Opening channel for service path '/services/electron-window'.
Opening channel for service path '/services/ide-updater'.
2023-05-30T16:21:50.291Z root WARN Frontend a.configure took longer than the expected maximum 100 milliseconds: 196.3 ms [Finished 1.616 s after frontend start]