arduino/arduino-ide #788: IDE v2 SO SLOW and other problems

MikeyMoMo, 2022-01-29:
I just put on the IDE 2 rc 3 and it is SO SLOW!  It takes 10 minutes to compile a small program.  FAR longer than 1.18.  

But even worse is the large blue-green logo that comes up while it is starting up (currently 90 seconds) that is ON TOP and won't go to the back.  That is just, plain WRONG!!!  Please take off the On Top setting.  I am typing most of this in the blind because it won't go away!

Waiting... waiting... that logo is REALLY irritating hiding everything else and won't go away.

How does one turn off the Output window?  It takes up about 20% of the programming area and I really want back.  I can't find any way to close it.

So far, v2 is a total loser.  Slow and far more irritating and harder to use than 1.18!

per1234, 2022-01-29:
Hi @MikeyMoMo. Thanks for your feedback.

In the future, please open a separate issue for each individual topic. Making multiple unrelated reports in a single issue makes tracking them very difficult and the related discussion confusing and inefficient, meaning the reports are less likely to be resolved.

>it is SO SLOW! It takes 10 minutes to compile a small program.

Does the problem occur on every compilation, or only sometimes?

Do you have  verbose output enabled in **File > Preferences**? There is a bug that makes the output printing very slow (https://github.com/arduino/arduino-ide/issues/68, https://github.com/arduino/arduino-ide/issues/405). When there is a lot of output, this can give the impression that the compilation/upload is very slow, when really the process finished long before the output finishes trickling in through the bottleneck.

Other than that, a common cause of slow compilations is "on access scanning" by your antivirus software. The IDE starts many processes during the compilation and if the antivirus wants to do a scan on each one then it really slows things down. You cah check this by **TEMPORARILY** disabling your antivirus for a single compilation to see if the problem goes away, then turn the antivirus back on.

If the problem doesn't occur with the antivirus off you will need to adjust the settings of your antivirus to put the appropriate file, folder, or process on the "allow list" so it doesn't interfere with compilation.

Please be cautious about working without an antivirus. This is only about temporarily disabling it for a quick test. If you don't feel comfortable doing that, fine. You can try going straight to configuring the antivirus to not interfere with the Arduino software.

>even worse is the large blue-green logo that comes up while it is starting up (currently 90 seconds) that is ON TOP and won't go to the back. That is just, plain WRONG!!! Please take off the On Top setting.

This is already being tracked here: https://github.com/arduino/arduino-ide/issues/324 and we have a proposed improvement here: https://github.com/arduino/arduino-ide/pull/717

>How does one turn off the Output window? It takes up about 20% of the programming area and I really want back. I can't find any way to close it.

It is the icon on the status bar here:

[image]

Related: https://github.com/arduino/arduino-ide/issues/600 https://github.com/arduino/arduino-ide/issues/308

felixd, 2022-01-30:
I have confirm really slow working of latest night build on Windows 10 (STM32Duino https://github.com/stm32duino/Arduino_Core_STM32).

It is really so slow that I still keep v1 just for compiling, monitoring etc. I just keep IDEv2 for code editing (colors, libraries support etc). 

It for sure not Antivirus thing (I don't use any).

This short message is just ot point this problem. Once I will have more detailed information about this problem I will try to update this ticket as well.

Cheers, Paweł

felixd, 2022-01-30:
It's "indexing" things forever... it's already running for more than an hour ;) I am not quite sure what's being indexed.

[image]

Now, when this message is gones it's "Building sketch" (that takes some time even in IDEv1 but I haven't initiated that (Sket building).

[image]

... and now it's "Indexing" again ;) 

[image]

per1234, 2022-01-30:
I will request that we restrict this issue exclusively to the discussion of the one original problem reported by @MikeyMoMo:

> It takes 10 minutes to compile a small program.

Slowness of the indexing/building process could be another symptom of a slow compilation, since it is a similar operation. But it is not exactly the same because this indexing/building is the operation of [the Arduino Language Server](https://github.com/arduino/arduino-language-server), which only provides the code-aware features such as parameter hints, problem detection, hover, goto, etc., but is not involved in the sketch compilation process at all.

There is another report of issues related to this indexing/building process here:
https://github.com/arduino/arduino-ide/issues/714
and another here: https://github.com/arduino/arduino-ide/issues/431 (which should be fixed when using the nightly, so probably not the same)

MikeyMoMo, 2022-01-31:
Sounds right.  Again, sorry for the poor posting etiquette.

per1234, 2022-02-01:
No worries. When you get a chance, please answer the questions I asked in my previous reply:

>Does the problem occur on every compilation, or only sometimes?

> Do you have verbose output enabled in File > Preferences? There is a bug that makes the output printing very slow (#68, #405). When there is a lot of output, this can give the impression that the [...]

per1234, 2022-03-11:
Closing due to lack of response.

@MikeyMoMo if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

closed 2022-03-11 by per1234

MikeyMoMo, 2022-06-12:
V2 has so many problems I stopped using it.  Don't have anything further since it is so badly messed up.  Maybe I will try again in a few months.  For now, a non-starter!  Deleted from my system.

VladimirVecera, 2023-06-10:
I have Arduino IDE 2.1.0 and compiling the code is terrible, but really terrible. It takes a few minutes and if I have to press the ESP32 boot, then if I jump back because I can't wait for maybe 5 minutes and I forget the boot button, I have to compile the code again. And if I test the code .... it just doesn't work. I have Windows 10.