arduino/arduino-ide #800: Windows 10: Improper Port name shown in GUI on Teensy devices.

KurtE, 2022-02-02:
**Describe the bug**
After installing the first version of the Teensyduino Board Manager version for testing the Arduino IDE and Arduino IDE 2,  the GUI is showing the address field of the device and not the label. 

Right now the GUI is showing things like: Teensy MicroMod on usb:0/140000/0/1/4

As @PaulStoffregen mentioned in the PJRC forum post:
https://forum.pjrc.com/threads/53548-Arduino-CLI-Alpha-Release-Teensy-Support?p=299387&viewfull=1#post299387

According to the [Pluggable Discovery Specification](https://arduino.github.io/arduino-cli/0.20/pluggable-discovery-specification/#list-command)

It appears like the GUI is showing the address and not the label.

**To Reproduce**
Steps to reproduce the behavior:
1. I installed the nightly builds of the IDE yesterday
2. Install the first board manger version of Teensyduino, instructions in: 
3. https://forum.pjrc.com/threads/53548-Arduino-CLI-Alpha-Release-Teensy-Support?p=298914&viewfull=1#post298914
4. Plug in a teensy example Micromod Teensy
5. Look at the names shown for the Serial ports

**Expected behavior**
I am expecting that instead of: Teensy MicroMod on usb:0/140000/0/1/4
It would show something like: Teensy MicroMod on COM118

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 10

**Additional context**
Not sure if this may be related to the issue: https://github.com/arduino/arduino-ide/issues/567

And also may be partially related to the Pluggable Monitor issue: https://github.com/arduino/arduino-ide/issues/769

KurtE, 2022-02-02:
Note this is not specific to Windows.  I just verified again on my Ubuntu 20.04 machine with last nights ide build
[image]

And from his the discovery tool:
```
kurte@kurte-XPS-8300:~$ .arduino15/packages/teensy/tools/teensy-discovery/1.56.1/teensy-discovery
LIST
{
"eventType": "list",
"ports": [
{
"address": "usb1/1-1/1-1.4/1-1.4.1",
"label": "/dev/ttyACM0 Everything",
"protocol": "teensy",
"protocolLabel": "Teensy Ports",
"properties": {
"modelID": "0x26",
"name": "Teensy MicroMod"
}
}
]
}
```

PaulStoffregen, 2022-02-03:
Here is the specific piece of the Pluggable Discovery spec relevant to this issue.

[image]

KurtE, 2022-02-04:
I thought I would mention, that I noticed one location in the GUI that it does show the label:

[image]

But it is interesting it shows it when you drop down the list, but it does not show it in the top selected one instead it shows the address