arduino/arduino-ide #802: Sort Board menus like the classic IDE does

per1234, 2022-02-02:
## Describe the current behavior

Arduino IDE 1.x sorts the menu for each platform in **Tools > Board** (e.g., **Tools > Board > Arduino AVR Boards**) according to the order of occurrence of the `name` property for each board definitions in the platform's [`boards.txt` configuration file](https://arduino.github.io/arduino-cli/dev/platform-specification/#boardstxt). This allows the platform developer to sort the boards in some logical order in the menu (e.g., putting commonly used boards at the top of the menu, or grouping related boards together).

Arduino IDE 2.x sorts them by alphabetical order, which gives the platform author no control. For example, in the Arduino IDE 2.x menu for the **Arduino AVR Boards** platform menu, the first 8 boards are 3rd party or retired.

Arduino IDE 1.x:

[image]

Arduino IDE 2.x:

[image]

## To reproduce

1. Select **Tools > Board > Arduino AVR Boards** from the Arduino IDE menus.
1. Note the order of the boards in the menu.
1. Compare this to the order of occurrence of the `name` properties for each board definition in the platform's `boards.txt` file:
   https://github.com/arduino/ArduinoCore-avr/blob/1.8.4/boards.txt

🙁 the Arduino IDE menu sort order does not match the order in `boards.txt`

## Describe the request

**Tools > Board** menu for each platform is sorted according to the order in `boards.txt`.

## Desktop

- OS: Windows 10
- Version: 2.0.0-rc3-snapshot.759b410
Date: 2022-01-31T13:15:29.574Z
CLI Version: 0.20.2 [13783819]

## Additional context

### Related:

- https://github.com/arduino/arduino-ide/pull/486
- https://github.com/arduino/arduino-ide/issues/801

KurtE, 2022-02-02:
Note, I do wish at times there were some extensions to the ordering of the boards in the boards manager, for example on my Windows Machine, my menu looks something like:
[image]

Now I probably do about 90% of my stuff using Teensyduino, so for example I wish there were quick ways for me to choose these boards. 

Now the new drop down list with The Teensy Micromod on COM18 that would great that you can choose them directly. 

But as an extension to the current stuff,  it might be fun if the tools->boards had an optional show the last N boards selected or some such mechanism.

Another low priority possible extension of ability to allow the user to order the menu...

per1234, 2022-02-03:
Hi @KurtE. There is a previous request for that here: https://github.com/arduino/arduino-ide/issues/330 and a related one here: https://github.com/arduino/arduino-ide/issues/86

AlbyIanna, 2022-08-11:
Blocked by: https://github.com/arduino/arduino-cli/issues/1834

PR #1520 (2022-10-17): Fix boards listing
### Motivation
The IDE needs to sort the menu for each platform in **Tools > Board** (e.g., **Tools > Board > Arduino AVR Boards**) according to the order of occurrence of the `name` property for each board definitions in the platform's [`boards.txt` configuration file](https://arduino.github.io/arduino-cli/dev/platform-specification/#boardstxt).

### Change description
- use `listall` instead of `search` command in the arduino-cli
- add a proper order to the menu item when registering a board
- update the arduino-cli to use the sorting fix

### Other information
Fixes #802. [... cut]
closed 2022-10-17 by AlbyIanna