arduino/arduino-ide #803: Serial Plotter (IDE 2.0) issue

kubichekster, 2022-02-02:
**Is your feature request related to a problem? Please describe.**

The new Serial Plotter displays only 50 data values, compared to the previous version, which displayed 500 values.
This is a problem because events stream by 10 times faster and are much harder to see. With 500 points we see a much clearer time history of the physical measurements we're making. I use the Plotter a lot in my K-12 STEM workshops, and this will really have a negative impact.

**Describe the solution you'd like**
Change the number of plot values back to 500. Or make it a user selectable option

**Describe alternatives you've considered**
I can modify my old programs to slow Serial.print outputs so the data stream doesn't disappear so quickly. It kinda works but it's suboptimal.
 
**Additional context**
Add any other context or screenshots about the feature request here.

ratio-x, 2022-03-07:
urgently expected...

IanBUK, 2022-04-18:
Export to Excel would be nice as well.

fralbo, 2022-07-21:
In fact, the best would be to let us define the horizontal scale.

InstantMuffin, 2022-08-10:
I, too, stumbled upon this.

Obviously it would be nice to have specific features like mentioned above, but I think just to have on par featureset with the former IDE it would be wise to readd display of 500 data points like in the earlier versions. It would be quite odd to have a successor that does not have or has less of a specific feature than the predecessor.

I tried changing this line
https://github.com/arduino/arduino-serial-plotter-webapp/blob/04a460e0c21609a63c4064f561990c0128c5a0ed/src/ChartPlotter.tsx#L46
in the bundled package that is being fetched/created when building arduino-ide. The build process did not edit the file back to what it was before, however, unfortunately, it did not increase the number of data points.
Being unfamiliar with the build process, I don't know if this is in fact the wrong line of code to adapt, or if the change just doesn't make it into the final arduino-ide build for some reason.

The change in itself would seem to be rather trivial, given that and the featureset of the predecessor I would really like to see the this implemented before the final release.

SaturdayScience, 2022-09-15:
I'm running rc9.4 on Linux Mint 21 and the plotter frustrates me. With data once per second plotter shows only 5 X-axis data sets no matter how large I make the window. At fast rate such as 100 data sets / second it shows 50. I can't seem to display 500 seat.

Farhankhosravi, 2022-09-16:
Hi..
yup, this is so URGENT and needed to make plotter scalable... 50 is really small!
only make two sliders to change x/y zoom manually! not a hard work, but VERY life savior.

I'm waiting for this too, hope they fix this ASAP.

SvenRosvall, 2022-09-23:
Another vote for this enhancement.

fralbo, 2022-09-24:
Anyone in the development team to take care about our requests

rhettherman, 2022-09-29:
Either going back to the 500-point serial plotter, or (better yet) allowing for perhaps several options for the number of points to plot, would be most appreciated.

pointsplitter, 2022-09-29:
Either going back to the 500-point serial plotter, or (better yet) allowing for perhaps several options for the number of points to plot, would be most appreciated.

ArnieO, 2022-09-29:
Yes, please do the requested enhancement / fix!
I fortunately have IDE 1.8.19 installed in parallel - it still displays 500 points.

bruno-zarokian, 2022-10-09:
To add to that, my plotter would sometimes display only 5 samples! Some random series of action usually gets it back to 50 samples. Seems like jumping between multiple projects does/undoes the issue but haven't explored well enough. All my projects use serial over USB btw.

per1234, 2022-10-09:
>my plotter would sometimes display only 5 samples! Some random series of action usually gets it back to 50 samples. Seems like jumping between multiple projects does/undoes the issue but haven't explored well enough.

Thanks for your report @bruno-zarokian. If you are able to find a procedure that can be followed to reproduce this bug, please submit a dedicated issue for it.

InstantMuffin, 2022-10-10:
I expected a few comments in regards to parity to the 1.X feature state, but I didn't expect so many.

I, too, still want this to be an par or to allow for more samples than 500.

per1234, 2022-10-10:
Hi all. We really appreciate the feedback about how the Arduino IDE 2.x Serial Plotter can be improved. However, once the message has been received, additional comments that don't add any new information can add up to a lot of notifications for those of us who watch this repository to keep up with the development work.

If you only want to express support then I suggest the use of [GitHub's "Reactions" feature](https://github.blog/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/).

Discussion is always welcome over on the Arduino Forum:

https://forum.arduino.cc/

Thanks in advance for your cooperation.
Per

Alex-Tsyganok, 2022-12-05:
The serial plotter is a web application that is installed as part of Arduino IDE since it's running as a web application, bundled source code can be modified on a local machine.

The package can be found in the installation folder of Arduino IDE 
`\Arduino IDE\resources\app\node_modules\arduino-serial-plotter-webapp\build\static\js\`
`main.35ae02cb.chunk.js` can be modified to change values threshold value, need to replace `U=Object(o.useState)(50)` with `U=Object(o.useState)(5000)`  it will increase maximum data points to 5000. 

Tested with Arduino IDE 2.0.2 version.
[image]

fralbo, 2022-12-05:
🤪
Thanks Alex.
Better layer than never...

kubichekster, 2022-12-06:
> ◆ This message was sent from a non-UWYO address. Please exercise caution when clicking links or opening attachments from external sources.

Thanks Alex - nice solution! I will give it a try.
However, I hope the developers will build this into the IDE asap. I work with K-12 teachers across the state and the fix needs to be plug-and-play before they can use it at their schools. They also need a portable version, but that's another story.
Thanks again

[image]<https://user-images.githubusercontent.com/5132633/205741021-223c3130-0a71-4264-9ac1-38eb364deb82.png>

falk-stefan, 2022-12-29:
@Alex-Tsyganok Thank you for that quick fix!

But I have to tell you that I felt the strong urge to wash my hands after that 😂

zapta, 2023-01-14:
I tried to modified the js file but not luck here, still only 50 points.  Windows 10, Arduino IDE 2.0.3, notepad++ editor.  I also restarted Windows, in case there is some caching involved. Any suggestions?

[image]

Leomirdroid, 2023-01-26:
Worked for me as well. To save the changes, I had to open notepad in admin mode, change the value and save (without the txt ending by clicking on all files next to the save button). @zapta did it work for you now? Did you open the correct Arduino version? Did you try uninstalling all older versions and make a clear reinstall?

Ralf-W, 2023-02-02:
@Alex-Tsyganok Thank you a lot!

RolfHut, 2023-02-10:
I'll briefly add that while the fix by @Alex-Tsyganok works on my Mac, this is not a solution for teachers (like me) who rely on the IDE in their teaching. Students can't be expected to dig this deep into the code of an application. I have to resort to either advising students to fall back on the old IDE, or use third party things (like processing), but neither of this is good teaching.

Can the fix by @Alex-Tsyganok just be implemented for now, while people work on a more lasting solution like allowing users to change the x-axis?

Dlloydev, 2023-05-05:
As it stands now, the plot width auto scales to full width as the data fills to 50 samples, then the plotter becomes a sliding window.

[image]

It would be useful to have the plot width auto scale to full width as the data fills to 500 samples (or a user set limit), then the sliding window activates.

Farhankhosravi, 2023-05-06:
Hello again...
Came to ask that how long this feature will be take?! Just asking for more samples like IDE 1.8 and older! It's really necessary but after a long while, it doesn't added to IDE V2...

Still waiting for such small feature. Please take a look!

zapta, 2023-05-06:
@per1234, @AlbyIanna, any news on this issue?  It was assigned a year ago but so far there is no indication here of any progress.  Can you please look into it?

Pkanna10, 2023-05-18:
> The serial plotter is a web application that is installed as part of Arduino IDE since it's running as a web application, bundled source code can be modified on a local machine.
>
> The package can be [...]

I just tested, and the solution works!! Thanks a lot. Any idea on how to make it go slower?

kubichekster, 2023-05-18:
OK, thanks, that sounds useful!
But I have a newbie question: What do I do after I make the change to the .js file? Do I need to recompile or rebuild something in order to update the IDE? Any details would be appreciated.
As a general comment, since the fix seems fairly trivial it seems crazy they don't default the number of data points to the version 1.x value (i.e., 500) so all the legacy Arduino programs would still work as designed.
thanks again

per1234, 2023-05-18:
Hi @kubichekster. This issue is used to track the request for the Arduino developers to increase the number of displayed data points or to make it user configurable. Comments that add unique information that is relevant to that development work are welcome, but this is not an appropriate place to get assistance with [the hack that was shared](https://github.com/arduino/arduino-ide/issues/803#issuecomment-1338149431).

If anyone needs assistance or has questions about that hack, please post on Arduino Forum:

https://forum.arduino.cc/

I'm sure we'll be able to help you out over there.

samstafford, 2023-07-25:
@Alex-Tsyganok  thank you for this patch; was very helpful to me.

per1234, 2023-07-29:
A PR for adding a menu to control the number of data points has been submitted to the repository where the Serial Plotter codebase is hosted:

https://github.com/arduino/arduino-serial-plotter-webapp/pull/32

Please refrain from commenting on that PR thread unless you have some significant and unique information to add. If you only want to express your support for the PR or a comment in the thread, use the ["Reactions" feature](https://github.blog/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to add a 👍, etc. If the thread becomes a source of notification noise due to the addition of comments that don't add significant and unique information, as happened here, I will be forced to lock the conversation, which might prevent the addition of comments that are actually of value.