arduino/arduino-ide #806: Serial Monitor - Autoscroll

KurtE, 2022-02-03:
**Describe the bug**
Sorry Not sure if bug or feature request

I am having some fun trying out the new IDE.  I hope reports like this are useful. 

But with the current daily build (as of yesterdays), if I am running a sketch that outputs lots of data to the Serial Monitor.

And so far  I have found an obvious way to tell the the monitor window to not scroll, so hard to check out what was displayed earlier on.    That is for example if I do something like: ctrl+home to get to the start of the window, it will jump back to the bottom as soon as new data is received.

**To Reproduce**
Steps to reproduce the behavior:

Have a sketch that outputs a lot.  I was trying to help out user on PJRC forum, so slightly modified sketch:
```
#include <MemoryHexDump.h>

//2D array to hold display feedback
byte channelStripData[8][9]; //byte array to store incoming midi display data

byte i;

unsigned long checkTime;
unsigned long startTime;
unsigned long timeOut;

void setup() {
  while(!Serial && millis() < 5000) ;
  Serial.begin(9600);
  MemoryHexDump(Serial, (void*)channelStripData, sizeof(channelStripData), true);
  checkTime = millis();
  timeOut = millis();
  memset((void*)channelStripData, '1', sizeof(channelStripData));
}

void loop() {

  timeOut = millis(); //Disables timeout

  if (millis() - checkTime > 1000)// Set frequency of table updates
  {
    printStuff();
    checkTime = millis();
  }

} // End of loop

void printStuff()
{
  Serial.print("CHAN");
  Serial.print("\t");
  Serial.print("SCRIB");

  Serial.println();

  for (i = 0; i < 8; i++)
  {
    Serial.print("CH ");
    Serial.print(i + 1);
    Serial.print("\t");

    Serial.write(channelStripData[i][0]); 
    Serial.write(channelStripData[i][1]);
    Serial.write(channelStripData[i][2]);
    Serial.write(channelStripData[i][3]);
    Serial.println();
  }
}
```
Again nothing special here.  But wanted to verify the contents of the call to my MemoryHexDump, and tried to scroll to the top and the screen kept moving back to end.

**Expected behavior**
A clear and concise description of what you expected to happen.

Expected something like a checkbox like you have in 1.8.19:

**Screenshots**

[image]

Alternative, do like what TyCommander does: 
If you scroll off the bottom, it will stop doing autoscroll, until you manually scroll it back to the bottom.

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 
 - 
[image]

**Additional context**
Add any other context about the problem here.

per1234, 2022-02-03:
Hi @KurtE.

>I have found an obvious way to tell the the monitor window to not scroll

The Serial Monitor autoscroll behavior is toggled by this icon:

[image]

It is not so clearly indicated as the Arduino IDE 1.x autoscroll control.

>If you scroll off the bottom, it will stop doing autoscroll, until you manually scroll it back to the bottom.

I have used a couple of other applications that have that behavior, and I like it.

>I hope reports like this are useful.

Bug reports and feature requests are always welcome. Thanks!

KurtE, 2022-02-03:
Thanks, somehow I missed it... probably good to close

closed 2022-02-03 by KurtE