arduino/arduino-ide #807: Allow 3rd parts tools in Tools menu

PaulStoffregen, 2022-02-03:
Earlier versions of Arduino IDE allowed third party tools to be added to the Tools menu.

For example, Teensy uses this feature to provide a tool that manages keys used for code security.

[image]

IDE 2.0 has similar tools in its Tools menu.  The same "Wifi101 / WifiNINA Firmware Updater" tool is present.

[image]

Hopefully an API can be provided for 3rd party tools to be added.  Obviously it can't be Java.  Existing 3rd party tools will need to be rewritten.  That's ok.  All that's needed is a mechanism and API to do so.

per1234, 2022-02-04:
Thanks for taking the time to submit an issue @PaulStoffregen .

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/58.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-02-04 by per1234

PaulStoffregen, 2022-02-04:
Opps, I did a few searches but somehow missed #58