arduino/arduino-ide #810: Serial Monitor with Teensy T4.x (probably others) does not receive input

KurtE, 2022-02-04:
**Describe the bug**
I wrote a simple sketch to output so many lines to the terminal monitor as fast as possible.  I am curious to see if the new monitor can keep up with a T4.x which runs at full speed USB...

The sketch is nothing special, but loops waiting for input data to allow me to type in how many lines to output. 
I then start up the sketch and do what is necessary to get it back to comm port (more later), and then it shows up my first prompt, I type in lets say 100 in what I am pretty sure is the input window for it and hit enter and the sketch does not receive any data and as such does not run any farther. 

@PaulStoffregen - not sure if any of this touches the Teensy integration code or not.  It should be using the default Serial monitor?

Again secondary issue: I believe that with earlier versions of Arduino like 1.8.19, if you chose to use the default Arduino monitor code, versus the Teensy updated monitor code, that the user choice was preserved between builds. 

**To Reproduce**
Steps to reproduce the behavior:
1. Install the Teensy boards see https://forum.pjrc.com/threads/53548-Arduino-CLI-Alpha-Release-Teensy-Support?p=298914&viewfull=1#post298914 for board manager string
2. start up the sketch...  It is shown at the end
3. Build and download it.  Note: this one used the default USB type of Serial, default CPU speed...
3. connect the Serial monitor and try to type in something.

**Expected behavior**
A clear and concise description of what you expected to happen.

I expected it in this case to output 100 lines plus summary to the serial monitor

Side note: At least on windows, when I now rebuild the sketch it switches the Serial port setting and terminal monitor will not work until I reset it. 

That is I am trying the sketch: and the top shows I am on COM115
[image]

I then do a sketch upload and notice it changed what the active Serial is:
[image]

It then does not allow me to use Serial monitor:
[image]

so I go back to tools ports and switch back (Probably should have done screen shot after I selected the COM line.
[image]
And then we are back to:

[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Nightly build 20210201

**Additional context**
```
void setup() {
  // put your setup code here, to run once:
  while (!Serial && millis() < 5000);
  Serial.begin(115200);

}

void loop() {
  while (Serial.read() != -1); // remove any thing still on queue
  Serial.println("\nEnter number of lines to output");
  uint32_t count_of_lines = 0;
  for(;;) {
    int ch;
    while ((ch = Serial.read()) == -1 ) ;
    Serial.write(ch);
    if ((ch >= '0') && (ch <= '9')) count_of_lines = count_of_lines * 10 + ch - '0';
    else break;
  }
  elapsedMillis em = 0;
  for (uint32_t line_num = 0; line_num < count_of_lines; line_num++) Serial.printf("Line %u\n", line_num);
  uint32_t delta_time = em;
  Serial.printf("*** Output %u lines in %u millis\n", count_of_lines, delta_time);
}
```

KurtE, 2022-02-04:
Quick update: the not getting input appears to happen with Arduino Uno as well...

I had to modify the sketch some to remove printf and elapsedMillis...
```
void setup() {
  // put your setup code here, to run once:
  while (!Serial && millis() < 5000);
  Serial.begin(115200);

}

void loop() {
  while (Serial.read() != -1); // remove any thing still on queue
  Serial.println("\nEnter number of lines to output");
  uint32_t count_of_lines = 0;
  for(;;) {
    int ch;
    while ((ch = Serial.read()) == -1 ) ;
    Serial.write(ch);
    if ((ch >= '0') && (ch <= '9')) count_of_lines = count_of_lines * 10 + ch - '0';
    else break;
  }
  //elapsedMillis em = 0;
  uint32_t start_time = millis();
  for (uint32_t line_num = 0; line_num < count_of_lines; line_num++) {
    Serial.print("Line ");
    Serial.println(line_num);
  }
  uint32_t delta_time = millis() - start_time;
  //Serial.printf("*** Output %u lines in %u millis\n", count_of_lines, delta_time);
  Serial.print("*** Output "); 
  Serial.print(count_of_lines);
  Serial.print(" lines in ");
  Serial.print(delta_time);
  Serial.println(" millis");
}
```
[image]

PaulStoffregen, 2022-02-04:
Tried it here with Linux.  Seems to work better.

[image]

KurtE, 2022-02-04:
Another quick update.  I then closed serial monitor and reopened it a couple of times on the UNO and now it is showing:
[image]

That you are supposed to hit CTRL+enter to send message... 

I will see if that works on Teensy as well.  

Two things:
1) It was not showing in previous runs... not sure why
2) Sure wish if the keyboard cursor is in the text box and you hit enter it sends
3) (can not count) If not obvious - maybe should also have button (Like Tycommander) that has Send on it...

PaulStoffregen, 2022-02-04:
But even on Linux, I'm seeing signs of "buffer bloat" issues....

[image]

KurtE, 2022-02-04:
Thanks Paul, I verified it worked as well...  

Again to developers - unclear why I did not see the CTRL+enter on it earlier... Maybe again me having to switch back from USB, I do see it did show up after I switched back... but probably by then eyes skipped over it. 

Side note: maybe it was also some frustration from trying to type that sketch in on Ubuntu...  Maybe something is screwy with my machine, 

But: I had the IDE running
And want to create the sketch so hit ctrl+n
new window opens up,  
I move the cursor to the end of the line put your setup... and hit enter
It first moves cursor to under the comment and then magically moves it back to column one... I type a new line in:
```
while (!Serial && millis() < 5000);
```
It stays in column 1 so I hit tab key and it moves out and it moves back to Col 1, hit tab again it moves to the next line before the }...  Maybe something strange with machine,  I did reboot and still the same...  What the heck is going on.

I finished typing it, in and built the sketch,  ran it... then I clicked the close button on the window and window was closed and the program was lost without ever being prompted to save it... 

So the one I posted was the second attempt which I did on windows and it did not have the ghost moving the text cursor

per1234, 2022-02-05:
>when I now rebuild the sketch it switches the Serial port setting and terminal monitor will not work until I reset it.

I am able to reproduce this with nightly build 20210201, but not with the latest nightly, so I think it must have been fixed by https://github.com/arduino/arduino-ide/pull/792

>Sure wish if the keyboard cursor is in the text box and you hit enter it sends

This is being tracked at https://github.com/arduino/arduino-ide/issues/572

>It was not showing in previous runs... not sure why

Likely because in addition to the new trigger being unintuitive and non-standard, the message that explains it to you disappears as soon as you start typing text in the field. I pointed that poor user experience out in https://github.com/arduino/arduino-ide/issues/572

>should also have button (Like Tycommander) that has Send on it...

Tracked here: https://github.com/arduino/arduino-ide/issues/348

>I'm seeing signs of "buffer bloat" issues....

There were some previous reports of poor Serial Monitor performance at https://github.com/arduino/arduino-ide/issues/79 and https://github.com/arduino/arduino-ide/issues/519, but those were closed as fixed by https://github.com/arduino/arduino-ide/pull/524

So please open a new issue to track this bug @PaulStoffregen.

> I clicked the close button on the window and window was closed and the program was lost without ever being prompted to save it...

Tracked here: https://github.com/arduino/arduino-ide/issues/595

>magically moves it back to column one.

I think it must be this bug: https://github.com/arduino/arduino-ide/issues/437

---

This high quality feedback is very valuable, but please try to make a dedicated issue report for each distinct bug or feature request. When many separate things are reported in the same issue it makes it very difficult to effectively and efficiently track them and we risk accidentally missing one.

closed 2022-02-05 by per1234

KurtE, 2022-02-05:
Thanks, 

I was actually in the middle of summarizing the stuff like you mentioned.  Had found 572 and 595 (and maybe 677) and was going to suggest maybe close as most of it covered by other things. 

quick update, the original thing I saw when I was starting this report was after a build I saw the not connected, which I see on the 0201 build will try with new one...
[image]

I can type 100 and even 100 + ctrl+enter and nothing happens yet it is showing the output from the sketch in the window...

Actually in this case it is showing/scrolling the text in both IDE2 windows I have open.

Thanks again,

per1234, 2022-02-05:
>maybe 677

677 is another thing that can cause loss of unsaved work, but that one is related to the design of the save on exit dialog (which is presented if you have auto-save disabled in **File > Preferences** and exit with a sketch in a "dirty" state), while the problem you described was about no prompt at all on exit, indicating the sketch was not in a "dirty" state (https://github.com/arduino/arduino-ide/issues/595)

>I saw the not connected

It is probably this one: https://github.com/arduino/arduino-ide/issues/753

>100 + ctrl+enter and nothing happens yet it is showing the output from the sketch in the window...

This is https://github.com/arduino/arduino-ide/issues/752, which is another symptom of the same bug that causes https://github.com/arduino/arduino-ide/issues/753

>0205 appears to solve the first part. thanks

Yay! I was very happy to discover it fixed when I gave it a try with my Teensy 4.0.

PaulStoffregen, 2022-02-06:
> There were some previous reports of poor Serial Monitor performance at https://github.com/arduino/arduino-ide/issues/79 and https://github.com/arduino/arduino-ide/issues/519, but those were closed [...]

I'm going to wait until #769 is fixed.  Then I'll investigate by creating a pluggable monitor to serve as a reliable test case without depending on hardware specifics.