arduino/arduino-ide #812: Serial Monitor - How to select all text and do a copy to the clipboard?

KurtE, 2022-02-05:
**Describe the bug**
Lets say I have 1000 lines of debug output in the serial monitor window, how do I do a select all, such that I can copy it to the
clipboard?

With 1.8.19, I click the mouse into the monitor window and hit CTRL+a and it selects all of the text.  I can then hit CTRL+c
to copy to clipboard. 

**To Reproduce**
Steps to reproduce the behavior:

Run a sketch and try to select all of the text... 

**Expected behavior**

I expected hitting ctrl+a would select everything in that window not the source window.

**Screenshots**
[image]

And/Or alternative solution - context menu like for example TyCommander

[image]

**Desktop (please complete the following information):**
 - OS:  Windows
 - Version: Daily Build 0205

**Additional context**

@per1234 wondering on how fine a granularity you prefer for these issues?  For example this is specific to ctrl+a select all of the text, but I also suggest an alternative.  

Or would this better to to make slightly more generic with a title like: Keyboard Interface for Serial Monitor.

And for example also include:  how can I use the keyboard to scroll through the l000 lines of text.  I clicked once inside the Serial Monitor text area and hit the Page down and it looked like it scrolled one page, I hit the page down (or up) again and nothing happened.

KurtE, 2022-02-06:
Probably related to: https://github.com/arduino/arduino-ide/issues/105

per1234, 2022-02-06:
>@per1234 wondering on how fine a granularity you prefer for these issues?

In the past, I thought of the ideal level of granularity as being the same as what a developer would address in a single atomic pull request. However, I'm finding recently that level is often not turning out to be granular enough in this project.

https://github.com/arduino/arduino-ide/issues/105 is a perfect example. It is likely that replacing the current Serial Monitor implementation with a Monaco editor would resolve multiple distinct defects and add multiple desired enhancements. The issue to track the proposal of using a Monaco editor is definitely valuable. However, using that issue exclusively to track all the bug reports and feature requests is just not effective.

Affected users will be best served by reports tracking the symptoms they are experiencing or feature they are missing. Having a separate report for each distinct significant symptom or enhancement allows efficiently verifying that a solution resolved them, and continuing to track any that remain outstanding.

>For example this is specific to ctrl+a select all of the text, but I also suggest an alternative.

That is fine for a single issue.

>And for example also include: how can I use the keyboard to scroll through the l000 lines of text. I clicked once inside the Serial Monitor text area and hit the Page down and it looked like it scrolled one page, I hit the page down (or up) again and nothing happened.

I would put that in a dedicated issue. As you suspected, there is a good chance that https://github.com/arduino/arduino-ide/issues/105 would resolve both, but I still think the individual defect/enhancement reports are worth making, especially in light of the fact that https://github.com/arduino/arduino-ide/issues/105 was submitted 1.5 years ago.

KurtE, 2022-02-06:
> In the past, I thought of the ideal level of granularity as being the same as what a developer would address in a single atomic pull request. However, I'm finding recently that level is often not [...]
Thanks, 
It is always hard to know from project to project and team to team.  

At times I wish there was an issue or WIKI or some such thing (and maybe there is), that for example for the Serial monitor that would try to spell out expected behavior as well as user interface descriptions for how to do things, that then for those things that don't work would refer to... This might also include things like accessibility issues (Not sure if I described this very well).

Example topics:

If I mouse click in the Serial monitor, is there supposed to be any visual indication that the focus is now on that window Pane?  Like maybe a text cursor that looks distinct from the one in edit windows?  If so what?   Right now if I click there and type an A nothing shows up, so obviously the text focus moved to there.  

Navigation keys: again assuming the indicator, what should different keys/mouse moves do for example:
Arrow keys up, down, left, right, plus home, end, page up/down...
what about selection (ctrl or shift...)...

Or How do I by only using the keyboard navigate to working with the Serial Monitor...

Note: The above are just example things, I have not looked yet to see if anything covers them, but back on the topic
scrolling through 1000 lines mentioned above. 

I agree #105 is too generic and maybe only one possible solution where you may choose others. 

And from what you are saying: Opening one that is titled something like: keyboard navigation within Seriial Monitor, 
may also be too generic, although that would be more the level I would typically try to do as, I believe it is one main issue of not setting an maintaining a consistent input focus on the window.

But one could open maybe 10 issues like:
click + pg down only works for one pg down
click + pg up only works for one.
click + home - take me home but only works for one
click + end - takes me to end but only once.
click + down arrow - worked for a couple of presses
...

I will probably open just one as I think it is the right thing, but can always expand to many

Thanks again

Defragster, 2022-07-24:
Confirmed this failed behavior still showing in RC9, Win 10.
No right click 'context' option to select, and click or double click a word in SerMon Ouput and 'Ctrl+A' does select ALL ... but in CODE window.

The 'text entry to send "Message"' does offer context menu and 'Ctrl+A' desired behavior.

Defragster, 2022-07-27:
Interesting screen display! Serial output was a static print waiting for user input. But the MATH was such that that last line was CLIPPED and then recalculated and redrawn repeatedly.
When it was in this state on Win 10 laptop a 2 finger scroll up/down would go clear and when full bottom would repeat the blurred redraw:
[image]

sellensr, 2022-09-14:
Inability to copy/paste large amounts of data and only 50 points in the plotter means I will have to downgrade my engineering class to use v1.8.x. This is disappointing, as V2.0 brings so many other good things.

Eric-FR, 2022-10-10:
Issue still present in IDE 2.0.0/Win 10. No workaround. I also tried the left-click and scroll down but faced the #1295 issue. Would also be nice to have a search option in serial monitor. Will open an issue.

lenniea, 2022-10-17:
I agree Ctrl+A selects all text in the IDE source window not the Serial Monitor.  And when I select text with the mouse only the "visible lines" get copied to the clipboard.  Seems like the "Right Click" Popup menu solution (with Select All and Copy (to clipboard) would be best.  Additionally fixing the selection of text with the keyboard would also be helpful.

sellensr, 2022-10-19:
An alternate approach might be an option to log Serial data to a file. Copy and paste is awkward with large amounts of data.

A menu item for Tools/Serial Logging could be switched on/off at will, generating a new file at 
[sketch folder]/Logs/SerYYYYMMDDhhmmss.log whenever it is toggled on or a new compile/load takes place while toggled on.

Eric-FR, 2022-10-19:
Good idea. Open a separate issue and ask for it.

Funwar, 2023-07-25:
I can confirm this is still the case for ArduinoIDE 2.1.1 on windows 10. And therefore I am reverting to 1.8

cghiee, 2023-07-26:
Another voice that the missing "Select All" has forced me to revert to 1.8x. 
I and other professors use the select all feature in classes we teach and saving to a file is too slow when trying to quickly analyze results.

closed 2026-11-11 by giacomocusinato