arduino/arduino-ide #816: Log file location cannot be changed

samindaperamuna, 2022-02-07:
Even after the project folder is changed to the desired location, a seperate folder is created for the log files in the home directory.

To reproduce

1. Change the default sketchbook location
2. Delete the Arduino folder in the home directory
3. Restart Arduino IDE

[image]

OS : Fedora Linux
Version : 2.0.0-rc3

per1234, 2022-02-07:
Hi @samindaperamuna. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/394.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-02-07 by per1234

drws, 2022-02-16:
@per1234 This one doesn't seem to be a direct duplicate of #394. I have an additional proposition for this one which cannot go into #394. It is that log file location should not only be configurable, but the logging itself should be able to be turned off. I+m proposing a solution that this one be reopened.

I've managed to hack around it with `ln -s /dev/null ~/.arduino15/logs/application.log`. Things seem to be working, but the workaround is not well-tested yet so use with care.

per1234, 2022-02-16:
>This one doesn't seem to be a direct duplicate of https://github.com/arduino/arduino-ide/issues/394

That is true, but my impression is that this was [an "XY problem"](https://en.wikipedia.org/wiki/XY_problem). The "Y" is to make the IDE stop creating this `undefined` folder in whatever working directory you start it from. That is clearly a bug, so better to fix the bug than to add an advanced configuration option every user will need to hunt down and set.

If the bug was fixed and the IDE saved the logs to a reasonable location, would you have any desire to change that location?

>I've managed to hack around it with `ln -s /dev/null ~/.arduino15/logs/application.log`

Aha! I see you also are presenting an "XY problem". That is the logs path used by Arduino IDE 1.x. This discussion and repository is about Arduino IDE 2.x, which is a complete rewrite of the Arduino IDE. Arduino IDE 2.x has a completely different logging system and never writes anything to `~/.arduino15/logs`.

Logging [has been permanently disabled](https://github.com/arduino/Arduino/pull/11730) in Arduino IDE 1.x, starting from the 1.8.19 release. So all you need to do is update to the latest production version of the Arduino IDE and your problem will be solved. You can get the latest Arduino IDE here:

https://www.arduino.cc/en/software

>the logging itself should be able to be turned off

I think it is a reasonable request. Please open a new issue for it.

drws, 2022-02-21:
> If the bug was fixed and the IDE saved the logs to a reasonable location, would you have any desire to change that location?

It's only that I see this one as a superset of the earlier #394 and not the other way around.

> Arduino IDE 2.x has a completely different logging system and never writes anything to `~/.arduino15/logs`.

Thank you for your explanation, I see missed the intended repository.

> Logging [has been permanently disabled](https://github.com/arduino/Arduino/pull/11730) in Arduino IDE 1.x, starting from the 1.8.19 release.

Again, thank you for the info. That solves my problem.