arduino/arduino-ide #818: Find and Show next error (or warning)

KurtE, 2022-02-08:
**Is your feature request related to a problem? Please describe.**
Lets say I am building a reasonably complex sketch, which often includes multiple files.  Either within the same sketch folder and/or
using libraries or core code that I am often working on. 

And my build generates some errors or warnings.   How do I find the messages within the build window, and then how do I navigate the source files to get to each error?

I know there are some different issues already open on parts of this like:
#229 to display the error lines in red
#118 Which directly related as to maybe click on error and it positions.

**Describe the solution you'd like**
I would like some User Interface mechanism, such as button and/or menu item and/or hotkey that navigates through the build output and finds the next message (and scrolls it to make it visible) and it locates the source file and likewise makes that line visible in the editor. 

**Describe alternatives you've considered**

These days I do most of my serious development using sublime text.  For the Teensy boards on Windows, some of us have a setup (https://github.com/Defragster/Tset) that integrates with sublimetxt's build system.  We generate batch files, with settings for which Teensy, and different options, like what type of usb... Then we can do builds.  And if there are any compile errors or warnings, I can hit the F4 key to navigate through these errors, which includes both the files of the sketch as well as core or library files.   

**Additional context**
Here shows an example, that I did deliberately.  Also shows sublime project that allows me easy quick access to several libraries.
Which is great, but outside of this request.
[image]

Again not sure if this might be considered a duplicate of others like #118 if so feel free to close, I mainly opened this one as I did not see one that tied the pieces to each other.

Lorandil, 2022-02-12:
+1 for this feature!

KurtE, 2022-02-14:
Side note: I believe some of this is related to #608 

Hopefully when that one is worked on, the solution will be extended to allow  the user  to the next error.

kittaakos, 2022-06-10:
> and then how do I navigate the source files to get to each error?

Unfortunately, when the CLI compiles, it will fail with the first error. The CLI does not provide multiple errors, but the first one; hence the IDE2 team cannot do much at this point.

I will remove the `topic: code` label (CC: @per1234)

-----

However, if a board was selected and the LS is running, the error navigation is possible like this:

https://user-images.githubusercontent.com/1405703/173067596-380c7baf-3322-4c35-b9ac-09980c316e0a.mp4

Or use the _Problems_ view:

 - Open the _Command Paletter_ with <kbd>F1</kbd>,
 - See the errors and click on them if you want to navigate to the error location.

https://user-images.githubusercontent.com/1405703/173068375-de6a445a-00d1-401d-9698-f14bf203256e.mp4

I hope one of them is valuable 👆, and #608 will be avail via the nightly in a couple of days.

kittaakos, 2022-06-10:
> it will fail with the first error. The CLI does not provide multiple errors, but the first one;

No, I was wrong.

```
sketch_jun1b:6:1: error: 'voisssd' does not name a type; did you mean 'void'?
       voisssd  foo() {
 ^     ~
 void
sketch_jun1b:35:1: error: 'sssvoid' does not name a type; did you mean 'void'?
 sssvoid loop() {
 ^~~~~~~
 void
sketch_jun1b:6:7: error: 'voisssd' does not name a type; did you mean 'void'?
       voisssd  foo() {
       ^~~~~~~
       void
sketch_jun1b:35:1: error: 'sssvoid' does not name a type; did you mean 'void'?
 sssvoid loop() {
 ^~~~~~~
 void
sketch_jun1b:56:1: error: expected unqualified-id before 'this'
 this is also trash
 ^~~~
exit status 1
'voisssd' does not name a type; did you mean 'void'?
```

It's doable on the IDE2 side.

kittaakos, 2022-06-15:
I implemented it as part of https://github.com/arduino/arduino-ide/pull/1064. Please try it out, and help us with your feedback. If you have any questions about how to get the build artifacts from GitHub, please ask your questions on the PR.

closed 2022-06-21 by kittaakos