arduino/arduino-ide #819: Compile error occurs only with *.cpp file including wiring_private.h, and SAMD1 board: "EPioType has not been declared"

compulsivetinkerer, 2022-02-08:
I have a file in a large project that I just converted from .ide to .cpp.  This caused an error:
C:\Users\doug\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.12\cores\arduino/wiring_private.h:32:36: error: 'EPioType' has not been declared
int pinPeripheral( uint32_t ulPin, EPioType ulPeripheral );
                                                 ^~~~~~~~

EPioType is in the SAMD21 board directory.  This error occurs only with .ccp files that include wiring_private.h, and only for SAMD21 boards.  

I have prepared a short two file example.  You will find it compiles without error with the file named SAMD21.ino, but change it to SAMD21.cpp and it throws the error.

`//filename:  wiring_private_test.ino

//setup: Set board to   "SparkFun SAMD21 Dev Breakout"  or "Arduino Zero (Native USB Port)"

// Sketch will compile with SAMD21.ino, but change the file name to SAMD21.cpp and it will throw this error. 
//    C:\Users\doug\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.12\cores\arduino/wiring_private.h:32:36: error: 'EPioType' has not been declared
//    int pinPeripheral( uint32_t ulPin, EPioType ulPeripheral );
//                                       ^~~~~~~~

// Is compiler finding the wrong WVariant.h file?

//prototypes
void SAMD21_Setup();
void SAMD21_Loop();

void setup() {
  SAMD21_Setup();
}

void loop() {
  SAMD21_Loop();
}`

Following file should be tried as both .ino and .cpp

`//filename: Try remote.ino (no compile error) and remote.cpp (compile error in board library)

#include <wiring_private.h>

void SAMD21_Setup() {
  return;
}

void SAMD21_Loop() {
  return;
}`

per1234, 2022-02-09:
Hi @compulsivetinkerer.

>it compiles without error with the file named SAMD21.ino, but change it to SAMD21.cpp and it throws the error.

The reason for this is the special preprocessing done on `.ino` files at the start of the sketch build process:

https://arduino.github.io/arduino-cli/latest/sketch-build-process/#pre-processing

>If not already present, `#include <Arduino.h>` is added to the sketch. This header file (found in the core folder for the currently selected board) includes all the definitions needed for the standard Arduino core.

It is this `Arduino.h` file that [contains the `#include` directive for `WVariant.h`](https://github.com/arduino/ArduinoCore-samd/blob/1.8.12/cores/arduino/Arduino.h#L44), which in turn [provides the declaration of `EPioType`](https://github.com/arduino/ArduinoCore-samd/blob/1.8.12/cores/arduino/WVariant.h#L186).

The sketch preprocessing is only done on `.ino` files, so you must provide all necessary `#include` directives if you chose to use `.cpp` files for your sketch source code.

closed 2022-02-09 by per1234

compulsivetinkerer, 2022-02-09:
Ahh yes! An easy one I see.  Thanks.