arduino/arduino-ide #822: Mouse freezes while working in the Arduino IDE

cnobile2012, 2022-02-09:
After a few minutes, my mouse freezes while working within the IDE. At that point, it freezes for the whole system. This never happens when not using the IDE and it only freezes while the mouse is on the IDE window. This didn't start happening until the other day after having used the IDE for several months.

I know this sounds crazy, but it is happening. I'm an experienced software engineer of over 40 years so I know what I'm talking about.

I'm using Kubuntu 20.04.
This is the version of the IDE: arduino-ide_2.0.0-rc3_Linux_64bit

If you need any other info please let me know.

cnobile2012, 2022-02-11:
So I figured I'd try the original Arduino editor and guess what? That one causes the same issue now, so I definitely think it's a changed dependency on the Linux box I'm using

The editor I used is arduino-1.8.13.

ubidefeo, 2022-02-16:
@cnobile2012 has your distro maybe updated the rendering engine?