arduino/arduino-ide #829: Saved sketch is not opened after "Save As" operation

per1234, 2022-02-13:
## Describe the bug

The established behavior of the Arduino IDE and applications in general is that the saved file/project is opened after a "**Save As...**" operation.

🙁 Arduino IDE does not open the saved sketch after a "**Save As...**" operation. Instead, the sketch that was open in the IDE before the operation remains open.

## To Reproduce

1. Select **File > New** from the Arduino IDE menus.
   🛈: The bug is not specific to new sketches.
1. Select **File > Save As...** from the Arduino IDE menus.
1. Save the sketch with the name "**SaveAsTarget**".

🙁 The "**SaveAsTarget**" sketch  is not opened by the IDE.

## Expected behavior

Target sketch is opened in the IDE after a "**Save As...**" operation.

## Desktop

- OS: Windows 10
- Version: 2.0.0-rc3-snapshot-2f9bf86
Date: 2022-02-11T14:52:06.341Z
CLI Version: 0.21.0 [10107d24]

## Additional context

I bisected the bug to https://github.com/arduino/arduino-ide/commit/112153fb965f63d952d126c8244cd3f84f0a1a1b (it does not occur in the build from the previous commit https://github.com/arduino/arduino-ide/commit/69ac1f4779589d0d21ce3d37c180b3393ad6156c).

---

The target sketch is created.

AlbyIanna, 2022-02-28:
Another issue happens when closing an unsaved new file. I think it might be related

https://zube.io/files/arduino/df8bbd1580e1a7a32a856589dec9726e-screen-recording-2022-02-28-at-15-35-24.mov

closed 2022-03-03 by AlbyIanna

per1234, 2022-03-10:
>Another issue happens when closing an unsaved new file

Thanks @AlbyIanna. I finally got around to looking into this.

The loss of unsaved changes during a "Save As..." operation you observed is a long standing bug that has been tracked at https://github.com/arduino/arduino-ide/issues/38

The description of https://github.com/arduino/arduino-ide/issues/38 focused primarily on the unexpected sketch close, since that was the first problem the user encountered along the chain of "Save As..." misbehavior.

I believe the unexpected close part of the issue was fixed by https://github.com/arduino/arduino-ide/pull/791, but that fix was difficult to see because it was somewhat masked by https://github.com/arduino/arduino-ide/issues/829 (this issue). After https://github.com/arduino/arduino-ide/issues/829 was fixed in https://github.com/arduino/arduino-ide/pull/880, we finally get directly to the unsaved changes loss bug without the unexpected close bug to muddy the waters.

I decided that it would be most effective to track the unsaved changes loss bug in a dedicated issue and mark https://github.com/arduino/arduino-ide/issues/38 as resolved. The new issue is here: https://github.com/arduino/arduino-ide/issues/896