arduino/arduino-ide #831: Allow closing tabs from files in `src` subfolder of sketch

per1234, 2022-02-13:
## Describe the current behavior

When a sketch is opened in the Arduino IDE, all source files in the root of the sketch folder are opened in editor tabs.

All these source files are compiled into a single program. Since this fact may not be apparent to users who are accustomed to single file sketches and thus think of the `.ino` file as the sketch unit rather than the entire folder, a control for closing them is intentionally omitted from the primary UI.

Complex sketches (e.g., [Marlin](https://github.com/MarlinFirmware/Marlin)) may contain many source files. In this case, it can be convenient to expose only a subset of the files to the user. This is accomplished by putting the files under [the `src` subfolder of the sketch](https://arduino.github.io/arduino-cli/dev/sketch-specification/#src-subfolder). The contents of this folder are compiled recursively, but they are not opened in the Arduino IDE editor by default.

It is possible to open the files from the `src` subfolder via the IDE's "**Sketchbook**" view.

🙁 Once opened, there is no way to close these files again via the primary UI.

## To reproduce

1. Put this "**HasSrcFolder**" sketch to your sketchbook: [HasSrcFolder.zip](https://github.com/arduino/arduino-ide/files/8056883/HasSrcFolder.zip)
1. Open the "**HasSrcFolder**" sketch in the Arduino IDE.
1. Select **File > Preferences...** from the Arduino IDE menus.
1. Check the box next to "**☐ Show files inside Sketches**.
1. Click the <kbd>OK</kbd> button.
1. Open the **Sketchbook** view.
1. Click on **HasSrcFolder/src/InSrcFolder.cpp**
   🙁 There is no way to close the **InSrcFolder.cpp** tab via the primary UI.
1. Select **File > Quit** from the Arduino IDE menus.
1. Start the Arduino IDE.
   🙁 The **InSrcFolder.cpp** tab is still open even after a restart.

[image]

## Describe the request

Add **X** icons to all editor tabs *except* the source files in the root of the sketch folder.

## Desktop

- OS: Windows 10
- Version: 2.0.0-rc3-snapshot-2f9bf86
Date: 2022-02-11T14:52:06.341Z
CLI Version: 0.21.0 [10107d24]

## Additional context

Another primary UI path to opening these files will be provided once https://github.com/arduino/arduino-ide/issues/722 is fixed.

IanBUK, 2022-04-18:
This please. This is one of the major changes that would massively improve the ide.

francwalter, 2022-07-13:
Very annoying behavior, took me hours to find this "bug" as the reason for enduring compiling errors.
I had put a changed cpp file in the folder where my ino file is (in my svn repo) and made the big mistake to open it in Arduion IDE 2.0.8rc, never could close it and always have compiling errors. Only way is to delete the file and save it some other place 😠