arduino/arduino-ide #842: Redundant test build version indicators

per1234, 2022-02-17:
## Describe the bug

Test builds are generated for every PR creation and push as well as pushes to the `main` branch. These make it easy for anyone to contribute to the project through beta testing.

It is important that the specific test build in use be easily identifiable. The Git commit hash is used for this purpose. For example, the version assigned to this commit: https://github.com/arduino/arduino-ide/commit/9ecff86bbefe3a34549e0e546d95ee1ff3439f97 is `2.0.0-rc3-snapshot-9ecff86`

🐛 The version indicator is now added twice to the test build versions. For example, the the version assigned to this commit: https://github.com/arduino/arduino-ide/commit/9b1f15def893f1f3df58e459e66a87c3fcdb930f is `2.0.0-rc4-snapshot.9b1f15d-snapshot-9b1f15d`

## To Reproduce

1. Open the list of `push` or `pull_request` event-triggered runs of the `Arduino IDE` workflow:
   https://github.com/arduino/arduino-ide/actions/workflows/build.yml?query=event%3Apull_request+event%3Apush+is%3Asuccess
1. Click on any recent item in the list of runs.
   For example: https://github.com/arduino/arduino-ide/actions/runs/1858133816
1. Under the "**Artifacts**" section of the page, click on the download link of the artifact for your operating system.
1. Wait for the download to finish.
1. Extract or install the downloaded file.
   🐛 The extracted folder contains redundant version indicators (e.g., `arduino-ide_2.0.0-rc4-snapshot.9b1f15d-snapshot-9b1f15d_Windows_64bit`)
1. Start the Arduino IDE test build.
   🐛 The window title contains redundant version indicators (e.g., "**sketch_feb17a | Arduino IDE 2.0.0-rc4-snapshot.9b1f15d-snapshot-9b1f15d**")
1. Select **Help > About Arduino IDE** from the Arduino IDE menus.
1. Click the <kbd>Copy</kbd> button.
1. Paste the contents of the clipboard into any text editor.
   🐛 The pasted text contains redundant version indicators.
   For example:
   ```text
   Version: 2.0.0-rc4-snapshot.9b1f15d-snapshot-9b1f15d
   Date: 2022-02-17T10:40:50.003Z
   CLI Version: 0.21.0 [10107d24]

   Copyright © 2022 Arduino SA
   ```

## Expected behavior

The tester builds don't have redundant version indicators (e.g., `2.0.0-rc4-snapshot.9b1f15d`).

## Desktop

- OS: Windows 10, Ubuntu 20.04
- Version: 2.0.0-rc4-snapshot.9b1f15d-snapshot-9b1f15d
  Date: 2022-02-17T10:40:50.003Z
  CLI Version: 0.21.0 [10107d24]

## Additional context

The change was introduced at https://github.com/arduino/arduino-ide/commit/f660058c7550aaeeab81a6bc59f060d1ca2b901a

There was a previous change from the `snapshot.<hash>` (e.g., `2.0.0-rc3-snapshot.69ac1f4`) to `snapshot-<hash>` format (e.g., `2.0.0-rc3-snapshot-112153f`), which was introduced in https://github.com/arduino/arduino-ide/commit/112153fb965f63d952d126c8244cd3f84f0a1a1b

---

The additional characters use up some of [the precious allowance of path length doled out by Microsoft](https://github.com/arduino/arduino-ide/issues/106) for the beta testers on Windows.

---

Seeing this identifier in the filename redundantly, but with two different formats (note the `.` in one and `-` in the other) makes me nervous that it is an indication of something fragile.

AlbyIanna, 2022-02-17:
Looks like we messed up when merging code into the branch (most probably my fault 😮‍💨 )
https://github.com/arduino/arduino-ide/commit/f660058c7550aaeeab81a6bc59f060d1ca2b901a#diff-3e05d14f91a7dddba50598a767e65b6278c000064034da31da684affbaf1d471R81-R95

@per1234 thank you so much for reporting this! Fixing it right away!

per1234, 2022-03-07:
Fixed by https://github.com/arduino/arduino-ide/pull/846
Thanks Alberto!

closed 2022-03-07 by per1234