arduino/arduino-ide #843: The built-in IDE updater does not work with my dev build

kittaakos, 2022-02-17:
**Describe the bug**

I did not notice any IDE malfunction but  I see `Error: Cannot find channel "stable-mac.yml" update info` errors in the backend logs. It could be an `electron-update` issue (https://github.com/arduino/arduino-ide/pull/797) or an infra problem and the updater cannot access the S3 bucket.

<details>
  <summary>Click for the full stacktrace</summary>
  
```
Error: Error: Cannot find channel "stable-mac.yml" update info: HttpError: 404 
"method: GET url: https://downloads.arduino.cc/arduino-ide/stable-mac.yml?noCache=1fs4p9v6r\n\nPlease double check that your authentication token is correct. Due to security reasons, actual status maybe not reported, but 404.\n"
Headers: {
  "date": "Thu, 17 Feb 2022 21:27:14 GMT",
  "content-type": "text/html",
  "cf-ray": "6df219357bf47276-HAM",
  "etag": "W/\"2e388576eefd92a15967868d5f566f29\"",
  "last-modified": "Fri, 12 Apr 2019 12:31:08 GMT",
  "vary": "Accept-Encoding",
  "via": "1.1 dcfe318aa268be733627ea7884246be4.cloudfront.net (CloudFront)",
  "cf-cache-status": "MISS",
  "expect-ct": "max-age=604800, report-uri=\"https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct\"",
  "x-amz-cf-id": "KL8WMTjO19jeBjIs5fgrsxpTaHNpJVy93_P7qQx2RooM00kFjT4XIg==",
  "x-amz-cf-pop": "MUC50-P1",
  "x-amz-replication-status": "COMPLETED",
  "x-amz-version-id": "VqbLlYOrmhwZcBiGDmQhZPztJf8PNquh",
  "x-cache": "Error from cloudfront",
  "set-cookie": [
    "__cf_bm=eS4hAbXDhb7yM6jeYydoqwF2WMUznAL18Uz1bLZcR2M-1645133234-0-AYTyZn0S4Zo9TobH+hIPFerjz9jxandik1dObz9LuRWoXMG2eZLDyfKa9mfWBxj4vC59CNrKFkg2ac+XWGVbWQU=; path=/; expires=Thu, 17-Feb-22 21:57:14 GMT; domain=.arduino.cc; HttpOnly; Secure; SameSite=None"
  ],
  "server": "cloudflare",
  "content-encoding": "gzip",
  "alt-svc": "h3=\":443\"; ma=86400, h3-29=\":443\"; ma=86400"
}
    at createHttpError (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/electron-updater/node_modules/builder-util-runtime/out/httpExecutor.js:14:12)
    at ElectronHttpExecutor.handleResponse (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/electron-updater/node_modules/builder-util-runtime/out/httpExecutor.js:116:20)
    at ClientRequest.<anonymous> (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/electron-updater/node_modules/builder-util-runtime/out/httpExecutor.js:82:26)
    at ClientRequest.emit (node:events:394:28)
    at SimpleURLLoaderWrapper.<anonymous> (node:electron/js2c/browser_init:105:6816)
    at SimpleURLLoaderWrapper.emit (node:events:394:28)
    at Object.newError (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/electron-updater/node_modules/builder-util-runtime/out/index.js:47:19)
    at GenericProvider.getLatestVersion (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/electron-updater/out/providers/GenericProvider.js:27:50)
    at async MacUpdater.getUpdateInfoAndProvider (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/electron-updater/out/AppUpdater.js:298:19)
    at async MacUpdater.doCheckForUpdates (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/electron-updater/out/AppUpdater.js:312:24)
    at async IDEUpdaterImpl.checkForUpdates (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/lib/node/ide-updater/ide-updater-impl.js:40:52)
    at async JsonRpcProxyFactory.onRequest (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/@theia/core/lib/common/messaging/proxy-factory.js:127:24)
Request checkForUpdates failed with error: Cannot find channel "stable-mac.yml" update info: HttpError: 404 
"method: GET url: https://downloads.arduino.cc/arduino-ide/stable-mac.yml?noCache=1fs4p9v6r\n\nPlease double check that your authentication token is correct. Due to security reasons, actual status maybe not reported, but 404.\n"
Headers: {
  "date": "Thu, 17 Feb 2022 21:27:14 GMT",
  "content-type": "text/html",
  "cf-ray": "6df219357bf47276-HAM",
  "etag": "W/\"2e388576eefd92a15967868d5f566f29\"",
  "last-modified": "Fri, 12 Apr 2019 12:31:08 GMT",
  "vary": "Accept-Encoding",
  "via": "1.1 dcfe318aa268be733627ea7884246be4.cloudfront.net (CloudFront)",
  "cf-cache-status": "MISS",
  "expect-ct": "max-age=604800, report-uri=\"https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct\"",
  "x-amz-cf-id": "KL8WMTjO19jeBjIs5fgrsxpTaHNpJVy93_P7qQx2RooM00kFjT4XIg==",
  "x-amz-cf-pop": "MUC50-P1",
  "x-amz-replication-status": "COMPLETED",
  "x-amz-version-id": "VqbLlYOrmhwZcBiGDmQhZPztJf8PNquh",
  "x-cache": "Error from cloudfront",
  "set-cookie": [
    "__cf_bm=eS4hAbXDhb7yM6jeYydoqwF2WMUznAL18Uz1bLZcR2M-1645133234-0-AYTyZn0S4Zo9TobH+hIPFerjz9jxandik1dObz9LuRWoXMG2eZLDyfKa9mfWBxj4vC59CNrKFkg2ac+XWGVbWQU=; path=/; expires=Thu, 17-Feb-22 21:57:14 GMT; domain=.arduino.cc; HttpOnly; Secure; SameSite=None"
  ],
  "server": "cloudflare",
  "content-encoding": "gzip",
  "alt-svc": "h3=\":443\"; ma=86400, h3-29=\":443\"; ma=86400"
}
    at createHttpError (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/electron-updater/node_modules/builder-util-runtime/out/httpExecutor.js:14:12)
    at ElectronHttpExecutor.handleResponse (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/electron-updater/node_modules/builder-util-runtime/out/httpExecutor.js:116:20)
[... 48 lines cut ...]

**To Reproduce**
Steps to reproduce the behavior:
1. Open the IDE from a terminal. On macOS it is `open /Applications/Arduino\ IDE.app/Contents/MacOS/Arduino\ IDE`
2. Check the error logs in the terminal

**Expected behavior**
Maybe the app could log/warn: update failed.

**Screenshots**
n/a

**Desktop (please complete the following information):**
 - OS: macOS 11.6.3 (20G415)
 - Version:
```
Version: 2.0.0-rc4-snapshot.7b5c9d7-snapshot-7b5c9d7
Date: 2022-02-17T15:10:36.531Z
CLI Version: 0.21.0 [10107d24]

Copyright © 2022 Arduino SA
```

**Additional context**
Add any other context about the problem here.

kittaakos, 2022-02-17:
Maybe related: https://github.com/arduino/arduino-ide/commit/77b430675d31610efd8619a97d99d3bffdb4b7d3

per1234, 2022-03-03:
Thanks @kittaakos. The check on the "stable" update channel was failing because the necessary "channel update info file" for that channel is only uploaded on release and there had not been one since the time the updater was added. Now that [the 2.0.0-rc4 release](https://github.com/arduino/arduino-ide/releases/tag/2.0.0-rc4) has been made, the error you reported should no longer occur.

Due to even the release versions using the pre-release syntax (e.g., `2.0.0-rc4` < `2.0.0-rc4-snapshot.7b5c9d7-snapshot-7b5c9d7`), we still won't get a genuine update offer from the "stable" channel until the next release comes out (`2.0.0-rc5` > `2.0.0-rc4`), but the "nightly" update channel (`"arduino.ide.updateChannel": "nightly"`) is already offering daily updates.

It is also now easy to produce artificial stable channel updates using this feature: https://github.com/arduino/arduino-ide/pull/865

closed 2022-03-03 by per1234