arduino/arduino-ide #852: Pressing Ctrl+R to Verify/Compile screen goes white with only tool bar and circling dots.

n4idaham, 2022-02-21:
After IDE is started and you've made a code change you have a choice of mouse clicking Compile/Verify on tool bar
or just pressing the keys Ctrl+R. Pressing the key combination causes screen to go all white except for tool bar at top
and circling dots in the middle of the screen. This persists forever. There is a [ X ] at top right of screen also, and if pressed 
the entire IDE exits.

You have to restart IDE to recover. As long as you use the mouse on the tool bar to Verify/Compile there doesn't see to be
a problem.
One more note, on some very rare occasions this does not happen.

Al

per1234, 2022-02-21:
Hi @n4idaham. Thanks for your report.

Please provide the following information, which will be required for us to investigate your report:

Which operating system are you using?

Which version of the Arduino IDE are you using? You can find it by selecting **Help > About Arduino IDE** from the Arduino IDE menus.

>After IDE is started and you've made a code change

Are these conditions essential to reproduce the bug?

Does it only happen after starting the IDE, or does it happen any time?

Is it required for you to have made a code change, or does the bug happen even if you just compile the sketch that was open when you started the IDE without any modifications?

If we are to effectively investigate and fix this issue, it is essential for us to be able to reproduce the bug. So please make sure to provide detailed instructions I can follow to reliably reproduce the bug.

n4idaham, 2022-02-21:
Wll do.

Al

n4idaham, 2022-02-21:
OS - Windows 10 Pro w/latest updates.
IDE - 2.0.0-RC4-nightly-20220220-nightly20220220

You do NOT have to do anything after starting IDE to produce this problem. Simply press Ctrl+R key combination.
This problem seems to happen at any time. throughout your entire time using the IDE.
Like I said above, there was on remote isolated case where it didn't do it, but I would say it fails 98+% of the time.

I just reproduced it several times by simply starting the IDE and doing nothing more that pressing Ctrl + R.

al

per1234, 2022-02-22:
OK, thanks for the clarification.

I am able to reproduce it. I see it was introduced in https://github.com/arduino/arduino-ide/commit/112153fb965f63d952d126c8244cd3f84f0a1a1b (does not occur in the build for https://github.com/arduino/arduino-ide/commit/69ac1f4779589d0d21ce3d37c180b3393ad6156c).

A couple of observations:

---

The default keyboard shortcut for the "**Reload Window**" command is also <kbd>Ctrl</kbd>+<kbd>R</kbd>. However, the issue still occurs even after I change the shortcut for that command. So the issue isn't simply the collision caused by sloppy default keyboard shortcut allocations.

However, it does appear to be the same mechanism, since I get the identical behavior from the IDE if I trigger the "**Reload Window**" command via the custom keyboard shortcut or the Command Palette.

---

The window title changes to "**index.html**" while the reload is in progress.

This does not happen when I run the "**Reload Window**" command while using the build at https://github.com/arduino/arduino-ide/commit/69ac1f4779589d0d21ce3d37c180b3393ad6156c

---

n4idaham, 2022-02-22:
Good news on reproducing it. One other item I forgot, but it’s really moot now that we know the cause is that Ctrl+U does not exhibit the behavior.

Thanks for the update.

 

Al

 

I am able to reproduce it. I see it was introduced in  <https://github.com/arduino/arduino-ide/commit/112153fb965f63d952d126c8244cd3f84f0a1a1b> 112153f (does not occur in the build for  <https://github.com/arduino/arduino-ide/commit/69ac1f4779589d0d21ce3d37c180b3393ad6156c> 69ac1f4).

The default keyboard shortcut for the "Reload Window" command is also Ctrl+R. However, the issue still occurs even after I change the shortcut for that command. So the issue isn't simply the collision caused by sloppy default keyboard shortcut allocations.

This does not happen when I run the "Reload Window" command while using the build at  <https://github.com/arduino/arduino-ide/commit/69ac1f4779589d0d21ce3d37c180b3393ad6156c> 69ac1f4

  _____

KurtE, 2022-03-02:
For what it is worth, I was checking to see if it had been reported.

I am seeing it on Windows 10 on the nightly build 0301...

Travelbacon, 2022-03-12:
To add information. CTR+R also reloads the page since RC 5 on my system.
Debian Buster with Gnome.

per1234, 2022-03-19:
As originally reported at https://forum.arduino.cc/t/ctr-f5-goes-to-index-html/970705, a non-configurable <kbd>F5</kbd> keyboard shortcut was also introduced by https://github.com/arduino/arduino-ide/commit/112153fb965f63d952d126c8244cd3f84f0a1a1b

Probably it has the same origin as this <kbd>Ctrl</kbd>+<kbd>R</kbd> bug.

msujew, 2022-04-04:
Related to a bug in Theia, see https://github.com/eclipse-theia/theia/pull/10704 and https://github.com/eclipse-theia/theia/pull/10869. Updating Theia should address this issue by itself.

closed 2022-05-25 by kittaakos