arduino/arduino-ide #854: Platform installation only offered if port is selected

per1234, 2022-02-22:
## Describe the bug

The "**Select Board**" dialog of the Arduino IDE lists boards from two sources:

- Installed boards platforms (same as **Tools > Board** menu)
- Boards from platforms which are not installed, but available for installation via [the primary package index](https://downloads.arduino.cc/packages/package_index.json) and any other [package indexes](https://arduino.github.io/arduino-cli/latest/package_index_json-specification/) the user has configured via the "**Additional Boards Manager URLs**" preference.

When a board from the latter source is selected, Arduino IDE offers to install the board's platform.

🐛 The installation offer is not made if a port has not been selected by the user.

Not all boards use a port, either because they are a pure HID device (e.g., [Digispark](https://github.com/digistump/DigistumpArduino)), or because they are programmed via an ISP programmer.

The user may also wish simply to compile for a board. In this use case there is no need to have the hardware attached to the computer.

For these reasons, it is not appropriate to impose an arbitrary requirement that a port be selected. The port selection is not in any way necessary for the IDE to identify the necessary platform.

## To Reproduce

1. Quit the IDE if it is running.
1. Rename or delete (:warning: cautiously) the following folder.
   (🛈 this is done as a reliable way to achieve the state of having no port selected)
   - Windows:
     ```
     C:\Users\<user name>\AppData\Roaming\arduino-ide
     ```
   - Linux:
     ```
     ~/.config/arduino-ide
     ```
   - macOS:
     ```
     ~/Library/Application Support/arduino-ide
     ```
1. Start the Arduino IDE.
1. Open the "**Board Selector**".
1. Click on "**Select other board and port...**".
1. From the "**BOARDS**" menu, select any board name that has a gray color (the color indicates the platform is not installed.
   ❗ Do not select a port from the "**PORTS**" menu.
1. Click the <kbd>OK</kbd> button.
   🐛 The IDE does not offer to install the missing platform.
1. Open the "**Board Selector**".
1. Click on "**Select other board and port...**".
1. Select any port from the "**PORTS**" menu.
   🛈 It doesn't matter what is producing the port. There is no need to have the actual board connected to your computer.
1. Click the <kbd>OK</kbd> button.

🙂 The IDE offers to install the missing platform:

[image]

## Expected behavior

Offer to install the platform whenever a board of a platform that is not currently installed is selected via the "**Select Board**" dialog.

## Desktop

- OS: Windows 10
- Version: 2.0.0-rc4-snapshot-2bfc434
Date: 2022-02-22T10:25:20.375Z
CLI Version: 0.21.0 [10107d24]

kittaakos, 2022-06-22:
Breaking: https://github.com/arduino/arduino-ide/commit/0f82e913809ad8882ebcd3de985a16d96f62eb9a

per1234, 2022-06-22:
@kittaakos it was a fix for an issue in the private tracker. I have now published that issue to make the information accessible to all interested parties: https://github.com/arduino/arduino-ide/issues/1100

closed 2022-07-06 by davegarthsimpson