arduino/arduino-ide #869: Adding platform specific examples - for example Teensy

KurtE, 2022-02-27:
**Is your feature request related to a problem? Please describe.**
With Arduino 1.8.x releases along with Teensyduino releases, the initial list of examples is augmented with the "Teensy" menu
[image]

The question is what is the best way to do this with IDE2? 

Sorry in advance if this is a duplicate and has been asked/answered before.  I did not find anything specific, so thought best to ask.
This is sort of related to issue: #817  but in reverse.  That is the majority of the top bracket of examples may seldom be used except maybe the real basic ones like blink. 

And as when the IDE first comes up, or right after you change board types to a different platform, these are the only items in the example menu:
[image]
Not sure if that is by design or not, but wish it at least indicated somehow that this menu is incomplete...

**Describe the solution you'd like**

Some way for the Teensy and other platforms to somehow specify some way to add and maybe remove items from this top level list.
Like what @paulstoffregen does with teensy installs for V1 of IDE. 

**Describe alternatives you've considered**

Alternative is to do it like I have seen some others do it, like Robotis with OpenCM9.04

They install a dummy library with their install, called OpenCM9.04, which contains empty files for .h and .cpp, and then has examples under it for the platform.  I tried hacking up a version for Teensy, but so far has not shown up.  Maybe the data is cached somewhere, so would need to clear out all of the data again... 

**Additional context**

Note: I am mostly testing this with Windows 10, currently running nightly build 20220225, which I believe is reasonably up to date.

per1234, 2022-02-27:
>They install a dummy library with their install, called OpenCM9.04, which contains empty files for .h and .cpp, and then has examples under it for the platform.

Yes, this is the common approach. Some other prominent examples:

- https://github.com/esp8266/Arduino/tree/master/libraries/esp8266
- https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32
- https://github.com/MCUdude/MightyCore/tree/master/avr/libraries/AVR_examples
- https://github.com/SpenceKonde/ATTinyCore/tree/master/avr/libraries/ATTinyCore
- https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/libraries/A_STM32_Examples

There is some additional discussion on the subject in regards to Arduino IDE 1.x here: https://github.com/arduino/Arduino/issues/3297

>I tried hacking up a version for Teensy, but so far has not shown up. Maybe the data is cached somewhere, so would need to clear out all of the data again...

Make sure to add [a `library.properties` metadata file](https://arduino.github.io/arduino-cli/latest/library-specification/#library-metadata) to your dummy library so that its examples are not subject to https://github.com/arduino/arduino-ide/issues/40

Make sure you have one of the boards of the Teensy platform selected from the **Tools > Board** menu. The IDE only shows examples of a [platform bundled libraries](https://arduino.github.io/arduino-cli/latest/platform-specification/#platform-bundled-libraries) when a board of its platform is selected (because the library is only accessible when using one of those boards).

The **File > Examples** menu is currently unreliable (https://github.com/arduino/arduino-ide/issues/262). It should eventually work after a few rounds of opening new windows or restarting the IDE. It should work just fine.

For example, here is the Examples menu item for the library from the ESP8266 platform which I linked to above:

[image]

KurtE, 2022-02-27:
Note: That did work to rearrange it more and put files in src, examples and a library properties...

But does make very easy to lose the tree in the forest....
[image]

closed 2024-07-06 by KurtE