arduino/arduino-ide #874: Postponed update is installed on exit

per1234, 2022-02-28:
## Describe the bug

Arduino IDE now has an auto update capability (https://github.com/arduino/arduino-ide/pull/797).

After downloading an update, the user is offered two options:

[image]

From https://github.com/arduino/arduino-ide/pull/797#issue-1121815724

> when a newer version is downloaded, the user can restart the application to apply the new changes, or just close the window; in the last case, the changes will be applied **at the next start-up**

🐛 In contradiction to the described behavior, after you select <kbd>Not now</kbd> and then close the Arduino IDE, the postponed update is installed immediately.

## To Reproduce

1. In order to force an update offer, download an outdated nightly build, using the variant that supports auto-update for your OS.
   Latest outdated URLs at the time of writing are here:
   - [**Windows**](https://downloads.arduino.cc/arduino-ide/nightly/arduino-ide_nightly-20220227_Windows_64bit.exe)
   - [**Linux**](https://downloads.arduino.cc/arduino-ide/nightly/arduino-ide_nightly-20220227_Linux_64bit.AppImage)
   - [**macOS**](https://downloads.arduino.cc/arduino-ide/nightly/arduino-ide_nightly-20220227_macOS_64bit.dmg)
  
   These download URLs can be derived from that of [the current nightly](https://github.com/arduino/arduino-ide#nightly-builds) by decrementing the date component.
1. Install the downloaded application as usual
1. Start the installed Arduino IDE.
1. Open the "**Command Palette**".
   (<kbd>Ctrl</kbd>+<kbd>Shift</kbd>+<kbd>P</kbd> or <kbd>Command</kbd>+<kbd>Shift</kbd>+<kbd>P</kbd> by default)
1. Run the "**Preferences: Open Settings (UI)**" command.
1. Change the "**Arduino › Ide: Update Channel**" (`arduino.ide.updateChannel`) setting to "**nightly**".
1. Select **File > Quit** from the Arduino IDE menus.
1. Start the installed Arduino IDE.
1. Wait for the IDE to finish loading.
1. Click the <kbd>Download</kbd> button on the "**Update Available**" dialog.
1. Wait for the download to finish.
1. Click the <kbd>Not now</kbd> button on the "**Arduino IDE \_\_\_ has been downloaded.**" dialog.
1. Select **File > Quit** from the Arduino IDE menus.
1. Watch the Arduino IDE installation folder.

🐛 The IDE is updated immediately after exit, not on the next start-up.
The installation takes some time and there is no feedback to the user that an operation is in progress, so they might try to open the IDE again or shut down their machine during the installation.

## Expected behavior

Postponed update behavior matches described behavior: update occurs on the next start-up.

## Desktop

- OS: Windows 10
- Version: 2.0.0-rc4-snapshot-354b74c
Date: 2022-02-25T16:28:30.426Z
CLI Version: 0.21.0 [10107d24]

AlbyIanna, 2022-03-01:
I see. This could be an issue when the user tries to open/modify the IDE when the installation is in progress. I'm not sure if/how we can leverage electron-updater to postpone the installation until the next start-up.
I don't think this is very common to happen, so maybe we "fix" this temporarily by changing the copy to make it more clear that the installation is going to happen when the IDE is closed and not on the next start-up. @per1234 what do you think of it?

per1234, 2022-03-02:
I think the best fix is to always run the interactive installer. This will clearly communicate to the user what is happening and also give them the option to cancel the process.

With that change, I don't think any adjustments to the dialog text are needed.

I found it odd that I got a different installation behavior from clicking the <kbd>Close and Install</kbd> button than I did from clicking the <kbd>Not now</kbd> button and then closing the IDE.

per1234, 2022-03-02:
>they're not different behaviour

They are different behaviors. With one you get an interactive installation. With the other you get a silent installation.

## Close and Install

1. Click the <kbd>Close and Install</kbd> button on the "**Arduino IDE \_\_\_ has been downloaded.**" dialog.

The IDE closes and you get the interactive NSIS installation for the update:

[image]

## Not now

1. Click the <kbd>Not now </kbd> button on the "**Arduino IDE \_\_\_ has been downloaded.**" dialog.
1. Select **File > Quit** from the Arduino IDE menus.

The IDE is updated silently:

[image]

per1234, 2022-03-02:
> I think it's not a good idea to quit the IDE automatically immediately after the download finishes, we must warn the user before taking such an action.

Of course. I think you misunderstood my proposal.

Here is the current behavior:

1. Click the <kbd>Not now </kbd> button on the "**Arduino IDE \_\_\_ has been downloaded.**" dialog.
1. Select **File > Quit** from the Arduino IDE menus.
1. The IDE is updated silently.

AlbyIanna, 2022-03-02:
Oh, sorry! I didn't get it at all.
On MacOS (and I think on Linux too) you don't get any interactive installation, even when you click **Close and Install**. 

So, as far as I understand, this is an issue on Windows only. @per1234 is it right?

per1234, 2022-03-02:
>I think on Linux too

That is correct.

>this is an issue on Windows only.

Well, the Linux behavior also does not match [what was described](https://github.com/arduino/arduino-ide/pull/797#issue-1121815724) ("the changes will be applied at the next start-up"). However, the install on close is not a problem because it is only a replacement of a single file and happens almost instantly even when I tried it on a slow old laptop.

The update takes a significant amount of time on Windows and is not atomic. So this is where the silent update caught me by surprise.

per1234, 2022-03-09:
I just got an update in VS Code and thought it would be interesting to see how they handle this "postponed update" condition:

1. Accept the update download by clicking the <kbd>Install Update</kbd> button on the update offer notification:
   [image]
1. Wait for the download to finish.
1. Postpone the installation by clicking the <kbd>Later</kbd> button on the "**Restart Visual Studio Code to apply the latest update.**" notification:
   [image]
1. Select **File > Exit** from the VS Code menus to exit all windows.

The installation occurs immediately. The process is communicated to the user by displaying a progress indicator until it is completed:

[image]

AlbyIanna, 2022-03-09:
Nice one @per1234! I tried getting a grasp on how vscode does automatic updates, but it's quite complex. I'll definitely get back to it when I'll have more time for this issue.

For reference: [updateService.win32.ts](https://github.com/microsoft/vscode/blob/4a130c40ed876644ed8af2943809d08221375408/src/vs/platform/update/electron-main/updateService.win32.ts)