arduino/arduino-ide #876: Takes x 6 times to load in compare to the old IDE

zaid894, 2022-03-02:
**Describe the bug**
Takes so much time to launch and load sketches that you allmostly forget what have you been waiting for, the load time is really too long in compare to the old IDE.

I have an M.2 and 2 x  4GB 3200 MHz RAM though it takes the 2.0 IDE around 31 Seconds to start and load a sketch while 4.5 Seconds at version 1.8.

**To Reproduce**
Steps to reproduce the behavior:
1. Run a sketch 
2. Use a stop watch.

**Expected behavior**
I  was expecting for much  better launch time.

**Desktop (please complete the following information):**
 - Windows 11 
 - 2.0.0 RC 3

ubidefeo, 2022-03-02:
@zaid894 
We are aware of the startup-time issue, but in order to better investigate can you give us more info?
- number of installed libraries (you can count the folders in the Sketchbook's `libraries` folder)
- number of platforms installed and which ones
example: 
- AVR
- SAMD
- MBED
- ESP32
- ESP8266
- STM32Duino

Thank you :)

ckuethe, 2022-03-03:
In my current installation, I have 47 libraries (I used to have over 280) and 21 platforms. I just restarted the IDE and it took 19 seconds from typing `arduino-ide` in a terminal until my sketch was open

Platforms:
```
daemon INFO Loaded platform [platform: {ID:"AmbaSat-1:avr",Installed:"1.0.4",Latest:"1.0.4",Name:"AmbaSat-1 Satellite Board"}]
daemon INFO Loaded platform [platform: {ID:"CubeCell:CubeCell",Installed:"1.4.0",Latest:"1.4.0",Name:"CubeCell Development Framework"}]
daemon INFO Loaded platform [platform: {ID:"Heltec-esp32:esp32",Installed:"0.0.6",Latest:"0.0.6",Name:"Heltec ESP32 Series Dev-boards"}]
daemon INFO Loaded platform [platform: {ID:"MightyCore:avr",Installed:"2.1.3",Latest:"2.1.3",Name:"MightyCore"}]
daemon INFO Loaded platform [platform: {ID:"Sanguino:avr",Installed:"1.0.3",Latest:"1.0.3",Name:"Sanguino"}]
daemon INFO Loaded platform [platform: {ID:"Seeeduino:samd",Installed:"1.8.2",Latest:"1.8.2",Name:"Seeed SAMD Boards"}]
daemon INFO Loaded platform [platform: {ID:"arduino:avr",Installed:"1.8.5",Latest:"1.8.5",Name:"Arduino AVR Boards"}]
daemon INFO Loaded platform [platform: {ID:"arduino:mbed_nano",Installed:"2.7.2",Latest:"2.7.2",Name:"Arduino Mbed OS Nano Boards"}]
daemon INFO Loaded platform [platform: {ID:"arduino:mbed_rp2040",Installed:"2.7.2",Latest:"2.7.2",Name:"Arduino Mbed OS RP2040 Boards"}]
daemon INFO Loaded platform [platform: {ID:"arduino:sam",Installed:"1.6.12",Latest:"1.6.12",Name:"Arduino SAM Boards (32-bits ARM Cortex-M3)"}]
daemon INFO Loaded platform [platform: {ID:"arduino:samd",Installed:"1.8.13",Latest:"1.8.13",Name:"Arduino SAMD Boards (32-bits ARM Cortex-M0+)"}]
daemon INFO Loaded platform [platform: {ID:"arduino:samd_beta",Installed:"1.6.25",Latest:"1.6.25",Name:"Arduino SAMD (32-bits ARM Cortex-M0+) Beta Boards"}]
daemon INFO Loaded platform [platform: {ID:"digistump:avr",Installed:"1.6.7",Latest:"1.6.7",Name:"Digistump AVR Boards"}]
daemon INFO Loaded platform [platform: {ID:"esp32:esp32",Installed:"1.0.6",Latest:"1.0.6",Name:"esp32"}]
daemon INFO Loaded platform [platform: {ID:"esp8266:esp8266",Installed:"3.0.2",Latest:"3.0.2",Name:"esp8266"}]
daemon INFO Loaded platform [platform: {ID:"heltec-esp8266:esp8266",Installed:"0.0.4",Latest:"0.0.4",Name:"Heltec ESP8266 Series Dev-boards"}]
daemon INFO Loaded platform [platform: {ID:"megax4:avr",Installed:"1.0.3",Latest:"1.0.3",Name:"ATmega644P/1284P Boards"}]
daemon INFO Loaded platform [platform: {ID:"megax8:avr",Installed:"1.0.3",Latest:"1.0.3",Name:"ATmega168P/328P Boards"}]
daemon INFO Loaded platform [platform: {ID:"rabid prototypes:samd",Installed:"1.6.5",Latest:"1.6.5",Name:"Rabid Prototypes SAMD Boards"}]
daemon INFO Loaded platform [platform: {ID:"tiny:avr",Installed:"1.0.6",Latest:"1.0.6",Name:"ATtiny extra Boards"}]
daemon INFO Loaded platform [platform: {ID:"breadboard:avr",Name:"breadboard-avr"}]
```

Libraries
```
Adafruit_AMG88xx_Library
Adafruit_BME280_Library
Adafruit_BusIO
Adafruit_Circuit_Playground
Adafruit_GFX_Library
Adafruit_GPS_Library
Adafruit_ILI9341
Adafruit_INA219
Adafruit_LED_Backpack_Library
Adafruit_MLX90614_Library
Adafruit_NeoPixel
Adafruit_seesaw_Library
Adafruit_SHTC3_Library
Adafruit_SleepyDog_Library
Adafruit_SSD1306
Adafruit_ST7735_and_ST7789_Library
Adafruit_STMPE610
Adafruit_TouchScreen
Adafruit_Unified_Sensor
Adafruit_Zero_DMA_Library
Adafruit_Zero_FFT_Library
Adafruit_Zero_PDM_Library
ArduinoINA219
AsyncElegantOTA
DebounceEvent
DS18B20
ElegantOTA
ESP32_ESP32S2_AnalogWrite
ESP32Time
GY39
HamShield
Heltec_ESP32_Dev-Boards
HX711
IRremoteESP8266
Keyboard
Max44009
MAX44009_library
Mouse
OneWire
RTClib
SD
[... 39 lines cut ...]
daemon INFO Checking signature [error: opening signature file: open /tmp/988278816.sig: no such file or directory, index: /tmp/988278816, signaturefile: /tmp/988278816.sig]
daemon INFO URL: https://rabidprototypes.com/arduino/package_rabidprototypes_index.json
daemon INFO Updating index [url: https://rabidprototypes.com/arduino/package_rabidprototypes_index.json]
root INFO Updating index: package_rabidprototypes_index.json completed.
daemon INFO Checking signature [error: opening signature file: open /tmp/220566271.sig: no such file or directory, index: /tmp/220566271, signaturefile: /tmp/220566271.sig]
daemon INFO URL: https://raw.githubusercontent.com/Lauszus/Sanguino/master/package_lauszus_sanguino_index.json
daemon INFO Updating index [url: https://raw.githubusercontent.com/Lauszus/Sanguino/master/package_lauszus_sanguino_index.json]
root INFO Updating index: package_lauszus_sanguino_index.json completed.
daemon INFO Checking signature [error: opening signature file: open /tmp/360745042.sig: no such file or directory, index: /tmp/360745042, signaturefile: /tmp/360745042.sig]
daemon INFO URL: https://resource.heltec.cn/download/package_CubeCell_index.json
daemon INFO Updating index [url: https://resource.heltec.cn/download/package_CubeCell_index.json]
root INFO Updating index: package_CubeCell_index.json completed.
daemon INFO Checking signature [error: opening signature file: open /tmp/662187401.sig: no such file or directory, index: /tmp/662187401, signaturefile: /tmp/662187401.sig]
daemon INFO URL: https://resource.heltec.cn/download/package_heltec_esp32_index.json
daemon INFO Updating index [url: https://resource.heltec.cn/download/package_heltec_esp32_index.json]
root INFO Updating index: package_heltec_esp32_index.json completed.
[... cut ...]
daemon INFO Updating index [url: https://resource.heltec.cn/download/package_heltec_esp8266_index.json]
root INFO Updating index: package_heltec_esp8266_index.json completed.
daemon INFO Checking signature [error: opening signature file: open /tmp/213301411.sig: no such file or directory, index: /tmp/213301411, signaturefile: /tmp/213301411.sig]
daemon INFO Updating libraries index
root INFO Updating index: library_index.json.gz completed.
root INFO Updating index: library_index.json.sig completed.
```

ckuethe, 2022-03-03:
```
root WARN Frontend keybindings.onStart is slow, took: 193.1 ms
root ERROR DefaultJsonSchemaContribution.registerSchemas is taking more than 500.0 ms, new schemas are ignored.
...
root WARN Frontend EditorNavigationContribution.onStart is slow, took: 19406.7 ms
...
root INFO [b35162f2-f205-4035-9499-3ce2055c3d9c] Waiting for backend deployment took: 250.6 ms [Finished 21.166 s after frontend start]
root INFO [b35162f2-f205-4035-9499-3ce2055c3d9c] Sync of 5 plugins took: 253.8 ms [Finished 21.420 s after frontend start]
Update for version 2.0.0-rc4 is not available (latest version: 2.0.0-rc4, downgrade is allowed).
...
root INFO [b35162f2-f205-4035-9499-3ce2055c3d9c] Load contributions of 5 plugins took: 88.6 ms [Finished 22.200 s after frontend start]
root INFO Finished loading frontend application after 22.231 seconds
```

ubidefeo, 2022-03-04:
thank you @ckuethe for the comprehensive output.
we'll look into why all those signature errors show up

u.

ckuethe, 2022-03-04:
I don't think the signature errors are as important as not blocking app startup while checking for updates. When I turn off my network the IDE loads much more quickly: `root INFO Finished loading frontend application after 5.734 seconds`

ckuethe, 2022-03-14:
Wow. rc5 is even worse. 

```
root WARN Frontend EditorNavigationContribution.onStart is slow, took: 259339.6 ms
root INFO Board config changed: 
root INFO Board config changed: 
root INFO [hosted-plugin: 1581552] PLUGIN_HOST(1581552) starting instance
root INFO Changed application state from 'init' to 'started_contributions'.
root INFO Changed application state from 'started_contributions' to 'attached_shell'.
root INFO >>> Restoring the layout state...
Checking for update
root INFO [29567868-3050-46bf-ae98-8526924c68fe] Waiting for backend deployment took: 295.9 ms [Finished 260.987 s after frontend start]
root INFO Board config changed: 
serial-service INFO >>> Disposing serial connection...
serial-service WARN <<< Not connected. Nothing to dispose.
serial-service INFO >>> Disposing serial connection...
serial-service WARN <<< Not connected. Nothing to dispose.
root INFO [29567868-3050-46bf-ae98-8526924c68fe] Sync of 5 plugins took: 279.8 ms [Finished 261.267 s after frontend start]
Found version 2.0.0-rc5 (url: arduino-ide_2.0.0-rc5_Linux_64bit.AppImage)
root INFO <<< The layout has been successfully restored.
```

kittaakos, 2022-03-17:
> as not blocking app startup while checking for updates

The IDE starts to `await` for the `index-update` in this commit: https://github.com/arduino/arduino-ide/commit/9aff90b0aff46503c6f87c133a0a9d1f4de5a895#diff-f463d1c276a9080ec522ead62c6dd596e5bf330b093b7d51b71738c9795d0fdaL60-R60.

Auto index-update CLI changes:
 - https://github.com/arduino/arduino-cli/pull/1043
 - https://github.com/arduino/arduino-cli/pull/1063
 - https://github.com/arduino/arduino-cli/pull/1237 (👈 this could be interesting)

I did some measurements on the `HEAD` (using `console.time`):

No 3rd party URLs:

1. app start:
```
root INFO $$$ update-index: 966.274ms
root INFO $$$ update-lib-index: 980.132ms
root INFO $$$ init-instance: 1.174s
root INFO $$$ all: 3.129s
```

2. app start:
```
root INFO $$$ update-index: 954.208ms
root INFO $$$ update-lib-index: 896.597ms
root INFO $$$ init-instance: 948.474ms
root INFO $$$ all: 2.800s
```

----

My 3rd party URLs:

```
http://digistump.com/package_digistump_index.json,http://www.leonardomiliani.com/repository/package_leonardomiliani.com_index.json,https://ambasat.com/boards/package_ambasat-1.com_index.json,https://arduino.esp8266.com/stable/package_esp8266com_index.json,https://dl.espressif.com/dl/package_esp32_index.json,https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json,https://mcudude.github.io/MightyCore/package_MCUdude_MightyCore_index.json,https://rabidprototypes.com/arduino/package_rabidprototypes_index.json,https://raw.githubusercontent.com/Lauszus/Sanguino/master/package_lauszus_sanguino_index.json,https://resource.heltec.cn/download/package_CubeCell_index.json,https://resource.heltec.cn/download/package_heltec_esp32_index.json,https://resource.heltec.cn/download/package_heltec_esp8266_index.json
```

1. app start (with `rm -rf ~/Library/Arduino15/ && rm -rf ~/Documents/Arduino/libraries`):
```
root INFO $$$ update-index: 8.020s
root INFO $$$ update-lib-index: 1.010s
root INFO $$$ init-instance: 2.332s
root INFO $$$ all: 11.362s
```

2. app start (with `rm -rf ~/Library/Arduino15/ && rm -rf ~/Documents/Arduino/libraries`):
```
root INFO $$$ update-index: 9.953s
root INFO $$$ update-lib-index: 862.065ms
root INFO $$$ init-instance: 2.107s
root INFO $$$ all: 12.922s
```

3. app start:
```
root INFO $$$ update-index: 8.702s
root INFO $$$ update-lib-index: 875.767ms
root INFO $$$ init-instance: 499.915ms
root INFO $$$ all: 10.078s
```

4. app start:
```
root INFO $$$ update-index: 8.343s
root INFO $$$ update-lib-index: 904.521ms
root INFO $$$ init-instance: 642.374ms
root INFO $$$ all: 9.890s
```

(_Note: the numbers do not necessarily add up, but `all >=` the sum of the sub-tasks._)

On my dev env, zero cores and libs are installed. Still, the core index update takes a long time and heavily depends on the 3rd party URLs config.

stodge, 2022-03-18:
It can take 30s+ to load on my laptop, and seems to take at least 15s before there's any visual indication that it's loading. Admittedly I have an HDD and not an SSD, but something like Visual Code doesn't have this problem. I'm on Windows 10 - is there a way to show startup logs?

per1234, 2022-03-18:
>is there a way to show startup logs?

Hi @stodge. You should get everything of interest by starting the Arduino IDE from the command line.

I'll provide instructions for how you can do that and then share the output here:

1. Start **File Explorer** (AKA "**Windows Explorer**").
1. Open the Arduino IDE installation folder
   (usually `C:\Program Files\Arduino IDE` or `C:\Users\<user name>\AppData\Local\Programs\Arduino IDE`).
1. In the folder listing of the Arduino IDE installation folder, hold the **<kbd>Shift</kbd>** key while clicking the right hand button on the mouse.
1. From the context menu, click "**Open PowerShell window here**".
   PowerShell will now open with the current directory set to the Arduino IDE installation folder.
1. Type `& ".\Arduino IDE"`.
1. Press the **<kbd>Enter</kbd>** key.
   Debug output should now be printed to the PowerShell window as the Arduino IDE starts (or fails to start).
1. Wait for the Arduino IDE to finish starting, or failing to start.
1. Switch back to the PowerShell window if the Arduino IDE window took the focus.
1. Right click on the title bar of the PowerShell window.
   This will open a context menu.
1. Select **Edit > Select All** from the context menu.
   This will select all the text in the PowerShell window.
1. Press **<kbd>Ctrl</kbd>+<kbd>C</kbd>**.
   This will copy the contents of the PowerShell window to the clipboard.
1. Paste (**<kbd>Ctrl</kbd>+<kbd>V</kbd>**) the output in a reply to this GitHub issue thread.

---

Please let me know if you have any questions or problems while following those instructions.

stodge, 2022-03-18:
Thanks. I ran the command as you suggested, and it did nothing for 5 minutes until I pressed RETURN in Power Shell. It still has the problem where you the splash logo sits over everything on the screen and you can't bring any other window on top of it. Will post the logs in a bit.

I started the IDE from the command line before you replied, but as a foreground process. Again, it didn't do anything for 5+{ minutes until I pressed return in PowerShell and both instances started at the same time.

I'll close both instances and restart. More later... Thanks.

stodge, 2022-03-18:
Ok, so if I run the command, it tells me that the background process has started. But the app doesn't load until I hit return again. Then the splash screen and window are displayed, but it doesn't finish loading until I interact with the window (i.e. use the basic menu). Then it loads quickly. If I don't press return the second time at startup, it never loads. If I don't interact with the window during startup, the splash screen never goes away and the app never loads. This is rc5.

> PS C:\Program Files\Arduino IDE> & ".\Arduino IDE"
PS C:\Program Files\Arduino IDE>
PS C:\Program Files\Arduino IDE> Starting backend process. PID: 12660
PS C:\Program Files\Arduino IDE> Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///c%3A/Users/mike/.theia'
root INFO Theia app listening on http://localhost:55479.
root WARN The local plugin referenced by local-dir:/c%3A/Users/mike/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/c%3A/Users/mike/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:C:\Users\mike\.arduinoProIDE\plugins does not exist.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root WARN Backend DefaultWorkspaceServer.onStart is slow, took: 2296.7 ms
root INFO Deploying backend plugin "cortex-debug@0.3.10" from "C:\Program Files\Arduino IDE\resources\app\plugins\cortex-debug\extension\dist\extension"
root INFO Deploying backend plugin "vscode-arduino-tools@0.0.2-beta.2" from "C:\Program Files\Arduino IDE\resources\app\plugins\vscode-arduino-tools\extension\dist\bundle"
root INFO Deploying backend plugin "cpp@1.52.1" from "C:\Program Files\Arduino IDE\resources\app\plugins\vscode-builtin-cpp\extension"
root INFO Deploying backend plugin "json@1.46.1" from "C:\Program Files\Arduino IDE\resources\app\plugins\vscode-builtin-json\extension"
root INFO Deploying backend plugin "json-language-features@1.46.1" from "C:\Program Files\Arduino IDE\resources\app\plugins\vscode-builtin-json-language-features\extension\client\dist\jsonMain"
root INFO Deploy plugins list took: 2314.8 ms
daemon INFO Starting daemon from C:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
root WARN Backend ConfigServiceImpl.onStart is slow, took: 200.5 ms
daemon INFO Using config file: c:\Users\mike\.arduinoIDE\arduino-cli.yaml
daemon INFO arduino-cli.exe version 0.21.0
daemon INFO Executing `arduino-cli daemon`
daemon INFO Setting up Prometheus metrics on :9090/metrics
daemon INFO {"IP":"127.0.0.1","Port":"55480"}
daemon INFO Daemon is running.
daemon INFO Adding libraries dir [dir: c:\Users\mike\Documents\Arduino\libraries, location: user]
daemon INFO URL: https://downloads.arduino.cc/packages/package_index.json
daemon INFO Updating index [url: https://downloads.arduino.cc/packages/package_index.json]
root INFO Git askpass helper is listening on http://127.0.0.1:57446.
root ERROR Error: Git installation not found.
    at C:\Program Files\Arduino IDE\resources\app\node_modules\find-git-exec\lib\find-git-exec.js:85:69
    at async DefaultGitInit.init (C:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\lib\node\theia\git\git-init.js:22:49)
root WARN Frontend CommonFrontendContribution.configure is slow, took: 422.4 ms
root WARN Failed to resolve module: @types/body-parser
root WARN Failed to resolve module: @types/cookie
root WARN Failed to resolve module: @types/dompurify
root WARN Failed to resolve module: @types/express
root WARN Failed to resolve module: @types/fs-extra
root WARN Failed to resolve module: @types/lodash.debounce
root WARN Failed to resolve module: @types/lodash.throttle
root WARN Failed to resolve module: @types/markdown-it
root WARN Failed to resolve module: @types/react
root WARN Failed to resolve module: @types/react-dom
root WARN Failed to resolve module: @types/react-virtualized
root WARN Failed to resolve module: @types/route-parser
root WARN Failed to resolve module: @types/safer-buffer
root WARN Failed to resolve module: @types/ws
root WARN Failed to resolve module: @types/yargs
root WARN Failed to resolve module: @theia/monaco
root WARN Failed to resolve module: @theia/editor
root WARN Failed to resolve module: @types/multer
root WARN Failed to resolve module: @types/rimraf
root WARN Failed to resolve module: @types/tar-fs
root WARN Failed to resolve module: @types/uuid
root WARN Failed to resolve module: @theia/navigator
root WARN Failed to resolve module: @types/p-queue
root WARN Failed to resolve module: @theia/preferences
[... cut ...]
daemon INFO Resolving include Arduino_ConnectionHandler.h for arch esp8266
daemon INFO   found better lib [lib: Arduino_ConnectionHandler, prio: 64D]
daemon INFO Resolving include ESP8266WiFi.h for arch esp8266
daemon INFO   found better lib [lib: ESP8266WiFi, prio: 64C]
daemon INFO Resolving include Arduino_DebugUtils.h for arch esp8266
daemon INFO   found better lib [lib: Arduino_DebugUtils, prio: 643]
daemon INFO Resolving include ArduinoMqttClient.h for arch esp8266
daemon INFO   found better lib [lib: ArduinoMqttClient, prio: 643]

per1234, 2022-03-18:
>the splash screen and window are displayed, but it doesn't finish loading until I interact with the window (i.e. use the basic menu)

This bug is tracked at https://github.com/arduino/arduino-ide/issues/193

>Then it loads quickly.

By this, do you mean that you are no longer experiencing the original problem you reported?:

>It can take 60s+ to load on my laptop, and seems to take at least 30s before there's any visual indication that it's loading.

ubidefeo, 2022-04-01:
@msujew 
let's have a chat about this one, as right now we wait for the Language Server's operations completion before we show the windows' content.
If we manage to defer the LS operations to a later time the IDE will be ready much earlier.
VS Code itself does not run intellisense before the windows are ready

ckuethe, 2022-04-25:
Ten minutes...

```
root INFO [09302298-f36c-4a36-8809-9f0436efc742] Load contributions of 5 plugins took: 116.5 ms [Finished 589.341 s after frontend start]
root WARN Frontend EditorPreviewTreeDecorator.onDidInitializeLayout is slow, took: 117.3 ms
root INFO Finished loading frontend application after 589.382 seconds
```

closed 2022-06-08 by fstasi

stodge, 2022-06-10:
rc7 is exactly the same for me, no improvement. Did you uninstall rc6 before installing rc7? I installed rc7 over rc6.

habazut, 2022-10-30:
IMHO the times are still far too long. 

```
Arduino IDE 2.0.1
Starting backend process. PID: 2366249
[2366269:1030/115436.958746:ERROR:sandbox_linux.cc(374)] InitializeSandbox() called with multiple threads in process gpu-process.
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///home/haba/.theia'
root INFO Backend Object.initialize: 3.6 ms [Finished 0.566 s after backend start]
root INFO Backend Object.configure: 2.8 ms [Finished 0.566 s after backend start]
root INFO Backend MessagingContribution.onStart: 1.3 ms [Finished 0.566 s after backend start]
root INFO Backend OSBackendApplicationContribution.configure: 0.3 ms [Finished 0.593 s after backend start]
root INFO Backend DefaultWorkspaceServer.onStart: 0.3 ms [Finished 0.593 s after backend start]
root INFO Backend LocalizationBackendContribution.configure: 0.2 ms [Finished 0.593 s after backend start]
root INFO Backend LocalizationBackendContribution.initialize: 26.4 ms [Finished 0.593 s after backend start]
root INFO Backend TaskBackendApplicationContribution.onStart: 0.2 ms [Finished 0.593 s after backend start]
root INFO Backend ElectronTokenBackendContribution.configure: 0.8 ms [Finished 0.594 s after backend start]
root INFO Backend PluginDeployerContribution.initialize: 0.3 ms [Finished 0.594 s after backend start]
root INFO Backend ConfigServiceImpl.onStart: 0.6 ms [Finished 0.595 s after backend start]
root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.3 ms [Finished 0.595 s after backend start]
root INFO Backend ArduinoDaemonImpl.onStart: 0.8 ms [Finished 0.596 s after backend start]
root INFO Backend NodeFileUploadService.configure: 1.5 ms [Finished 0.596 s after backend start]
daemon INFO Starting daemon from /usr/share/arduino-ide_2.0.1_Linux_64bit/resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
discovery-log INFO start
discovery-log INFO start new deferred
root INFO Backend BoardDiscovery.onStart: 0.7 ms [Finished 0.597 s after backend start]
root INFO Backend FileDownloadEndpoint.configure: 0.4 ms [Finished 0.597 s after backend start]
root INFO Backend AuthenticationServiceImpl.onStart: 0.4 ms [Finished 0.598 s after backend start]
root INFO Backend PluginApiContribution.configure: 0.4 ms [Finished 0.598 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.2 ms [Finished 0.599 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.0 ms [Finished 0.599 s after backend start]
root INFO Backend PlotterBackendContribution.configure: 0.3 ms [Finished 0.599 s after backend start]
root INFO Theia app listening on http://localhost:37849.
root INFO Finished starting backend application: 2.5 ms [Finished 0.601 s after backend start]
root WARN The local plugin referenced by local-dir:/home/haba/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/home/haba/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:/home/haba/.arduinoProIDE/plugins does not exist.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.5" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-bg" to a VS Code extension "vscode-language-pack-bg@1.48.3" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-cs" to a VS Code extension "vscode-language-pack-cs@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-de" to a VS Code extension "vscode-language-pack-de@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-es" to a VS Code extension "vscode-language-pack-es@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-fr" to a VS Code extension "vscode-language-pack-fr@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-hu" to a VS Code extension "vscode-language-pack-hu@1.48.3" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-it" to a VS Code extension "vscode-language-pack-it@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-ja" to a VS Code extension "vscode-language-pack-ja@1.53.2" with engines:
[... cut ...]
Marking workspace as a closed sketch. Workspace URI: file:///home/haba/Arduino/Pipi. Date: 1667127968731.
Skipped storing sketch as workspace root. Expected minimum threshold: <1667127967764>. Was: <1667127964259>.
Skipped storing sketch as workspace root. Expected minimum threshold: <1667127967764>. Was: <1667127964281>.
Storing the sketch as a workspace root: </home/haba/Arduino/Pipi>.
Skipped storing sketch as workspace root. Already visited: </home/haba/Arduino/Pipi>.
Stored workspaces roots: /home/haba/Arduino/Pipi
```