arduino/arduino-ide #881: Incorrect "Additional Boards Manager URLs" dialog field height

per1234, 2022-03-03:
## Describe the bug

Arduino IDE supports adding installing arbitrary boards platforms via [**Boards Manager**](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-board-manager) by adding the URLs of their [package index](https://arduino.github.io/arduino-cli/dev/package_index_json-specification/) file to the "**Additional Boards Manager URLs**" preference.

Since users may have several URLs, a dedicated "**Additional Boards Manager URLs**" dialog is offered with a multi-line field.

🐛 The height of the field is very small, to the point where it is not clear that it is an input field at all.

## To Reproduce

1. Select **File > Preferences...** from the Arduino IDE menus.
1. Click the button on the right side of the "**Additional Boards Manager URLs**" field.

🐛 The height of the field is very small:

[image]

## Expected behavior

Usable input field height in the "**Additional Boards Manager URLs**" dialog, as it was in previous versions:

[image]

## Desktop

- OS: Windows 10
- Version: 2.0.0-rc4-snapshot-0fc7c78
Date: 2022-03-03T08:39:37.612Z
CLI Version: 0.21.0 [10107d24]

## Additional context

I bisected the bug to https://github.com/arduino/arduino-ide/commit/112153fb965f63d952d126c8244cd3f84f0a1a1b (it does not occur when using the build from the previous commit https://github.com/arduino/arduino-ide/commit/69ac1f4779589d0d21ce3d37c180b3393ad6156c).

KurtE, 2022-03-03:
Thanks @per1234 was just about to report it after downloading rc4... 

I tried resizing the window using the resize part at lower right of text box, but it would only change horizontal size, not vertical. 
Windows 10

VE3EZB, 2022-03-11:
I have the same issue:

Processor	Intel(R) Core(TM) i5-6500 CPU @ 3.20GHz   3.19 GHz
Installed RAM	16.0 GB (15.9 GB usable)
Device ID	F2DD9474-AF46-4403-8077-82B34F6BBDBA
Product ID	00330-80000-00000-AA808
System type	64-bit operating system, x64-based processor

Arduino IDE 
Version: 2.0.0-rc5
Date: 2022-03-10T11:09:10.332Z
CLI Version: 0.21.0 [10107d24]

Copyright © 2022 Arduino SA

commit 0db119d7 (2022-04-05): #919, #881: Fixed 3rd party URLs-related issues (#920)

* Fixed empty string to URLs conversion

Closes #919.

* #881: Fixed height of the 3rd part URLs `textarea`

Closes #881.
closed 2022-04-04 by fstasi