arduino/arduino-ide #889: The installer icon is wrong

mklemarczyk, 2022-03-06:
**Describe the bug**
The installer icon is wrong.

**To Reproduce**
Steps to reproduce the behavior:
Download and run the installer for Arduino on Windows 10 (64 bit).
Choose the installation for all users.

**Expected behavior**
I would expect the generic MSI icon or Arduino icon.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 64 bits
 - Version: 2.0.0-rc4

**Additional context**
The icon can confuse the user that something wrong happened to the installation.

per1234, 2022-03-06:
This is how the 2.0.0-rc4 installer looks on my computer:

[image]

I wonder what the difference is between our systems? Is this your first time installing the Arduino IDE 2.x @mklemarczyk?

I believe this is configured by the [`build.nsis.installerIcon` key](https://www.electron.build/generated/nsisoptions#NsisOptions-installerIcon):

https://github.com/arduino/arduino-ide/blob/49971ada078da7c9aaef29adc30e894af67c0d1a/electron/build/template-package.json#L113

And the referenced file is here:

https://github.com/arduino/arduino-ide/blob/49971ada078da7c9aaef29adc30e894af67c0d1a/electron/build/resources/icon.ico

mklemarczyk, 2022-03-17:
I was installing it after having the rc-2 installed before. I assume it made an update from rc-2 to rc-4.

per1234, 2022-03-18:
I just tried the update from 2.0.0-rc2 to 2.0.0-rc4 and still wasn't able to reproduce the issue.

Please give the installer from the nightly build a try and let us know whether the problem still occurs:

https://github.com/arduino/arduino-ide#nightly-builds

mklemarczyk, 2022-03-19:
Testes with the latest version.
The icon is fine just until step of changing to administration for installation for all users. Thank it becomes red.
Installing for local user:
[image]

Installing for all users (tested with upgrade and clean installation - same behavior):
[image]

When uninstalling for users the icon is correct:
[image]