arduino/arduino-ide #898: In my window11  Arduino IDE 2 RC4 can't run!

kylongmu, 2022-03-10:
A double click on the icon of Arduino IDE 2 RC4, wait and nothing appear.
but the arduino 1.8.x is work well.

per1234, 2022-03-10:
Hi @kylongmu. Thanks for your report.

I'm going to ask you to provide some additional information that might help us to identify the problem.

Please do this:

1. Start **File Explorer** (AKA "**Windows Explorer**").
1. Open the Arduino IDE installation folder
   (usually `C:\Program Files\Arduino IDE` or `C:\Users\<user name>\AppData\Local\Programs\Arduino IDE`).
1. In the folder listing of the Arduino IDE installation folder, hold the **<kbd>Shift</kbd>** key while clicking the right hand button on the mouse.
1. From the context menu, click "**Open PowerShell window here**".
   PowerShell will now open with the current directory set to the Arduino IDE installation folder.
1. Type `& ".\Arduino IDE"`.
1. Press the **<kbd>Enter</kbd>** key.
   Debug output should now be printed to the PowerShell window as the Arduino IDE starts (or fails to start).
1. Wait for the Arduino IDE to finish starting, or failing to start.
1. Switch back to the PowerShell window if the Arduino IDE window took the focus.
1. Right click on the title bar of the PowerShell window.
   This will open a context menu.
1. Select **Edit > Select All** from the context menu.
   This will select all the text in the PowerShell window.
1. Press **<kbd>Ctrl</kbd>+<kbd>C</kbd>**.
   This will [copy](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the contents of the PowerShell window to the clipboard.
1. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) (**<kbd>Ctrl</kbd>+<kbd>V</kbd>**) the output in a reply to this GitHub issue thread.

---

Please let me know if you have any questions or problems while following those instructions.

ednieuw, 2022-03-11:
I solved the problem in W11. 
I noticed in the W11 window "Add and Remove programs" the RC3 installation, above the RC5 installation, that could not me removed. Maybe I installed RC3 "Only for me" instead "For all users"
I removed RC5, installed RC3 "Only for me".
In W11 there are two locations to "add and remove programs". With the old W7 windows location: 
`Control Panel\Programs\Programs and Features`
and the new W11 environment. In the W11 windows you can not delete RC3 but you can with `Control Panel\Programs\Programs and Features` you can start with this link in file explorer
Remove the installation of RC3.
and remove this folder: c:\Users\YOURNAME\.arduinoIDE\
Restart the PC 
Install RC5. In 10 seconds you see the editor.

The RC3 installation is still visible in the W11 "Add and Remove programs"  with is annoying.

kylongmu, 2022-03-16:
> Hi @kylongmu. Thanks for your report.
> 
> I'm going to ask you to provide some additional information that might help us to identify the problem.
> 
> Please do this:
> 
>     1. Start **File Explorer** (AKA "**Windows Explorer**").
> 
>     2. Open the Arduino IDE installation folder
>        (usually `C:\Program Files\Arduino IDE` or `C:\Users\<user name>\AppData\Local\Programs\Arduino IDE`).
> 
>     3. In the folder listing of the Arduino IDE installation folder, hold the **Shift** key while clicking the right hand button on the mouse.
> 
>     4. From the context menu, click "**Open PowerShell window here**".
>        PowerShell will now open with the current directory set to the Arduino IDE installation folder.
> 
>     5. Type `& ".\Arduino IDE"`.
> 
>     6. Press the **Enter** key.
>        Debug output should now be printed to the PowerShell window as the Arduino IDE starts (or fails to start).
> 
>     7. Wait for the Arduino IDE to finish starting, or failing to start.
> 
>     8. Switch back to the PowerShell window if the Arduino IDE window took the focus.
> 
>     9. Right click on the title bar of the PowerShell window.
>        This will open a context menu.
> 
>     10. Select **Edit > Select All** from the context menu.
>         This will select all the text in the PowerShell window.
> 
>     11. Press **Ctrl+C**.
>         This will [copy](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the contents of the PowerShell window to the clipboard.
> 
>     12. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) (**Ctrl+V**) the output in a reply to this GitHub issue thread.
> 
> 
> Please let me know if you have any questions or problems while following those instructions.
Sorry for late to reply, I tried your guide, after step 6, I got these info:
`PS C:\Users\THINKPAD\AppData\Local\Programs\arduino-ide> & ".\Arduino IDE"
PS C:\Users\THINKPAD\AppData\Local\Programs\arduino-ide>

#
# Fatal error in , line 0
# Version mismatch between V8 binary and snapshot.
#   V8 binary version: 8.3.110.13-electron.0
#    Snapshot version: 9.4.146.22-electron.0
# The snapshot consists of 164467 bytes and contains 3 context(s).
#
#
#
#FailureMessage Object: 00000001000FE900`

kylongmu, 2022-03-16:
> I solved the problem in W11. I noticed in the W11 window "Add and Remove programs" the RC3 installation, above the RC5 installation, that could not me removed. Maybe I installed RC3 "Only for me" [...]

Thanks for your help, I follows your way,   `Control Panel\Programs\Programs and Features` shows 3 install of arduino 2.0, so I remove all of them, and remove this folder: c:\Users\YOURNAME.arduinoIDE Restart the PC Install RC5. In 10 seconds I see the editor.
But, but the editor file pull down menu just shows only "exit", so I exit and try again, this time with nothing, check the system process shows 9 zobine procees with "Arduino IDE"  and a1 zobine process with "arduino-cli.exe", I try to kill them but failed.