arduino/arduino-ide #902: abs() not recognized as function if not after equals sign

blopa1961, 2022-03-10:
**Describe the bug**
In arduino IDE abs() get marked as "no matching function for all to 'abs'" if it does not follow and equals sign
(the program compiles without errors)

**To Reproduce**
Steps to reproduce the behavior:
in a new sketch select nodeMCU 1.0
write a statement like:

if (abs(i) < j) {
  etc...
}

abs (i) gets flagged as error in the IDE: "no matching function call for 'abs'"

k= abs(i); // this does not get flagged

same problem occurs if sprintf is not assigned to a variable, example:
sprintf(deciSecs, "%.1f",((float) timeStep) / 1000.0);
instead of
i= sprintf( etc... );

**Expected behavior**
I didn't expect this to be flagged

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Version: 21H2

per1234, 2022-03-11:
Hello @blopa1961. Thanks for your report.

We will need some additional information in order to effectively investigate this problem.

>select nodeMCU 1.0

Please provide more detailed instructions. I can guess about what you mean, but I might have guessed wrong, and that wrong guess might be the reason I can't reproduce the bug you are experiencing.

I can see that this is a board from a 3rd party boards platform, but which specific boards platform? 

Which version of that boards platform are you using?

Did you install it via Boards Manager, or did you install it to the sketchbook folder?

What is the exact name of the board as shown under the **Tools > Board** menu in the Arduino IDE (e.g., "**NodeMCU 1.0 (ESP-12E Module)**")?

>write a statement like:

Please provide a complete minimal sketch we can use to reproduce the problem.

Provide one sketch for the `abs` issue, and another for the `sprintf` issue.

blopa1961, 2022-03-11:
Hello @per1234 and thank you for your fast reply.

About the 3rd party board, it's a standard nodeMCU.
To follow me:
open Arduino IDE, open preferences
in Additional Board Manager URLs add "http://arduino.esp8266.com/stable/package_esp8266com_index.json"
in Tools/Board esp8266 select nodeMCU 1.0 (ESP-12E module)

I tried to create a new sketch with the abs() or sprintf() problems but guess what, I can't reproduce the problem in a small sketch...

In the sketch I am working on now (which is more than 2000 lines long) it still happens, I attached a couple of screen captures where you can see the problems I reported.
When I opened my sketch the IDE did not initially flag abs(), but after I searched for "sprintf" both sprintf() and abs() were flagged (is this an uninitialized variable problem?) sorry, I'm a programmer too ;)

BTW my computer is a Core i7 with 32Gb RAM and a Samsung 850 Pro SSD disk, so I don't think this is a resource issue...

[image]
[image]

Regards, blopa1961

blopa1961, 2022-03-11:
I keep investigating the issue, please take a look at this capture (!)
[image]
The abs() function got flagged, but it was underlined 4 lines below the actual abs() occurrence within a comment, but in the correct column. Please notice it is aligned below the abs() in line 557.
Also, please notice the } in line 559 (which ends the prevoius function) is also underlined and reports the same 'abs' problem. Is this a feature?

blopa1961, 2022-04-01:
Hello @ubidefeo 
I took the time to test a few of my small sketches and found one (attached) where printf gets flagged as you can see in the screen capture:
[image]
[NTP_Time.zip](https://github.com/arduino/arduino-ide/files/8401089/NTP_Time.zip)

I hope you can reproduce the problem this time.
Regards, blopa1961

Looky1173, 2022-05-29:
It seems like most built-in functions are getting flagged like so:
[image]

closed 2023-08-17 by blopa1961