arduino/arduino-ide #905: Serial monitor in Arduino IDE v. 2.0.0-rc4 doesn't pad timestamps correctly

colde, 2022-03-11:
**Describe the bug**
When enabling timestamps in the serial monitor, timestamps might be formatted as 18:2:13.521 instead of as expected 18:02:13.521

I.e. this is the data i got in my console:
```
17:56:13.570 -> New message received
17:58:24.586 -> New message received
18:0:14.054 -> New message received
18:2:13.521 -> New message received
18:2:18.212 -> New message received
```

**To Reproduce**
Steps to reproduce the behavior:
1. Have a board connected that outputs serial data
2. Enable the serial monitor timestamp feature 
3. Wait for a time where between 0 and 9 minutes past the hour
4. See error

**Expected behavior**
The timestamps should prepend a 0 for single digit minutes 

**Desktop (please complete the following information):**
 - OS: macOS Monterey 12.2.1
 - Version: 2.0.0-rc4

per1234, 2022-03-11:
Hi @colde. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/580.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-03-11 by per1234

colde, 2022-03-12:
Apologies. My search didn't find that.