arduino/arduino-ide #906: Implementation error. Neither 'download' nor 'task' is available

vielfarbig, 2022-03-12:
**Describe the bug**
In console there is printed the following uncaught exception. I didn't notice yet that something else is broken by that.  

**To Reproduce**
Steps to reproduce the behavior:
1. In Ubuntu 20.04 execute the `arduino-ide_2.0.0-rc5_Linux_64bit.AppImage` in terminal
2. file > preferences > additional boards manager URLs > `http://arduino.esp8266.com/stable/package_esp8266com_index.json` > OK
3. Restart IDE
4. Boards Manager > Install esp8266 (default) Version 3.0.1 (which is actually 3.0.2 according to terminal output?)

**Error message**
The error message that occured is:
```
daemon INFO Adding libraries dir [dir: /home/user/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries, location: platform]
root ERROR Uncaught Exception: 
root ERROR Error: Implementation error. Neither 'download' nor 'task' is available.
    at ClientReadableStreamImpl.<anonymous> (/tmp/.mount_arduing1VLF0/resources/app/node_modules/arduino-ide-extension/lib/node/grpc-installable.js:15:23)
    at ClientReadableStreamImpl.emit (node:events:394:28)
    at addChunk (node:internal/streams/readable:312:12)
    at readableAddChunk (node:internal/streams/readable:287:9)
    at ClientReadableStreamImpl.Readable.push (node:internal/streams/readable:226:10)
    at Object.onReceiveMessage (/tmp/.mount_arduing1VLF0/resources/app/node_modules/@grpc/grpc-js/build/src/client.js:321:24)
    at Object.onReceiveMessage (/tmp/.mount_arduing1VLF0/resources/app/node_modules/@grpc/grpc-js/build/src/client-interceptors.js:291:178)
    at /tmp/.mount_arduing1VLF0/resources/app/node_modules/@grpc/grpc-js/build/src/call-stream.js:202:74
    at processTicksAndRejections (node:internal/process/task_queues:78:11)
```

ubidefeo, 2022-03-24:
@vielfarbig
can you try and download the latest Nightly Build and see if you still run into this issue?
I have not been able to replicate on my system

per1234, 2022-03-25:
I can still reproduce it with the latest build (2.0.0-rc5-snapshot-4de7737) on both my Windows 10 and Linux machines.

It is not specific to the ESP8266 boards platform, Additional Boards Manager URLs, or even Boards Manager. It also happens when I install platforms from the primary package index as well as when I install libraries via Library Manager.

@ubidefeo try starting the IDE from the command line, install a few random libraries from Library Manager, then check the output in the terminal.

---

Even if it doesn't indicate a serious problem (I have no idea whether it does or not), I would really like to see the output cleaned up. An accumulation of errors and warnings desensitizes us to new ones that could indicate a serious problem, and they also frequently serve as "red herrings" for users experiencing some unrelated problem.

vielfarbig, 2022-03-25:
I can also reproduce with the following nightly build:
```
Version: 2.0.0-rc5-nightly-20220325
Date: 2022-03-25T03:06:41.670Z
CLI Version: 0.21.0 [10107d24]

Copyright © 2022 Arduino SA
```

kittaakos, 2022-05-05:
It still happens with 905b780.

Steps to reproduce:
 - Start the IDE for the first time, and the default `arduino:avr` package will be downloaded automatically.
 - See the error.

<details>
  <summary>Click to see full console output</summary>
  
  ```
/Users/a.kitta/dev/git/arduino-ide/node_modules/.bin/electron . --log-level=debug --hostname=localhost --no-cluster --app-project-path=/Users/a.kitta/dev/git/arduino-ide/electron-app --remote-debugging-port=9222 --no-app-auto-install --plugins=local-dir:../plugins --hosted-plugin-inspect=9339
Debugger listening on ws://127.0.0.1:49891/885403fa-d3c9-4dc3-916e-d06b19c7c531
For help, see: https://nodejs.org/en/docs/inspector
Debugger attached.
Debugger listening on ws://127.0.0.1:49895/e947db3d-ae36-43c5-af8f-234c5e825132
For help, see: https://nodejs.org/en/docs/inspector
Debugger attached.
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///Users/a.kitta/.theia'
root DEBUG Starting the deployer with the list of resolvers [
  LocalDirectoryPluginDeployerResolver {},
  LocalFilePluginDeployerResolver {},
  GithubPluginDeployerResolver {
    unpackedFolder: '/var/folders/z1/xkw1yh5n7rz4n8djprp1mdn80000gn/T/github-remote'
  },
  HttpPluginDeployerResolver {
    unpackedFolder: '/var/folders/z1/xkw1yh5n7rz4n8djprp1mdn80000gn/T/http-remote'
  }
]
root DEBUG Found the list of default plugins ID on env: undefined
root DEBUG Found the list of plugins ID on env: undefined
root DEBUG Found the list of default plugins ID from CLI: local-dir:../plugins
root DEBUG HTTP file upload URL path: /file-upload
root DEBUG Backend file upload cache path: /var/folders/z1/xkw1yh5n7rz4n8djprp1mdn80000gn/T/theia_upload
root INFO Theia app listening on http://localhost:49898.
root WARN The local plugin referenced by local-dir:/Users/a.kitta/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/Users/a.kitta/.arduinoIDE/extensions does not exist.
root DEBUG Backend Object.initialize took: 5.9 ms
root DEBUG Backend Object.configure took: 4.4 ms
root DEBUG Backend MessagingContribution.onStart took: 1.9 ms
root DEBUG Backend LocalizationBackendContribution.initialize took: 42.0 ms
root DEBUG Backend MessagingContribution.configure took: 0.9 ms
root DEBUG Backend DefaultWorkspaceServer.onStart took: 34.6 ms
root DEBUG Backend LocalizationBackendContribution.configure took: 2.6 ms
root DEBUG Backend ElectronTokenBackendContribution.configure took: 0.6 ms
root DEBUG Backend PluginDeployerContribution.initialize took: 0.5 ms
root DEBUG Backend NodeFileUploadService.configure took: 2.6 ms
root DEBUG Backend WebviewBackendSecurityWarnings.initialize took: 0.9 ms
root DEBUG Backend PlotterBackendContribution.initialize took: 0.2 ms
root DEBUG Backend FileDownloadEndpoint.configure took: 0.6 ms
root DEBUG Backend PluginApiContribution.configure took: 0.9 ms
root DEBUG Backend HostedPluginReader.configure took: 1.3 ms
root DEBUG Backend PlotterBackendContribution.configure took: 0.2 ms
root DEBUG Backend TaskBackendApplicationContribution.onStart took: 0.4 ms

DevTools listening on ws://127.0.0.1:9222/devtools/browser/3b67d1cd-1ed0-44e0-9bc3-13288825bfa4
root INFO PluginTheiaDirectoryHandler: accepting plugin with path /Users/a.kitta/dev/git/arduino-ide/plugins/cortex-debug
root DEBUG Resolving "cortex-debug" as a VS Code extension...
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines: { vscode: '^1.52.0' }
root INFO PluginTheiaDirectoryHandler: accepting plugin with path /Users/a.kitta/dev/git/arduino-ide/plugins/vscode-arduino-tools
root DEBUG Resolving "vscode-arduino-tools" as a VS Code extension...
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.2" with engines: { vscode: '^1.46.0' }
root INFO PluginTheiaDirectoryHandler: accepting plugin with path /Users/a.kitta/dev/git/arduino-ide/plugins/vscode-builtin-cpp
root DEBUG Resolving "vscode-builtin-cpp" as a VS Code extension...
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines: { vscode: '*' }
root INFO PluginTheiaDirectoryHandler: accepting plugin with path /Users/a.kitta/dev/git/arduino-ide/plugins/vscode-builtin-json
root DEBUG Resolving "vscode-builtin-json" as a VS Code extension...
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines: { vscode: '0.10.x' }
root INFO PluginTheiaDirectoryHandler: accepting plugin with path /Users/a.kitta/dev/git/arduino-ide/plugins/vscode-builtin-json-language-features
[... 1129 lines cut ...]
    at processTicksAndRejections (node:internal/process/task_queues:78:11)
root ERROR Request loadSketch failed with error: 5 NOT_FOUND: Can't open sketch: reading files: open /Users/a.kitta/Documents/Arduino/libraries/package-365500555/TFT-1.0.6/examples/Esplora: no such file or directory Error: 5 NOT_FOUND: Can't open sketch: reading files: open /Users/a.kitta/Documents/Arduino/libraries/package-365500555/TFT-1.0.6/examples/Esplora: no such file or directory
    at Object.callErrorFromStatus (/Users/a.kitta/dev/git/arduino-ide/node_modules/@grpc/grpc-js/build/src/call.js:31:26)
    at Object.onReceiveStatus (/Users/a.kitta/dev/git/arduino-ide/node_modules/@grpc/grpc-js/build/src/client.js:179:52)
[... cut ...]
  name: 'Arduino_BuiltIn',
  label: '',
  exampleUris: [],
  installable: true,
  deprecated: false,
  location: 0,
  author: 'Arduino',
  availableVersions: [ '1.0.0' ],
  includes: [],
  description: 'Shell Arduino library to provide the standard built-in libraries',
  moreInfoLink: 'https://github.com/arduino-libraries/Arduino_BuiltIn',
  summary: 'The sole purpose of this library is to install the standard Arduino libraries which are typically included with the Arduino IDE.'
}

  ```
</details>

kittaakos, 2022-06-30:
Here are the responses the IDE2 receives when installing/uninstalling a lib or updating the indexes. This example is from the index update, but the received messages clearly show the CLI sends a message where neither `task` nor `download` is defined.

```json
[
    [
        {
            "downloadProgress": {
                "url": "https://downloads.arduino.cc/packages/package_index.tar.bz2",
                "file": "Downloading index: package_index.tar.bz2",
                "totalSize": 41738,
                "downloaded": 0,
                "completed": false
            }
        },
        {
            "downloadProgress": {
                "url": "",
                "file": "",
                "totalSize": 0,
                "downloaded": 41738,
                "completed": false
            }
        },
        {
            "downloadProgress": {
                "url": "",
                "file": "",
                "totalSize": 0,
                "downloaded": 0,
                "completed": true
            }
        },
        {
            "downloadProgress": {
                "url": "https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json",
                "file": "Downloading index: package_esp32_index.json",
                "totalSize": -1,
                "downloaded": 0,
                "completed": false
            }
        },
        {
            "downloadProgress": {
                "url": "",
                "file": "",
                "totalSize": 0,
                "downloaded": 70385,
                "completed": false
            }
        },
        {
            "downloadProgress": {
                "url": "",
                "file": "",
                "totalSize": 0,
                "downloaded": 0,
                "completed": true
            }
        },
        {}
    ],
    [
        {
            "downloadProgress": {
                "url": "https://downloads.arduino.cc/libraries/library_index.json.gz",
                "file": "Downloading index: library_index.json.gz",
                "totalSize": -1,
                "downloaded": 0,
                "completed": false
            }
[... 153 lines cut ...]
                "downloaded": 0,
                "completed": false
            }
        },
        {
            "downloadProgress": {
                "url": "",
                "file": "",
                "totalSize": 0,
                "downloaded": 543,
                "completed": false
            }
        },
        {
            "downloadProgress": {
                "url": "",
                "file": "",
                "totalSize": 0,
                "downloaded": 0,
                "completed": true
            }
        },
        {}
    ]
]

commit 99723a99 (2022-07-02): Update package index on 3rd party URLs change.

Closes #637
Closes #906
commit 2e4719ad (2022-07-04): Update package index on 3rd party URLs change.

Closes #637
Closes #906
commit fd4625cb (2022-07-04): Update package index on 3rd party URLs change.

Closes #637
Closes #906
commit a2bec126 (2022-07-05): Update package index on 3rd party URLs change.

Closes #637
Closes #906
closed 2022-07-08 by kittaakos