arduino/arduino-ide #908: STM32 - selecting STM32CubeProgrammer (SWD)  - always return to "Serial"

AndKe, 2022-03-13:
**Describe the bug**

Selecting  "STM32CubeProgrammer (SWD) "   - does actually select the "Serial" variant.

[image]

**To Reproduce**
With STM32 cube programmer / ST-LINK programmer:
1. select  STM32CubeProgrammer (SWD) 
2. observe that  STM32CubeProgrammer (Serial)  is selected 

**Expected behavior**
would dexpect SWD to be selected

**Screenshots**
See above

**Desktop (please complete the following information):**
Ubuntu 21.10  
Arduino IDE 2.0 rc5

**Additional context**

per1234, 2022-03-14:
Thanks for your report @AndKe. I wasn't able to reproduce the issue:

[image]

Please provide more detailed instructions I can follow:

Did you install the STM32 boards using the Arduino IDE Boards Manager?
If so, what was the URL you put in the **File > Preferences > Additional Boards Manager URLs** field?
Which version did you install?

What do you have selected in the other custom menus (e.g., **Tools > Board part number**)?

When do you find the **Tools > Upload method** selection has changed to "**STM32CubeProgrammer (Serial)**"? Does it happen immediately after you selected **Tools > Upload method > STM32CubeProgrammer (SWD)**, or does it only change after some other action (e.g., switching boards, uploading, restarting the IDE)?

AndKe, 2022-03-14:
@per1234  thanks for looking into this.

My additional boards ULR:
http://dan.drown.org/stm32duino/package_STM32duino_index.json,https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json

My other settings are:
Arduino_Core_STM3
  Board: Generic STMF1
  Board Part : Bluepill F103C8
  U(S)ART support: Enabled(generic Serial)
  USB Support: CDC(generic "serial" supersede U(S)ART"
  USB Speed: Low/Full Speed
  Optimize: Smallest (-Os default)
  C Runtime Library : newlib Nano(default)

It changes back to "serial" instantly  - I open the menu, set it to SWD ,(the menu is collapsed automatically), than I open the menu again, and it's serial.

fpistm, 2023-01-18:
> @per1234 thanks for looking into this.
> 
> My additional boards ULR: http://dan.drown.org/stm32duino/package_STM32duino_index.json,https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json

https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
is deprecated since a while, you should use this one:
https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json

> My other settings are: Arduino_Core_STM3 Board: Generic STMF1 Board Part : Bluepill F103C8 U(S)ART support: Enabled(generic Serial) USB Support: CDC(generic "serial" supersede U(S)ART" USB Speed: [...]

I've tested with latest 2.3.0 and your option. I have no issue.