arduino/arduino-ide #909: Split Editor

markub3327, 2022-03-13:
Hello,

  I need to split the editor into twice windows. The same feature is in VS Code by pressing Split Editor Right (COMMAND+\ on macOS).

Thanks a lot.
Have a nice day.

kittaakos, 2022-03-14:
Possible related: https://github.com/arduino/arduino-ide/issues/552

You can do <kbd>F1</kbd> > type _Split Editors_ > and press <kbd>Enter</kbd> but then you cannot close the split tabs.

CptHolzschnauz, 2022-03-24:
Let's rephrase it to "we would like to have" 

Thank you contributors for all the work done!

PS: RC5 : Still two main issues IMHO: Split view and serial monitor

markub3327, 2022-03-24:
Thanks a lot. I wait to new releases.

CaptainCredible, 2023-09-17:
Really hoping this will be be implemented <3