arduino/arduino-ide #921: IDE crashes when setting invalid Additional Boards Manager URLs

thorv, 2022-03-20:
### Describe the problem

The Arduino IDE crashes at the end of the splash.

### To reproduce

Set the Prefereces / Additional Boards Manager URLs to 
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/
I know this is the wrong URL, but I still don't think the application should crash.

### Expected behavior

I'd like to be able to reconfigure the correct URL, in this case https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json, without the program crashing.
In addition, I think it would be better if the user was notified that the URL was incorrect.

### Arduino IDE version

Zip/Desktop 1.8.19

### Operating system

Windows

### Operating system version

11

### Additional context

I confirmed that the phenomenon is reproduced even in the Windows Sandbox environment.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

PaulStoffregen, 2022-03-20:
Confirmed, reproducible on Linux x86-64.
```
java.io.FileNotFoundException: /home/paul/.arduino15 (Is a directory)
	at java.io.FileInputStream.open0(Native Method)
	at java.io.FileInputStream.open(FileInputStream.java:195)
	at java.io.FileInputStream.<init>(FileInputStream.java:138)
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:234)
	at cc.arduino.contributions.packages.ContributionsIndexer.mergeContributions(ContributionsIndexer.java:179)
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:116)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:483)
	at processing.app.Base.<init>(Base.java:272)
	at processing.app.Base.main(Base.java:150)

```

PaulStoffregen, 2022-03-20:
But this repository is for the new/upcoming 2.0 IDE.  I'm pretty sure @per1234 will move it to the proper place.

thorv, 2022-03-20:
> But this repository is for the new/upcoming 2.0 IDE.
Oh, excuse me for posting to the wrong place.
Can you tell me the correct location for the issue report for version 1.8.x?