arduino/arduino-ide #922: IDE hangs during "Save As..." operation

per1234, 2022-03-20:
### Describe the problem

The IDE hangs permanently during the "**Save As...**" operation under the following conditions:

- IDE was last shut down via "**File > Quit**"
- Currently selected board is from a different platform than the one selected in the previous window
- The language server "**indexing**" process is in progress

### To reproduce

1. Start the Arduino IDE.
1. Select **File > Quit** from the Arduino IDE menus.
   ❗ This exact shutdown method is required. The bug does not occur after shutting the IDE down via **File > Close** or the native window close control on the title bar (i.e., **X** or **🔴**).
1. Start the Arduino IDE.
1. Select **File > New** from the Arduino IDE menus.
1. Select a board of a different platform from the **Tools > Board** menu in the Arduino IDE.
   **ⓘ** e.g., if you originally had "**Arduino Uno**" ("**Arduino AVR Boards**" platform) selected, you could select "**Arduino Nano 33 BLE**" ("**Arduino Mbed OS Nano Boards**" platform).
   ❗ The issue is timing sensitive, so you will have a short time window if selecting a board from a lightweight platform such as "**Arduino AVR Boards**". The boards of any of the "**Arduino Mbed OS \_\_\_\_ Boards**" platforms are recommended as these provide a large window.
1. Immediately select **File > Save As...** from the Arduino IDE menus.
1. Immediately click the <kbd>**Save**</kbd> button.
   ❗ This must be done while the IDE is still in the state where it shows something like "**indexing 42/188**" at the left side of the status bar.

🐛 The IDE hangs forever in the state where the window title is "**index.html**" and contains spinning dots:

[image]

### Expected behavior

The sketch that was produced by the "**Save As..." operation is loaded in the Arduino IDE.

### Arduino IDE version

2.0.0-rc5-snapshot-f36df02 Date: 2022-03-17T10:22:00.589Z

### Operating system

Windows

### Operating system version

10

### Additional context

The first version I can reproduce this fault with is https://github.com/arduino/arduino-ide/commit/e6b9d4e2aa0503a0cb99e472051e1e49f487e2c7. I can not reproduce the fault with https://github.com/arduino/arduino-ide/commit/69ac1f4779589d0d21ce3d37c180b3393ad6156c.

The "**Save As...**" process stopped working correctly at https://github.com/arduino/arduino-ide/commit/112153fb965f63d952d126c8244cd3f84f0a1a1b (https://github.com/arduino/arduino-ide/issues/829), so it is possible that the bug was introduced somewhere between https://github.com/arduino/arduino-ide/commit/69ac1f4779589d0d21ce3d37c180b3393ad6156c and https://github.com/arduino/arduino-ide/commit/e6b9d4e2aa0503a0cb99e472051e1e49f487e2c7 but was not visible due to https://github.com/arduino/arduino-ide/issues/829

---

I see the following output printed on the command line after I do a **File > Quit**:

```text
root ERROR Uncaught Exception:
root ERROR Error: Connection got disposed.
    at Object.dispose (C:\ide 2\18-arduino-ide_2.0.0-rc5-snapshot-f36df02_Windows_64bit\resources\app\node_modules\vscode-jsonrpc\lib\main.js:904:25)
    at C:\ide 2\18-arduino-ide_2.0.0-rc5-snapshot-f36df02_Windows_64bit\resources\app\node_modules\vscode-ws-jsonrpc\lib\socket\connection.js:14:41
    at CallbackList.invoke (C:\ide 2\18-arduino-ide_2.0.0-rc5-snapshot-f36df02_Windows_64bit\resources\app\node_modules\vscode-jsonrpc\lib\events.js:62:39)
    at Emitter.fire (C:\ide 2\18-arduino-ide_2.0.0-rc5-snapshot-f36df02_Windows_64bit\resources\app\node_modules\vscode-jsonrpc\lib\events.js:121:36)
    at closeHandler (C:\ide 2\18-arduino-ide_2.0.0-rc5-snapshot-f36df02_Windows_64bit\resources\app\node_modules\vscode-jsonrpc\lib\main.js:240:26)
    at CallbackList.invoke (C:\ide 2\18-arduino-ide_2.0.0-rc5-snapshot-f36df02_Windows_64bit\resources\app\node_modules\vscode-jsonrpc\lib\events.js:62:39)
    at Emitter.fire (C:\ide 2\18-arduino-ide_2.0.0-rc5-snapshot-f36df02_Windows_64bit\resources\app\node_modules\vscode-jsonrpc\lib\events.js:121:36)
    at WebSocketMessageReader.fireClose (C:\ide 2\18-arduino-ide_2.0.0-rc5-snapshot-f36df02_Windows_64bit\resources\app\node_modules\vscode-jsonrpc\lib\messageReader.js:111:27)
    at WebSocketMessageReader.fireClose (C:\ide 2\18-arduino-ide_2.0.0-rc5-snapshot-f36df02_Windows_64bit\resources\app\node_modules\vscode-ws-jsonrpc\lib\socket\reader.js:67:19)
    at C:\ide 2\18-arduino-ide_2.0.0-rc5-snapshot-f36df02_Windows_64bit\resources\app\node_modules\vscode-ws-jsonrpc\lib\socket\reader.js:24:18
```

I see the same output printed multiple times after the fault occurs.

---

I suspect it is related to https://github.com/arduino/arduino-ide/issues/657

---

I was only able to reproduce the fault by saving during the "indexing" operation triggered by the board selection. The fault did not occur for me when saving during subsequent indexing operation triggered by editing the sketch.

---

I was not able to reproduce the issue with the "**Arduino > Language: Log**" setting enabled.

---

Originally reported at https://forum.arduino.cc/t/saving-files-is-a-crap-shoot/970560/5

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

xinkiknix, 2022-05-01:
I get this issue (IDE2.0.0-rc6) when creating a new sketch while another one is open.  When saving the new one the IDE never returns to working state for the new sketch, the older sketch is still working ok.

ubidefeo, 2022-05-04:
@xinkiknix

very useful information, it helps digging into it.
Thank you

kittaakos, 2022-06-03:
Excellent job on the step. I could reproduce the bug on my slower Windows machine.

I do not know if it's related, but when I try to open the dev tools in the windows with the spinner, I see such warnings in the logs. I do not know yet if it's related:

```
Attempting to call a function in a renderer window that has been closed or released.
Function provided here: bundle.js:2:1080768
Attempting to call a function in a renderer window that has been closed or released.
Function provided here: bundle.js:2:1080768
Attempting to call a function in a renderer window that has been closed or released.
Function provided here: bundle.js:2:1080768
Attempting to call a function in a renderer window that has been closed or released.
Function provided here: bundle.js:2:1080768
```

per1234, 2022-06-11:
I am not able to reproduce this by following the procedure I described above using Arduino IDE 2.0.0-rc7. However, that result is meaningless because I am also no longer able to reproduce it using the 2.0.0-rc5-snapshot-f36df02 build I was using when I submitted this issue. This means something has changed in the environment the IDE runs in on my machine.

The issue was always somewhat intermittent, but I was able to reproduce it >50% of the time, and have done so occasionally with later versions of the IDE, in addition to running into the problem incidentally. However, I have made quite a bit of effort to reproduce it without any luck so I am not in a good position for further investigation or testing.

I do believe the issue still occurs in 2.0.0-rc7 because there is a report of it here:

https://forum.arduino.cc/t/high-contrast-theia-theme-changed-and-other-issues/1001842/10

>I getting a lot of white screens that that just rotate a circle of blue balls,
This is occuring quite often when using save file as....
The header [top screen] has small arduin img followed by index.html
the line below it has file edit sketch tools help......

If anyone finds a procedure that can be followed to reliably reproduce the issue, please provide detailed instructions here. That would be very valuable in any efforts to fix this.

ubidefeo, 2022-06-13:
@per1234 @kittaakos 
is this still an issue?

per1234, 2022-06-13:
It appear the answer is yes. Please see my previous comment for everything I know about the current status.

xinkiknix, 2022-06-20:
yesterday (rc7) I had an issue with IDE, I had 2 instances open with the same name, showing the directory name in tha Tab.  
[image]
[ptillisch](https://forum.arduino.cc/u/ptillisch) sugested to  "View: Reset Workbench Layout" . I executed the command and go the blank screen with :
[image]
I had to restart the IDE to get back to normal situation. Changes where saved.

Dracwula, 2022-12-16:
Reproduceable steps under IDE 2.0.3
-------------------------------------
1. Create a new sketch called "ABC", and [SAVE]
2. Optionally, Edit or Reload this same sketch ABC and [SAVE AS] new sketch name "XYZ" 
3. I think here at this point, it will open a new IDE app instance, notice the duplicated TAB XYZ, the internal contents are identical.

To eliminate the duplication tab under IDE 2.0.3
-------------------------------------
4. Close IDE that has XYZ sketch open
5. Go into where "XYZ.ino" is located via Windows Explorer
6. Rename XYZ as "XYZ-2.ino"
7. Double-click "XYZ-2.ino" from here, IDE will open and prompt you with new folder name to match "XYZ-2", [Confirm] it, this will eliminate the duplication tab under IDE 2.0.3

This maybe a quick & dirty method to eliminate the duplicate sketch tabs, unless someone can provide better solution.