arduino/arduino-ide #925: Show Sketch folder on examples different than IDE 1.

KurtE, 2022-03-22:
### Describe the problem

With RC5 of the IDE2, if I open an example sketch, like the ESP32 FFat_Test sketch

If I then do the command: Sketch->Show Sketch Folder 
It opens up a folder in my Temp directory...

If I do the same thing on IDE 18.19 it shows the example folder within the library. 

This may be by design. But if so should be documented. 

This may also be sort of another aspect for issue #595 

### To reproduce

Described in the problem description. 

### Expected behavior

Not sure, both have merit. 

With IDE 1 behavior - I would use this command to open it up in the library, and then from there I might open up an editor on the library to view/change the library code.   Or maybe get to the Readme file for the library.

However if you are doing editing of the example, and wish to see your actual edits, then showing the temporary folder may be correct. 

### Arduino IDE version

I am using RC5: but tried:  nightly build 0322

### Operating system

Windows

### Operating system version

Windows 10

### Additional context

[image]

I did a search but did not find a duplicate, but could have missed one.  Also as I mentioned maybe part of #595 
Also as I mentioned could be by design. 

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

ubidefeo, 2022-03-24:
@per1234 
as you know, every open workspace (Window) keeps a reference to temp file until saved.
Since the Examples are not editable I guess the Example Sketch is considered an unsaved copy.
This needs some investigation on that side

per1234, 2022-03-25:
I think it is the correct approach. Without it, the IDE would have to force a "Save As" operation on the users with "Auto save" enabled whenever they wanted to make a throw-away adjustment to an example (e.g., setting Wi-Fi credentials, changing the blink interval).

As @KurtE I noticed that it is no longer possible to use **Sketch > Show Sketch Folder** to access the library location when an example is open. I used to recommend that as a trick to the users who were having trouble finding the library files (e.g., platform bundled libraries which are under a hidden folder). However, the "go to" capabilities of the language server now gives us a much more powerful tool for accessing library source code, so I think in the long term the breakage of my old hacky approach won't be any loss.

So I would just kick it over to content and support to decide about whether to document the change.

KurtE, 2022-03-25:
@per1234, to me it is sort of an interesting question. that is with IDE1 you see the actual location AND you don't have to do a save as to make temporary changes.   However if you try to close it or do a save then it will ask you if you wish to save... 

Also side note: maybe different/related issue.  I believe when I open up an example with IDE1 it puts the item in the "Open Recent..." list and it appears like IDE2 does not.  Which at times is useful.  Like I come back in the morning and go which one of the examples was it I opened yesterday...  Let alone in my case having to scroll through 100s of libraries to get back to it in the example list.

per1234, 2022-03-31:
>you don't have to do a save as to make temporary changes

I see now that Arduino IDE 1.x gives example sketches special treatment by disabling the "**Save when verifying or uploading**" behavior, even if the user has it enabled.

So an alternative to the Arduino IDE 2.x current behavior would be to emulate the Arduino IDE 1.x behavior by opening examples in place, but with the auto save disabled. However, I'm not sure I would like that inconsistency in saving behavior as a user. I actually am not personally a fan of auto save at all, so I probably don't have the best perspective on what the users who do like it want. Maybe @91volt has some opinions on this.

>when I open up an example with IDE1 it puts the item in the "Open Recent..." list and it appears like IDE2 does not.

That is correct and I have also missed that behavior of Arduino IDE 1.x.

I think it is related in that it must be that the "open example as new sketch" behavior is causing the sketch to not be listed under the **File > Open Recent** menu. If I use **File > Open...** to open an example (which actually does open it in place, and editable). I then see the sketch under the **File > Open Recent** menu.

 Would you mind opening a dedicated issue about it? Even though related to the same underlying mechanism, I think it is distinct enough from the "Show Sketch Folder" behavior change that it will be more effective to track and decide on it separately. Even if the decision is to keep the current "open example as new sketch" behavior, it should still theoretically be possible for the IDE to add the example to the **File > Open Recent** menu.

KurtE, 2022-03-31:
Thanks I will open new one. 

Another maybe same/or related or different issue is something I have never used until trying now. 

was suppose I mark my library as being in development mode:
https://arduino.github.io/arduino-cli/0.21/library-specification/#development-flag-file

Now I open up an example from that library, such that I can make edits before releasing the library or new release of the library. 
Again on IDE1, If I do a save it does... 

But on IDE2, I have not tried, but the show sketch folder is still showing me the temp directory folder.

per1234, 2022-04-01:
>But on IDE2, I have not tried, but the show sketch folder is still showing me the temp directory folder.

That is correct. It seems that the presence of the file is not even [reported to the IDE by Arduino CLI](https://arduino.github.io/arduino-cli/latest/rpc/commands/#library).

The library developer does have the ability to edit examples by opening them using **File > Open**, but that is the same for all libraries regardless of whether they have a `.development` flag file.