arduino/arduino-ide #927: Spurious output in Serial Monitor after upload

per1234, 2022-03-24:
### Describe the problem

The first output printed to Serial Monitor after an upload is some "garbage" characters.

[image]

### To reproduce

#### Equipment

- Arduino board that uses the ATmega16U2 USB chip:
  - Arduino Uno
  - Arduino Uno Mini
  - Arduino Mega

#### Steps

1. Upload a sketch to your board that prints to `Serial`:
   ```cpp
   void setup() {
     Serial.begin(9600);
     Serial.println("hello world");
   }
   void loop() {}
   ```
1. Open the "**Serial Monitor**" view.

🐛 The output starts with some unexpected content:

```text
□□□□□□□□□□□hello world
```

### Expected behavior

Serial monitor output always reflects the data sent by the board.

### Arduino IDE version

2.0.0-rc5-snapshot-4de7737

### Operating system

Windows

### Operating system version

10

### Additional context

The number of `□` in the demonstration matches the number of characters that are printed by the sketch.

---

The spurious output does not occur when the output is triggered by resetting the board, so it is specific to the upload operation.

---

The issue does not occur when using Arduino IDE 1.x

---

I have only been able to reproduce this issue with the Arduino boards that use an ATmega16U2 USB to serial adapter chip.

I could not reproduce it when using boards with other USB interfaces:

- Native USB (Leonardo, Nano 33 IoT)
- FTDI FT232R (Nano, Pro Mini)
- WCH CH340 (3rd party boards)

---

Originally reported at https://forum.arduino.cc/t/serial-monitor-contains-garbage-after-upload/972312

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

woolseyj, 2022-05-04:
I am seeing similar issues with version **2.0.0-rc6** of the IDE running on **macOS 12.3.1** with an **Arduino Uno WiFi Rev2** board.  If I upload via the IDE, I get mostly garbled output in the Serial Monitor, however, if I upload via the **arduino-cli** tool, the output is correct whether it is viewed through the **screen** command or with **arduino-cli monitor**.

cmaglie, 2023-02-08:
Reopening because the fix has been reverted: https://github.com/arduino/serial-monitor/pull/39

We need to investigate this issue further and find a better solution.

ubidefeo, 2023-02-09:
@per1234 
I have noticed that the number of characters is equal to the ones printed, but the line ending is missing.
For "hello world" it should be 12 chars, rather than 11 when `println()` is used.

**I am wild-guessing here**, but it could be that on some Windows (not int'l English systems) the character encoding plays a role (ISO-8859-12 vs 8859-1 or other encoding).
As I climb the conspiracy theory ladder I might say that how that particular text output view handles the text is to be further investigated.

The flow of operations is as follows, and it causes a chopping of the output printed, but it shouldn't grant the replacement of valid characters with squares.

* upload recipe 
  * reset board
  * upload
  * reset board
* board is reset and runs through its setup, starting to print the string
* serial monitor opens serial port, resetting the board again
* board has not finished to print the output
* new output from second reset is printed after anything that was already printed before

shortening the string might produce a different output

these are our findings for now

SimonArrowhead, 2023-08-27:
@ubidefeo @per1234 I have the same issue but number of squares is deifferent than number of characters:

[image]

I tried switching system, keyboard and regional settings from polish to united states and number of squares was the same.
I switched arduino to usb 2.0 and squares disappeared but hello word was printed two time in serial monitor. Before uploading it to arduino and after uploading. The first one has some strange characters in it. 

[image]

Tried this on my old mac (MacBook Air 13 inch, Early 2015, macOS Monterey 12.6.5) and at the beginning it looked like it was also printing two times in one run but after I ran it twice it worked just fine (printed just once every run without any squares). I supposed something was left in the memory from previous tests.

[image]

Jorgetrujilloroman, 2025-05-13:
Hello,

In case it helps, I have been testing and replicating this issue on different boards and architectures, and here are my findings:

[image]

**1*** The result on the UNO R3 is correct, all the ASCII characters are displayed, but at the very beginning, there are some garbage characters displayed, as reported on the tracked GitHub Issue: 

[image]

**2*** Arduino Due is giving the same result that Uno R3: Displaying garbage but the entire list:

[image]

**3*** The Uno R4 WiFi board is failing (not printing the entire list):

[image]

**4*** Same issue as in the Uno R4 WiFi: Garbage characters and printing incomplete list

[image]

*The workaround is to press the RST button once, so the board resets and all the characters are correctly displayed. It also helps if you add a 2 seconds delay in the setup()*

jchunkins, 2025-09-07:
Just going to leave this here... I am using Mac OS 15.6.1 using the Arduino IDE Version: 2.3.6. I am using the [esp8266 board](https://github.com/esp8266/Arduino) with a D1 mini clone. 

After googling around for this issue, it was suggested that the baud rate for this board (and maybe other boards) defaults to outputting boot information on load at 74880. I switched my sketch and serial monitor to use this value and the output from the monitor was readable. To be clear I added this in my `setup()`:

```
Serial.begin(74880); 
```

Instead of gibberish, the boot information was shown:

```

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v0004a400
~ld
rf cal sector: 1020
freq trace enable 0
rf[112] : 0�
```

Maybe this will help.