arduino/arduino-ide #928: COM port number reset after upload

GClairet, 2022-03-24:
### Describe the problem

On Win10 Pro x64, with my Arduino Micro, after I upload the firmware, the COM port is resetted to the first COM port in the list (COM1) instead of returning to the COM9 in my case. With the serial monitor or serial plotter I have to reselect the COM port each time and I miss the beginning of the frames.

### To reproduce

Select a COM port, run the serial monitor, compile and upload firmware: after uploading, the COM port has charged.
Maybe something to help, when I click on the COM port list, 2 COM ports are selected: COM9 and COM1. I have not manage to unselect the COM1.
[image]

### Expected behavior

Returning automatically to the previously selected COM port (COM9) after uploading.

### Arduino IDE version

2.0.0-rc5

### Operating system

Windows

### Operating system version

10 Pro x64 (20H2)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-03-31:
Thanks for your report @GClairet.

>the COM port is resetted to the first COM port in the list (COM1) instead of returning to the COM9

We already have a report about the port change at https://github.com/arduino/arduino-ide/issues/710

>when I click on the COM port list, 2 COM ports are selected: COM9 and COM1. I have not managed to unselect the COM1.

I'm not aware of any previous reports of this bug (though I vaguely recollect encountering it myself), so let's make this issue about that exclusively. Feel free to comment on https://github.com/arduino/arduino-ide/issues/710 if you have any new information on the port switch bug.

I did my best to reproduce this multiple port selection bug but didn't manage to accomplish it. Does it occur for you consistently?

GClairet, 2022-04-06:
Hi per1234,
OK for the report #710, I will read it.
Concerning the 2 COM ports selected, the bug bothered me for 2 days before finding out what manipulation triggered it. I think I selected the COM port by the menu "Tools/Port: " and later, after having unplugged-replugged my Arduino Micro into another USB port without closing Arduino 2.0, I re-selected the port by the drop-down list near the save button. I also remember clicking the "Select other board and port..." link in that drop-down list. 
The bug remained after relaunching Arduino IDE 2.0. Only rebooting the PC cleared the selection of 2 COM ports.
Maybe the multiple ways to change this setting is the cause or maybe I used it wrong.
Now only I use the "Tools/Port: " menu to select the port and only use the drop list to check which port is selected.

kittaakos, 2022-09-01:
Was this fixed @davegarthsimpson?

https://github.com/arduino/arduino-ide/pull/1328/files#diff-e42c82bb67e277cfa4598239952afd65db44dba55dc7d68df619dfccfa648279R564

davegarthsimpson, 2022-09-01:
> Was this fixed @davegarthsimpson?

yes from my own testing this was fixed by [this PR](https://github.com/arduino/arduino-ide/pull/1328).

@per1234 if you get the chance it'd be great to get your confirmation!

per1234, 2022-09-01:
Thanks so much for those instructions Dave. I verified it was fixed by https://github.com/arduino/arduino-ide/pull/1328 (I can still reproduce it while using the build from the previous commit).

closed 2022-09-01 by per1234