arduino/arduino-ide #929: Serial Plotter with zoom and auto-scale button

GClairet, 2022-03-24:
### Describe the request

Adding some setting  to Serial Plotter:
- A button to to disable the auto-scale
- A zoom to have more points in the x-axis

### Describe the current behavior

The auto-scale do not allow me to have a good and rapid reading of the graph evolution. The too low number of points in x-axis does not allow me to have a view of the high speed datas for a long enough time.

### Arduino IDE version

2.0.0-rc5

### Operating system

Windows

### Operating system version

10 Pro x64 (20H2)

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-03-30:
Adding the requested zoom capability would likely also resolve https://github.com/arduino/arduino-ide/issues/803 ("Or make it a user selectable option").

fralbo, 2022-07-21:
I think auto-scale is maybe not really expectable, mainly for real time plotting, but it would be very nice to be able to define the horizontal scale.