arduino/arduino-ide #934: Lots of Serial Monitor content without line break causes IDE to become unresponsive

per1234, 2022-04-01:
### Describe the problem

If the Arduino board prints a large amount of data without a line break to Serial Monitor, the IDE becomes unresponsive.

### To reproduce

#### Equipment

- Any Arduino board

#### Steps

1. Upload the following sketch to the Arduino board:
   ```cpp
   void setup() {
     Serial.begin(115200);
   }
   void loop() {
     Serial.print("hello");
   }
   ```
1. Open the "**Serial Monitor**" view.
1. Select "**115200 baud**" from the dropdown baud rate menu at the top right corner of the "**Serial Monitor**" view.
1. Wait a couple minutes.
   **ⓘ** It will likely occur much sooner than that, especially with a board that has native USB capability
1. Try to use the Arduino IDE UI.

🐛 The UI is unresponsive.

### Expected behavior

Arduino IDE handles this condition gracefully.

That could even be a limit on the line length that will be displayed, since it is more likely that problematic line lengths would be caused by a missing line break in the sketch than intentionally formatting.

### Arduino IDE version

2.0.0-rc5-snapshot-c9b498f

### Operating system

Windows

### Operating system version

10

### Additional context

There were similar reports of Serial Monitor impact on CPU performance, but those were resolved by https://github.com/arduino/arduino-ide/pull/524

- https://github.com/arduino/arduino-ide/issues/397
- https://github.com/arduino/arduino-ide/issues/519

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

ubidefeo, 2022-06-13:
@per1234 
how does RC7 behave on this?

per1234, 2022-06-13:
@ubidefeo still same problem with 2.0.0-rc7